iiab/iiab #234: confusing use of mysql_service between 'os'.yml and computed_vars.yml

jvonau, 2017-09-09:
Should be handled like apache_config_dir is done, nothing in 1-prep and use os.yml to set the variable to account for differences in the distros.

holta, 2017-09-28:
@jvonau can you explain this during a live call if poss — or resolve?

holta, 2017-10-25:
@jvonau plz help bring to a conclusion as IIAB 6.5's feature freeze approaches in 5 days?

Or can we defer this to [IIAB 6.6](https://github.com/iiab/iiab/milestone/3)'s Milestone ?

tim-moody, 2017-10-25:
and should consider doing mysql the ansible way by removing from aggregates and triggering with meta for roles that need it. install and enable would be True

holta, 2017-11-01:
Database installation approach extensively discussed during Monday Oct 30 community/team call (http://minutes.iiab.io) and documented with PR #476 "Clarifies MySQL & PostgreSQL in local_vars.yml & default_vars.yml"

Plz re-open as nec for IIAB 6.6 as our Feature Freeze has now passed.

holta, 2017-11-02:
@jvonau asked me to re-open this during today's community/team call (http://minutes.iiab.io) even if this slips to IIAB 6.6 Milestone.

holta, 2017-11-12:
Moving to [IIAB 6.6](https://github.com/iiab/iiab/milestone/2) milestone as Code Freeze was last Thursday and Feature Freeze 2 weeks ago.

holta, 2017-11-13:
@tim-moody encouraged us to make this happen (during today's community/team call @ http://minutes.iiab.io) but understands there's no PR so all agreed to wait for the [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) timeframe.

holta, 2018-07-18:
@jvonau please update or close your tkt, if poss?

Thanks, as [IIAB 6.6](http://download.iiab.io/6.6/) release is now converging!

We can also move this to our [IIAB 6.7 Milestone](https://github.com/iiab/iiab/milestone/4) if you prefer.

holta, 2018-11-04:
@jvonau if this is important for IIAB 6.7 (release target December) please definitely shout.

holta, 2018-12-20:
PR #1279

holta, 2019-01-02:
Can I close this tkt now that computed_vars.yml no longer exists?

FYI `mysql_service:` values were just fixed across many of the [OS].yml files in /opt/iiab/iiab/vars thanks to yesterday's PR https://github.com/iiab/iiab/pull/1348

closed 2019-01-03 by holta

holta, 2019-01-04:
Closed during our community/team call (http://minutes.iiab.io) as the consensus:
- this is resolved by PR #1348
- computed_vars.yml lives on within by Stage 0