iiab/iiab #237: investage reported word press failure

jvonau, 2017-09-11:
TASK [wordpress : Get the WordPress software] **********************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true,
"msg": "failed to create temporary content file: timed out"}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=312  changed=57   unreachable=0    failed=1

holta, 2017-09-11:
@jvonau do you have more context what OS etc were involved?  (fresh install using master presumably?)

jvonau, 2017-09-11:
[GRS on xsce-devel](https://groups.google.com/forum/#!topic/xsce-devel/UQsDAU8So9U) was installing Ubuntu 16.04

holta, 2017-09-25:
@jvonau please resolve if poss?

arky, 2017-09-26:
This problem is causing Ubuntu, Debian, CentOS and Fedora builds to fail on Jenkins server.
 
Please consider marking this issue as 'critical' and 'release blocker'.

holta, 2017-09-26:
Thanks @arky for the report!

Do these CentOS and Debian ansible issues have anything to do with WordPress?
(If not, can you please consider a new ticket?)

holta, 2017-09-26:
I'm moved both of @arky's issues to #314 and #315 as they are WordPress downloading issues (generally a solved issue, insofar as all downloads are sovable!)

closed 2017-09-26 by holta

holta, 2017-09-26:
@jvonau: feel free to re-open this WordPress downloading issue if it's truly a systemic failure?

(Rather than an intermittent/inevitable downloading glitch...)

jvonau, 2017-09-26:
I feel that the fail to download might be related to the available
bandwidth available at the build location. Might have to state a longer
timeout or add a timeout value if one is not already present.