iiab/iiab #240: Sugarizer 0.8 to 0.9

holta, 2017-09-11:
Anyone know if this upgrade is as easy as changing [roles/sugarizer/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/sugarizer/defaults/main.yml) from:
`sugarizer_version: 'sugarizer-0.8'`
to:
`sugarizer_version: 'sugarizer-0.9'`

...while dropping sugarizer-0.9.tar.gz http://download.iiab.io/packages/ ?

holta, 2017-09-11:
Or: defer to [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) is this upgrade is complex?

jvonau, 2017-09-11:
Placing sugarizer-0.9.tar.gz on d.iiab.io would allow a quick edit for local testing. I'd use local_vars to define sugarizer_version: during testing as a quick down & dirty way to see how 0.9 behaves.

holta, 2017-09-13:
@georgejhunt thank you for having posted [download.iiab.io/packages/sugarizer-0.9.zip](http://download.iiab.io/packages/sugarizer-0.9.zip) from [github.com/llaske/sugarizer/archive/v0.9.zip](https://github.com/llaske/sugarizer/archive/v0.9.zip)

Is there something I can smoke test, per @jvonau suggestion above or similar?

holta, 2017-09-13:
RELATED: during every boot Sugarizer (0.8 for now) says:

   [FAILED] Failed to start sugarizer.service.

As reconfirmed by Curt Thompson, who finds this red error on boot to be jarring for new folk especially.

(Perhaps this boot-time ugliness can this be fixed with Sugarizer 0.9 now?)

jvonau, 2017-09-13:
from #240 warnings.txt 
> Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:9] Failed to parse sec value, ignoring: 10                       # Restart service after 10 seconds if node service crashes
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog               # Output to syslog
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog                # Output to syslog
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:1] Unknown section 'unit'. Ignoring.
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:9] Failed to parse sec value, ignoring: 10                       # Restart service after 10 seconds if node service crashes
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:10] Failed to parse output specifier, ignoring: syslog               # Output to syslog
Sep 10 07:28:41 raspberrypi systemd[1]: [/etc/systemd/system/sugarizer.service:11] Failed to parse output specifier, ignoring: syslog                # Output to syslog
Sep 10 07:28:41 raspberrypi systemd[25141]: sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory
Sep 10 07:28:41 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.
Sep 10 07:28:41 raspberrypi systemd[25147]: sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory
Sep 10 07:28:41 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.
Sep 10 07:28:42 raspberrypi systemd[25152]: sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory
Sep 10 07:28:42 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.
Sep 10 07:28:42 raspberrypi systemd[25168]: sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory
Sep 10 07:28:42 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.
Sep 10 07:28:42 raspberrypi systemd[25171]: sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory
Sep 10 07:28:42 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.
Sep 10 07:28:42 raspberrypi systemd[1]: sugarizer.service: Start request repeated too quickly.
Sep 10 07:28:42 raspberrypi systemd[1]: Failed to start sugarizer.service.
Sep 10 07:28:42 raspberrypi systemd[1]: sugarizer.service: Failed with result 'exit-code'.

sugarizer.service: Failed at step EXEC spawning /usr/bin/node: No such file or directory

holta, 2017-09-14:
@jvonau do we keep #240 open here?

(@georgejhunt has implemented Sugarizer 0.9 with #245 and #246)

jvonau, 2017-09-14:
I'd check the logs and see if the logging has shut up and confirm the installation first.

holta, 2017-09-14:
@georgejhunt suspects Sugar's Journal may still/not be working due to MongoDB -- removing MongoDB entirely may be an option, to save substantial disk space if so.

holta, 2017-09-14:
Error #193 [Ansible fails all subsequent runs on Raspbian Stretch: Sugarizer "npm : Depends: node-gyp (>= 0.10.9)"] appears to have returned with Sugarizer 0.9 ?  In any case, "git pull; ./runansible" incorporated @georgejhunt's #245 and #246 and this failure ensured:

```
TASK [sugarizer : Actually get it installed] ***********************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["apt-get", "install", "-y", "npm"], "delta": "0:00:02.213079", "end": "2017-09-14 12:25:47.241875", "failed": true, "rc": 100, "start": "2017-09-14 12:25:45.028796", "stderr": "E: Unable to correct problems, you have held broken packages.", "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " npm : Depends: node-gyp (>= 0.10.9) but it is not going to be installed"], "warnings": ["Consider using apt module rather than running apt-get"]}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=388  changed=89   unreachable=0    failed=1
```

holta, 2017-09-14:
FYI @jvonau @georgejhunt the red error on bootup that Curt was trying to help us get rid of (so we stop scaring away newcomers) is still present -- as confirmed in /var/log/boot.log during our very latest reboot:

`Failed to start sugarizer.service.`

georgejhunt, 2017-09-14:
Try "systemctl disable sugarizer.service". I think that will remove the
error message.

holta, 2017-09-14:
Indeed @georgejhunt the red error disappeared from boot (/var/log/boot.log) thanks.  Can you make this (or something like this) true for everyone now?

```
root@box:~# systemctl disable sugarizer.service
Removed /etc/systemd/system/multi-user.target.wants/sugarizer.service.
```

holta, 2017-09-15:
Indeed i've reconfirmed #193 is solved for Sugarizer 0.8

i.e. "./runtags sugarizer" works repeatedly on Raspbian Stretch, unlike earlier in September.

So: it remains unclear why #193 [Sugarizer "npm : Depends: node-gyp (>= 0.10.9)"] flared up again this morning for Sugarizer 0.9 (when #245 & #246 overlayed the new Sugarizer 0.9 upon its 0.8 predecessor during "git pull; ./runansible").

holta, 2017-09-18:
FYI the related MongoDB bloat issue #254 ("what's causing 801MB /var/lib/mongodb ?") may be greatly helped (or at least mitigated) by #260 ("MongoDB: 2 improvements to reduce DB bloat") that @georgejhunt & @jvonau are now working on.

holta, 2017-09-19:
#262 is also making progress here (see #254 for the latest overview!)

jvonau, 2017-09-22:
> ● sugarizer.service
   Loaded: loaded (/etc/systemd/system/sugarizer.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-09-21 18:19:46 EDT; 1min 44s ago
 Main PID: 931 (node)
   CGroup: /system.slice/sugarizer.service
           └─931 /usr/bin/node sugarizer.js
Sep 21 18:19:48 box.lan sugarizer[931]:     at Object.Module._extensions..js (module.js:579:10)
Sep 21 18:19:48 box.lan sugarizer[931]:     at Module.load (module.js:487:32)
Sep 21 18:19:48 box.lan sugarizer[931]:     at tryModuleLoad (module.js:446:12)
Sep 21 18:19:48 box.lan sugarizer[931]:     at Function.Module._load (module.js:438:3) code: 'MODULE_NOT_FOUND' }
Sep 21 18:19:48 box.lan sugarizer[931]: js-bson: Failed to load c++ bson extension, using pure JS version
Sep 21 18:19:48 box.lan sugarizer[931]: Thu, 21 Sep 2017 22:19:48 GMT express deprecated app.configure: Check app.get('env') in an if statement at sugarizer.js:13:5
Sep 21 18:19:49 box.lan sugarizer[931]: Thu, 21 Sep 2017 22:19:49 GMT connect deprecated multipart: use parser (multiparty, busboy, formidable) npm module instead at node_modules/connect/lib/middleware/bodyParser.js:56:20
Sep 21 18:19:49 box.lan sugarizer[931]: Thu, 21 Sep 2017 22:19:49 GMT connect deprecated limit: Restrict request size at location of read at node_modules/connect/lib/middleware/multipart.js:86:15
Sep 21 18:19:49 box.lan sugarizer[931]: Sugarizer server listening on port 8089...
Sep 21 18:19:49 box.lan sugarizer[931]: Presence Server is listening on port 8039...

#304 was developed on .42 while working out #267 Needs a clean install test

jvonau, 2017-09-22:
Sugarizer does not offer a logout option but Shutdown & Restart are available!!! 
[image]

holta, 2017-09-26:
@llaske replied that he's working on this login/logout requirement (https://github.com/llaske/sugarizer/issues/129) for the Sugarizer v1.0 milestone.

Sugarizer 0.9 will have to do for now, for the IIAB 6.4 release.

Feel free to re-open or re-package this issue for [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) later?

closed 2017-09-26 by holta