iiab/iiab #249: "apt install ansible" OR download.iiab.io/packages + pip to speed RPi installs?

holta, 2017-09-16:
Currently we install Ansible 2.2.0 as follows which takes about 7 minutes:
```
cd /opt/iiab/iiab/scripts/
./ansible
```
Is it possible we could accelerate our RPi routine (by 5 min or so) using either method below?
1.  "apt install ansible" (attractive if there's a flag to specify Ansible 2.2.0!)
2. "wget http://download.iiab.io/packages/<our-precompiled-ansible-2.2.0>" then pip install that?

jvonau, 2017-09-16:
Might get some inspiration from #164 but if you are looking to just deploy a tar file, think a virtual environment might be required to be able to scoop the files from one directory. This one looked useful and just over a year old so it's more current than whatever is in play now https://gist.github.com/dpalomar/96b826dac5c2e8b62cbf4c86dbd1c9df

holta, 2017-09-16:
Update for @tim-moody : to lower RPi contributors' very high barriers to entry (constant refrain at Open Source Summit in LA last week: "involved new developers within 1 HOUR MAXIMUM") @jvonau & @georgejhunt have bravely tried to help me find a fast-installing Ansible 2.2.0 for IIAB 6.4 (via apt, pip, .deb, whatever) but no luck just yet :/

@jvonau surmises that Admin Console jobs are "SCHEDULED" but never actually moving forward due to:

```
Think the jobs are running... the db is being updated... but can't retrieve the data from the db.... -rw------- 1 root root 12288 Sep 16 15:58 /opt/admin/cmdsrv/cmdsrv.0.2.db
root@box:/opt/iiab# date
Sat Sep 16 16:11:13 EDT 2017
```

jvonau, 2017-09-16:
> /root/.ansible_async:
416426711106.853  528555768525.10127
tail -f /root/.ansible*/*7
{"changed": true, "end": "2017-09-16 04:23:02.892949", "stdout": "                                     \n   _   __  ___    _     _ _          \n  | | / / / _ \\  | |   (_) |         \n  | |/ / / /_\\ \\ | |    _| |_ ___    \n  |    \\ |  _  | | |   | | __/ _ \\   \n  | |\\  \\| | | | | |___| | ||  __/   \n  \\_| \\_/\\_| |_/ \\_____/_|\\__\\___|   \n                                     \nhttps://learningequality.org/ka-lite/\n                                     \n         version 0.17.3\n                                     \n-------------------------------------------------------------------\nWARNING: You are installing KA-Lite as root user!\n    Installing as root may cause some permission problems while running\n    as a normal user in the future.\n-------------------------------------------------------------------\n\nCopying database file from /usr/local/lib/python2.7/dist-packages/kalite/database/templates/data.sqlite to /library/ka-lite/database/data.sqlite\nCopying /usr/local/lib/python2.7/dist-packages/kalite/database/templates/content_items/content_khan_en.sqlite to /library/ka-lite/database/content_khan_en.sqlite\nNot running in interactive mode; skipping content pack download.\nGenerating 2048-bit RSA encryption key (may take a few minutes; please wait)...\nDone!\nDevice 'raspberrypi' has been successfully initialized.\nSuperuser created successfully.\nCopying static media...\nAnnotating availability of all content, checking for content in this directory: /library/ka-lite/content\n[INFO] [2017-09-16 04:23:01,661] kalite: Annotating content for language: en, channel: khan\n\nThis may take several minutes depending on system resources...\n[INFO] [2017-09-16 04:23:01,667] kalite: Annotation complete for language: en, channel: khan\n\nCONGRATULATIONS! You've finished setting up the KA Lite server software.\nYou can now start KA Lite with the following command:\n\n    kalite start", "cmd": ["/usr/local/bin/kalite", "manage", "setup", "--username=Admin", "--password=changeme", "--noinput"], "start": "2017-09-16 04:22:32.248757", "delta": "0:00:30.644192", "stderr": "Generating a new secret key file /library/ka-lite/secretkey.txt...", "rc": 0, "invocation": {"module_args": {"creates": null, "executable": null, "chdir": null, "_raw_params": "/usr/local/bin/kalite manage setup --username=Admin --password=changeme --noinput", "removes": null, "warn": true, "_uses_shell": false}}, "warnings": []}

holta, 2017-09-17:
FYI a clean install starting with apt's Ansible 2.2.1 on Raspbian Stretch Lite was attempted using http://download.iiab.io/6.4/rpi/load-min-ansibletest-vpn.txt which at least ran to completion, even though Admin Console did not work.  Assisted by @jvonau.  Recap:
```
# apt -y install ansible python-kerberos python-selinux python-winrm (>= 0.1.1) python-xmltodict cowsay sshpass git bzip2 file findutils gzip hg svn sudo tar which unzip xz zip libselinux-python
# ABOVE LINE FAILED DUE TO THESE 6: (>= 0.1.1) hg svn which xz libselinux-python
# SO THE FOLLOWING WAS RUN INSTEAD

apt -y install python-kerberos python-selinux python-winrm python-xmltodict cowsay sshpass git bzip2 file findutils gzip sudo tar unzip zip

apt install ansible

root@raspberrypi:~# ansible --version
ansible 2.2.1.0
  config file = /etc/ansible/ansible.cfg
  configured module search path = Default w/o overrides
```
FYI near the very end of ./runansible, munin failed with the following error:
```
TASK [munin : Create admin user] >
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "invalid version number '1.7.0.post20170423015547'"}
        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

< PLAY RECAP >
127.0.0.1                  : ok=297  changed=212  unreachable=0    failed=1
```
So: vars/local_vars.yml was changed to "munin_install: False" and "munin_enabled: False" in order to allow the full bash script (load-min-ansibletest-vpn.txt) to complete...

jvonau, 2017-09-17:
> rpi/load-min-ansibletest-vpn.txt which at least ran to completion, even

>
Missing from the initial load list but installing in scripts/ansible and
was not installed, debs: python-keyczar   pips: pyrax pysphere boto
dnspython

jvonau, 2017-09-17:
Notes from first pass of apt ansible install [debs.txt](https://github.com/iiab/iiab/files/1308851/debs.txt)

jvonau, 2017-09-17:
console needs to adjust path to ansible

jvonau, 2017-09-17:
#181

holta, 2017-09-17:
apt correctly now installs Ansible 2.2.1 and has been tested with IIAB's Admin Console, thanks to @jvonau's heroic overnight work.

This is now part of IIAB 6.4 Beta -- full explanation here:
https://groups.google.com/forum/#!topic/unleashkids/SrNFM6FMnUU

closed 2017-09-17 by holta

jvonau, 2017-09-17:
#256 #259

jvonau, 2017-09-17:
>  Above installs a more future-proof Ansible 2.2.1+
sed -i 's/StrictVersion/LooseVersion/' /usr/lib/python2.7/dist-packages/ansible/modules/core/web_infrastructure/htpasswd.py
 ABOVE APPLIES TEMPORARY PATCH https://github.com/ansible/ansible/pull/20202
 needed by munin until Ansible 2.3.2 or 2.4.0, to overcome Ansible bug:
    TASK [munin : Create admin user] ********************************
    fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true,
    "msg": "invalid version number '1.7.0.post20170423015547'"}