iiab/iiab #254: what's causing 801MB /var/lib/mongodb ?

holta, 2017-09-16:
Sugarizer is *not* causing this (certainly not directly) as /var/lib/mongodb is only 33MB with Sugarizer 0.9 fully installed via medium-sized http://wiki.laptop.org/go/IIAB/local_vars.yml

This ballooning 801MB /var/lib/mongodb arises from http://wiki.laptop.org/go/IIAB/local_vars_big.yml (on master on RPi3) and so must be arising as a result of one of these 10:

1. squid
2. dansguardian
3. wondershaper
4. cups
5. samba
6. phpmyadmin
7. elgg
8. dokuwiki
9. moodle
10. monit

holta, 2017-09-16:
@georgejhunt observed this 801MB "pollution" is essentially just these 3 files:
```
root@box:/var/lib/mongodb# ls -l journal/*
-rw------- 1 mongodb nogroup 268435456 Sep 16 10:52 journal/prealloc.0
-rw------- 1 mongodb nogroup 268435456 Sep 16 04:37 journal/prealloc.1
-rw------- 1 mongodb nogroup 268435456 Sep 16 04:37 journal/prealloc.2
```
Could this be some kind of buggy attempt at a Sugar Journal ?

Still baffling to us that this appears on one Sugarizer RPi3 (built w/ [big local_vars](http://wiki.laptop.org/go/IIAB/local_vars_big.yml)) but not the other (built w/ [medium local_vars](http://wiki.laptop.org/go/IIAB/local_vars.yml)).

holta, 2017-09-16:
@llaske can you confirm Y/N if Sugarizer's Journal stored in /var/lib/mongodb/journal ?

Do you know why installing on Sugarizer 0.9 on RPi3 (part of IIAB 6.4, on lastest Raspbian Stretch Lite) mysteriously sometimes ends up with 769MB in here?
```
root@box:/var/lib/mongodb# ls -l journal/*
-rw------- 1 mongodb nogroup 268435456 Sep 16 10:52 journal/prealloc.0
-rw------- 1 mongodb nogroup 268435456 Sep 16 04:37 journal/prealloc.1
-rw------- 1 mongodb nogroup 268435456 Sep 16 04:37 journal/prealloc.2
```
_Thanks if you know what might be going on with the 3 mysterious files above!_

holta, 2017-09-17:
A 2nd two-hour-long install of http://download.iiab.io/6.4/rpi/load-big-vpn.txt confirms the problem, if in a slightly diff way &mdash; this occurs during the MongoDB's own installation:
```
root@raspberrypi:/var/lib/mongodb/journal# ls -l
total 524676
-rw------- 1 mongodb nogroup 268435456 Sep 17 00:05 j._0
-rw------- 1 mongodb nogroup    397312 Sep 17 00:04 prealloc.1
-rw------- 1 mongodb nogroup 268435456 Sep 17 00:05 prealloc.2
```
Line 101 of /opt/iiab/iiab/roles/mongodb/templates/mongod.conf looks like it may solve this:
```
# Disable data file preallocation - will often hurt performance (false by default)
#noprealloc = true
```
Can we officially change this to **noprealloc = true** ?  For Raspbian at the bare minimum?

llaske, 2017-09-17:
The MongoDB database is usually in /data/db directory.  My oldest DB - three years ago - has a size of 200Mb only.
The journal you mentioned should be the MongoDB journal, not the Sugarizer Journal.

holta, 2017-09-17:
@georgejhunt can you confirm if the following (32KB out of the box) database is in fact the Sugarizer database in use within IIAB?
```
root@box:~# du -hs /opt/iiab/moodle/mod/data/db/*
8.0K    /opt/iiab/moodle/mod/data/db/access.php
8.0K    /opt/iiab/moodle/mod/data/db/install.xml
4.0K    /opt/iiab/moodle/mod/data/db/log.php
4.0K    /opt/iiab/moodle/mod/data/db/services.php
4.0K    /opt/iiab/moodle/mod/data/db/subplugins.php
4.0K    /opt/iiab/moodle/mod/data/db/upgrade.php
32K     total
```

georgejhunt, 2017-09-17:
No, a file in /opt/iiab/moodle is not the sugarizer database

holta, 2017-09-17:
@georgejhunt has 2 MongoDB fixes smoke-tested ready: #260

holta, 2017-09-18:
@georgejhunt made progress with PR's #260 & #262 ("try to shrink mongodb") which appear to have reduced MongoDB's footprint to 1 single prealloc files...not perfect yet..._but far better than 2 or 3 such files of 256MB each:_
```
root@raspberrypi:/var/lib/mongodb/journal# ls -l
total 262148
-rw------- 1 mongodb nogroup 268435456 Sep 18 15:51 prealloc.0
```

holta, 2017-09-18:
Clarif: PR #260 added "noprealloc = true" in...
/etc/mongod.conf (for MongoDB >= 2.6)

PR #262 also added "noprealloc = true" in...
/etc/mongodb.conf (for MongoDB <= 2.4)

holta, 2017-09-19:
With the exact same code ("git clone https://github.com/iiab/iiab --depth 1") _we got yet another result this time around:_
```
root@raspberrypi:~# ls -l /var/lib/mongodb/journal
total 400
-rw------- 1 mongodb nogroup 409600 Sep 18 21:41 tempLatencyTest

root@raspberrypi:~# du -hsc /var/lib/mongodb/*
404K    /var/lib/mongodb/journal
0       /var/lib/mongodb/mongod.lock
404K    total
```

holta, 2017-09-19:
FYI @georgejhunt / All,

The results regrettably differ every time &mdash; even on identical RPi3 hardware &mdash; when running the exact same code: (iiab master with medium-sized local_vars.yml, via http://download.iiab.io/6.4/rpi/load-vpn.txt)
```
root@raspberrypi:/var/lib/mongodb# ls -l /var/lib/mongodb/journal
total 524288
-rw------- 1 mongodb nogroup 268435456 Sep 18 23:34 prealloc.0
-rw------- 1 mongodb nogroup 268435456 Sep 18 23:34 prealloc.1

root@raspberrypi:/var/lib/mongodb# du -hsc /var/lib/mongodb/*
513M    /var/lib/mongodb/journal
0       /var/lib/mongodb/mongod.lock
513M    total
```

holta, 2017-09-19:
Latest install's quite similar for once:
```
root@raspberrypi:~# ls -l /var/lib/mongodb/journal/
total 524292
-rw------- 1 mongodb nogroup 268435456 Sep 19 07:00 prealloc.0
-rw------- 1 mongodb nogroup 268435456 Sep 19 07:00 prealloc.1

root@raspberrypi:~# du -hsc /var/lib/mongodb/
513M    /var/lib/mongodb/
513M    total
```

holta, 2017-09-21:
FYI @georgejhunt this is the exact stage where 256MB prealloc files are created:
```
TASK [mongodb : enable services] ***********************************************
changed: [127.0.0.1] => (item={u'name': u'mongodb'})
```
(as determined by watching MongoDB's playbook live, during a new install on RPi3 w/ Medium-sized local_vars.yml via http://download.iiab.io/6.4/rpi/load-vpn.txt)

arky, 2017-09-21:
Here is an alternative solution: You can prevent mangodb creating large files by turning on "smallfiles=true". This makes sure all journal prealloc files are small.

See Journaling section of MangoDB conf documentation https://docs.mongodb.com/manual/core/journaling/#journaling-journal-files

holta, 2017-09-22:
@arky @georgejhunt MongoDB still pollutes our filesystem with a random number of 256MB files (0, 1, 2 or 3 such files...often just 1 single 256MB during many installs...but really a different number during each install, as described above).

Even after applying "smallfiles=true" and "noprealloc = true" to both /etc/mongodb.conf and /etc/mongod.conf &mdash; as reconfirmed on a clean-install-from-scratch of 6.4-RC2 + all patches from master.

Changes to both files are having no visible effect: is it possible our MongoDB is misconfigured and is completely ignoring /etc/mongodb.conf and /etc/mongod.conf both?

_In Any Case: before we resort to desperate measures like disabling "half-or-whatever" of Sugarizer's apps by eliminating MongoDB entirely out of frustration (which we'll do if we truly absolutely must!) let's try to get a more solid grip on cause-and-effect here first in coming days :-)_

holta, 2017-09-22:
Ref #263 ("Mongo conf")

arky, 2017-09-22:
Can we try turning off both "storage.mmapv1.smallFiles" and "storage.mmapv1.preallocDataFiles". This change will have performance impact. 

A closer reading of MangoDB docs was illuminating. There has been changes in namespace of storage API and they also recommend YAML config style now.  

**MangoDB 3.4(Older INI style)**
storage.mmapv1.preallocDataFiles = false
storage.mmapv1.smallFiles = true

**MangoDB 3.4 (YAML style)**
storage:
   mmapv1:
     preallocDataFiles: false
      smallFiles: true

https://docs.mongodb.com/manual/reference/configuration-options/#storage-options
https://docs.mongodb.com/manual/administration/configuration/

holta, 2017-09-22:
@georgejhunt can you help me implement @arky's "preallocDataFiles" suggestion on #254 here?

Unfortunately I find no reference to "preallocDataFiles" inside our Ansible playbook for MongoDB:
```
root@box:/opt/iiab/iiab/roles/mongodb# grep -ri preallocDataFiles *
<returns nothing>
```
@arky does this stand to work with MongoDB 2.4.14 that it appears (??) we are stuck with for now?
```
root@box:~# apt list mongodb
Listing... Done
mongodb/stable,now 1:2.4.14-4 armhf [installed]
```

arky, 2017-09-22:
@holta I have no idea why it doesn't work. In MangoDB 2.4 the setting is '`noprealloc = true`' which is already set 'mongdb.conf' template. 

https://github.com/iiab/iiab/blob/26de481e93e418d61e470011c9a8455129fcaebb/roles/mongodb/templates/mongod.conf

holta, 2017-09-25:
@arky regrettably placing your suggestion:
```
storage.mmapv1.preallocDataFiles = false
storage.mmapv1.smallFiles = true
```
into /etc/mongod.conf and /etc/mongodb.conf did not stop the creation of the 256MB prealloc file during a completely clean install of [IIAB 6.4 master](http://download.iiab.io/6.4/rpi/) (with [medium-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml)) on RPi3:
```
root@raspberrypi:~# ls -l /var/lib/mongodb/journal/
total 262148
-rw------- 1 mongodb nogroup 268435456 Sep 25 19:17 prealloc.0
```
Any other suggestions?!  Of course one could give up, cynically pointing out that "MongoDB" is true to its name, being Mongo Large even if accomplishing nothing else :/

@georgejhunt a rethink might be in order, as I'm suspicious after ~10 days that there may never be any cause-and-effect &mdash; between any changes we make to /etc/mongod.conf and /etc/mongodb.com &mdash; and actual filesystem/results.  _If for example we might possibly have MongoDB configured wrong to begin with, or some such?_

holta, 2017-10-13:
CLARIF- the up-to-three 256MB files polluting our filesystems are generated during the very 1st step of Ansible's mongodb playbook here: (a time-consuming step, that takes 2 minutes to complete as can be seen below)
```
2017-10-13 12:42:00,074 p=3000 u=root |  TASK [mongodb : Install mongodb required packages] *****************************
2017-10-13 12:43:09,709 p=3000 u=root |  changed: [127.0.0.1] => (item=mongodb-server)
2017-10-13 12:45:00,536...
```

holta, 2017-10-21:
Resolving for now, as we have a double-fisted workaround installed (#427).  As well as another to further ruggedize Sugarizer's install (#436) using Node.js 6.x

Plz re-open as nec, if MongoDB is in future being put a solid foundation, with a clean+reliable install path, e.g. for Sugar's Journal or other reasons.

arky, 2018-07-02:
@holta  Here is the definite answer to this problem.  If journal is enabled then the mongodb daemon will creates the prealloc files when the service is started. <b>MongoDB requires journal and prealloc to be enable for data integrity and crash recovery. Removing these files will also cause performance issues.</b> 

Anyway here is the solution. In /etc/mongodb.conf, set the following settings. 
journal=false
noprealloc = true

Works as expected in Debian 9 MongoDB 3.2.11-2+deb9u1 version. 

These [following commits #427](https://github.com/iiab/iiab/pull/427
) are not needed.

holta, 2018-07-02:
> MongoDB requires journal and prealloc to be enabled for data integrity and crash recovery. Removing these files will also cause performance issues.

Thanks @arky!

Please see the latest at:
- PR #814 (which re-activated MongoDB, for the new Sugarizer 1.0) in IIAB 6.6/master
- PR #851 (which offers some possible fixes to Sugarizer 1.0) though very regrettably it breaks short URL's like http://192.168.0.x/sugarizer (which teachers and students really do depend on)

holta, 2018-07-10:
Let's close this tkt if we confirm the ~768MB (3 prealloc files) are at least consistently being created this year.

Context: Last year (2017) these 3 files were Not being created consistently: some installations (on the very same RPi 3 hardware, identical conditions) created 1 such 256MB prealloc file, some two 256MB prealloc files, some all three 256MB prealloc files.

(Let's also check other OS's like Ubuntu 18.04, and Debian 9 to confirm.)

holta, 2018-07-10:
QUESTION: what's the best way to wipe all traces of MongoDB and Sugarizer from an existing system, to reinstall cleanly?

FYI/FWIW latest results below appear inconsistent, upon installing Sugarizer 1.0 with IIAB 6.6/master across ~4 common OS's...

...Raspbian Lite:
```
root@box:~# du -hsc /var/lib/mongodb/journal/*
257M    /var/lib/mongodb/journal/prealloc.0
257M    /var/lib/mongodb/journal/prealloc.1
256M    /var/lib/mongodb/journal/prealloc.2
769M    total
```
...Ubuntu 18.04/Server and 18.04/Desktop both show:
```
root@box:~# du -hsc /var/lib/mongodb/journal/*
100M    /var/lib/mongodb/journal/WiredTigerLog.0000000002
100M    /var/lib/mongodb/journal/WiredTigerPreplog.0000000001
100M    /var/lib/mongodb/journal/WiredTigerPreplog.0000000002
300M    total
```
...Ubuntu 16.04/Desktop:
```
root@box:~# du -hsc /var/lib/mongodb/journal/*
388K    /var/lib/mongodb/journal/tempLatencyTest
388K    total
```
...Debian 9.4:
```
(forthcoming)
```

arky, 2018-07-12:
@holta  Prealloc are essential files prefilled with zero bytes. They will not have consistent file sizes, if that's what you are looking for.

holta, 2018-07-12:
> They will not have consistent file sizes

FWIW the sizes are consistent after IIAB's installation of MongoDB.

But each machine can be very different (even reinstalling on the same machine can lead to a very different result FYI).  Very odd.

I am seeing a lot of MongoDB flakiness even when prealloc files are in place.  This morning @llaske provided these tips to try to deal with MongoDB, and Sugarizer's frequent inability to connect to its own database there:

> Regarding difficulty to connect to MongoDB, I can suggest a tips to avoid this sort of issue.
> The idea is to launch 2 times "mongod" at RPi startup:
> - The first time with an option "--repair" to fix existing locked file if any. Could happen if you unplug your RPi without doing a safe shutdown. It's why we bought a RPi box with a physical power off button. The command "mongod --repair" just fix database then exit (don't really launch MongoDB).
> - The second time like today "--port 27018 ...". This time it really launch MongoDB.
> 
> It's what we've done in our current docker container: https://github.com/llaske/sugarizer-server/blob/master/docker/Dockerfile-mongodb.tpl#L2

holta, 2019-02-23:
Thanks @llaske for opening https://github.com/llaske/sugarizer-server/issues/27 "MongoDB [allocates] up to 800MB during install process"

In place of https://github.com/llaske/sugarizer/issues/130

holta, 2020-08-28:
@jvonau can we resolve this now, in your opinion?

holta, 2020-09-03:
> @jvonau can we resolve this now, in your opinion?

This issue seems to have miraculously solved itself, at least judging from BIG-sized installs of IIAB 7.2 pre-releases (master) onto these 4 OS's:

- 64-bit RaspiOS on RPi 4
- 64-bit Ubuntu on RPi 4
- Debian 10.5 on VM
- Linux Mint 20 on VM

Let's close/resolve this for now (and by all means re-open if this recurs!)

closed 2020-09-03 by holta

llaske, 2020-09-03:
> This issue seems to have miraculously solved itself

Could be related to upgrade of MongoDB API we've done in [Sugarizer Server 1.2.0](https://github.com/llaske/sugarizer-server/releases/tag/v1.2.0) to remove deprecated API call.
Just guessing.