iiab/iiab #255: enable IIAB browsing *during* WhatsApp/IP messaging

holta, 2017-09-17:
Currently, a typical Android phone will cut off the user's 3G mobile data connection to the Internet when they're interacting with a nearby Internet-in-a-Box.  Increasingly this is not longer acceptable: a growing number of users will no longer accept interruption of their mobile data(*) when connecting their Android phone to a local Internet-in-a-Box (typically over USB tether, Bluetooth or WiFi).

(*) often this is a very limited 3G data plan in the developing world, used for messaging with family, which is increasingly replacing pay-per-use SMS, as it's perceived to be less expensive (and better) than classicac text messaging.

Is a separate APK browser perhaps necessary, so Android users can truly read/learn from their IIAB device in many/most/all diverse conditions?  What are the best way(s) we can enable Android phone users to reliably/instantly access their IIAB (or Internet-in-a-Purse) content whenever they want and need to, thruout their daily lives?

Recap: increasingly people need immediate and bureaucracy-free access to http://[IIAB-IP-ADDRESS] _at the exact same time as_ WhatsApp messaging continues over their personal + very limited 3G mobile data connection.  This reading/interacting with IIAB materials needs to work *while* Android phone users are chatting with their spouse/family via a 3G mobile data IP-based messaging app (like WhatsApp or similar).  An "Internet in a Box" APK (i.e. Android app, that is not much more that a browser) might be one great way to make such hassle-free local routing happen?

Background Context: pricey/classical SMS is (increasingly) falling out of favor among developing world go-getters especially, folks who increasingly no longer tolerate interruptions to their 3G mobile data connection (however meager...)  Consequently: they're urgently asking us to enable offline Wikipedia research and watching of offline medical videos (etc) that will enrich their minds _without_ disrupting their very basic 3G connection/lifeline~

_Thanks @darkenvy & all for your design feedback._

jvonau, 2017-09-17:
Noted that avahi is installed, configured, and appears to beacon what we want.
> Sep 10 06:46:14 raspberrypi avahi-daemon[358]: Service "MGMT console at raspberrypi " (/services/schoolserver.service) successfully established.
Sep 10 06:46:15 raspberrypi avahi-daemon[358]: Files changed, reloading.
Sep 10 06:46:15 raspberrypi avahi-daemon[358]: Loading service file /services/ssh.service.
Sep 10 06:46:16 raspberrypi avahi-daemon[358]: Service "raspberrypi" (/services/ssh.service) successfully established.

I did not note if there was a multicast route present, that would be important when dealing with a multi-homed host. Worth digging deeper as the apk could use mdns to find the "box" with a pre-defined service to home-in on without needing to know the hostname or ip address of the server beforehand.

jvonau, 2017-10-18:
Refs: https://www.quora.com/Is-it-possible-to-use-Wi-Fi-and-3G-simultaneously-on-a-smartphone-If-not-why-not
download booster: https://www.youtube.com/watch?v=cM9-q7jDMaY

holta, 2017-10-24:
@darkenvy: excellent preliminary discussion begun during today's community/team call [*] on pulling the critical+central need together &mdash; with or without an Android app, Keep In Touch!

[*] see http://minutes.iiab.io Item 3. @ Oct 23, 2017

_Context in a "3G 3rd World" : we need to nail this or similar for [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) to ship in January or so &mdash; such that middle-class users in emerging economies / developing countries can use their personal IIAB's for high-bandwidth educational videos etc unimpeded &mdash; while at the very same time **preserving** their family/professional/social lifelines (WhatsApp over 3G mobile data typically)._

CLARIF #1: WhatsApp on Android, or similar low-bandwidth social apps, have essentially replaced SMS in recent years, across many of these upper/middle-class demographics &mdash; while at the same time these are folk who still very much want + deserve a high-bandwidth (FREE) connection to their in-home/in-school IIAB library.

CLARIF #2: Sometimes WhatApp/etc use institutional WiFi for live Internet, offered by an employer or school, rather than 3G mobile data.  Yet another reason Bluetooth or a physical cable may be preferred for smartphone(s) needing access to IIAB / Internet-in-a-Purse.

holta, 2017-10-30:
Plz see summary notes from Oct 23 & Oct 26 calls @ http://minutes.iiab.io where this (#255) was discussed among many.

holta, 2018-08-24:
@darkenvy  is making actual progress designing an IIAB Android app — that allows near-simultaneous use of WhatsApp!

For now he's assuming IIAB will almost always be accessed over Wi-Fi.

But in future there a possibility IIAB might possibly be a accessed from smartphones over Bluetooth or with an actual USB wire, in an Internet-in-a-Purse model or similar.

Reno's using [React Native](https://facebook.github.io/react-native/) which is a [JavaScript library](https://en.wikipedia.org/wiki/React_(JavaScript_library)) from Facebook.  Its WebKit-based browser may be minimal but fully works for our purposes (not interrupting medical clinicians' workflow, who also want WhatsApp IM's while using IIAB!)"

Related: #608 "Captive Portal experimental support summary in FAQ.IIAB.IO ?"

_Please write up a quick outline here when you can Reno, so others in the field can help guide requirements & final product!_

arky, 2018-08-27:
@darkenvy  Please kindly keep me posted about your exploration with JavaScript libraries. 

Our friends at Mozilla are working on interesting Web API for decentralized web. I think these experimental API would be interesting to look at in future.

holta, 2018-10-10:
@darkenvy plz give us an update on the most plausible attack vectors to crack this nut?

(...to assist others in framing the dilemma at the very minimum?)

jvonau, 2025-04-06:
2025 now, what is the status?

psiie, 2025-04-06:
Okay, this made me laugh! 8 years! WOW

Yea, as far as I know, this is still a problem. Actually I just checked. I connected to a WiFi with a captive-portal-redirect. I marked it as "use connection as-is" to keep the captive-portal unresolved. All data ceased to flow inbound or outbound (as it is being routed through the WiFi now).

I don't think there will ever be a solution to this problem, not a REAL one anyways, as this is a design _feature_ for phones. 

Solutions?
1. drop the IIAB connection at intervals to allow messages to come in (bandaid)
2. actually... thats all I got :P

psiie, 2025-04-08:
Huh, I was wondering if a split tunnel VPN solution would work, but disregarded it thinking that the underlying issue would still be there.