iiab/iiab #257: "kalite manage retrievecontentpack download en" fails nightly

holta, 2017-09-17:
This download fails, freezing IIAB installs without warning, particularly (around) 6AM Eastern Time when KA Lite servers apparently go offline on a (nightly) basis:
http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip

This occurs as part of "kalite manage retrievecontentpack download en" during the last step of IIAB install scripts like http://download.iiab.io/6.4/rpi/load.txt

(Trapping this error might later be advisable, so that hobbyist installers aren't left high and dry?!)

nataren, 2017-09-22:
`http://pantry.learningequality.org/` should be reachable now.

holta, 2017-11-22:
Great News KA Lite is beefing up their download mechanisms &mdash; both of the following are marked as blockers that need to be solved prior to [KA Lite 0.17.4](https://github.com/learningequality/ka-lite/milestone/64)'s release (*) &mdash; with Benjamin Bach and others are working hard on these:
* https://github.com/learningequality/ka-lite/pull/5545 "Harden download robustness"
* https://github.com/learningequality/ka-lite/issues/5528 "Video downloads break when connection is unstable"

(*) possibly in December 2017??

benjaoming, 2017-11-24:
@holta I've changed `retrievecontentpack` to use the same download mechanisms as video file downloads in https://github.com/learningequality/ka-lite/pull/5545

Hopefully, unifying the download mechanisms can make KA Lite more reliable.

I don't know why pantry.learningequality.org would be down, I'll make sure it has monitoring configured in a few days. We haven't been keeping track of its uptime. If you notice anything periodic as you indicated, please don't hesitate to contact us.

The changes introduced would be grateful for a bit of testing, so will ping back in this issue. No changes affect the content packs or videos, so no one needs to re-download anything, it's just a matter of software improvements (without regressions we hope).

holta, 2018-06-12:
Has been working well enough for 6+ months so am closing now.

Feel free to re-open as nec!

closed 2018-06-12 by holta

benjaoming, 2018-06-14:
Thanks @holta - that's good to know!