iiab/iiab #258: ansible path

tim-moody, 2017-09-17:
add to local facts:

ansible_playbook_program - full path to ansible-playbook executable
ansible_program - full path to ansible executable

jvonau, 2017-09-17:
From PR#249, think this might work
ansible_program: {{ ansible_facts.ansible_env._ }}
ansible_playbook_program: {{ ansible_facts.ansible_env._}}-playbook

tim-moody, 2017-09-17:
"ansible_env": {
    "LANG": "en_US.UTF-8",
    "CMDSRV_DIR": "/opt/admin/cmdsrv",
    "JOURNAL_STREAM": "8:17439",
    "PWD": "/opt/admin/cmdsrv",
    "PATH": "/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin",
    "INVOCATION_ID": "14570bc06b5a40e8891c26b1d67d73cf"
  },

which variable? Nothing has the path to ansible and ansible-playbook

jvonau, 2017-09-17:
shows up in -m setup, that is where I pulled it from

tim-moody, 2017-09-17:
OK. I guess the _ variable name didn't parse; could they have chosen a poorer name. Wonder how long before someone will change it.

I would still prefer to set a more meaningful name in local facts and isolate future chaos.

jvonau, 2017-09-17:
`ansible_program` lives [here](https://github.com/iiab/iiab-admin-console/tree/master/vars) in the os.yml files.

closed 2017-09-17 by tim-moody