iiab/iiab #261: "iiab-cmdsrv-ctl GET-KIWIX-CAT" replies "No Response from IIAB-CMDSRV"

holta, 2017-09-18:
When installing IIAB 6.4 on #259 (using http://download.iiab.io/6.4/rpi/load-venvtest-vpn.txt calls "git clone https://github.com/jvonau/iiab -b python_stack") the command-line updating of Kiwix's catalog appears to fail:
```
root@raspberrypi:~# /usr/bin/iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Error": "**No Response from IIAB-CMDSRV**"}
```
Strangely Kiwix's catalog *is* updated (~1.5MB from download.kiwix.org/library/library.xml = http://mirror.download.kiwix.org/library/library.xml presumably) despite the above "spurious" error message?!
```
root@raspberrypi:/etc/iiab# ls -l kiwix_catalog.json
-rw-r--r-- 1 root root 1475743 Sep 18 09:31 kiwix_catalog.json
```
@jvonau can you shed some more light

tim-moody, 2017-09-18:
Wonder if this message is coming from the wrapper. iiab-cmdsrv-ctl sends a command to iiab-cmdsrv and polls for a return within the timeout period. If it times out it stops polling, but the command could continue in the cmdsrv. See iiab-cmdsrv-ctl line 39. Current timeout is 10 seconds.

tim-moody, 2017-09-18:
iiab-cmdsrv-ctl could prompt the user to continue and either stop or return to polling.

any good pythonic way to prompt?

holta, 2017-09-18:
New Info: on brand new installs (like http://download.iiab.io/6.4/rpi/load-venvtest-vpn.txt) GET-KIWIX-CAT _really does fail_ (the 1st time!) with this different error:
```
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Error": "IIAB-CMDSRV failed to initialize","Alert": "True"}
```
As seen in Admin Console (stale 2015-era ZIM files are offered for download, after running "systemctl restart iiab-cmdsrv.service" and waiting ~30 sec) and confirmed by this smaller < 1MB catalog here:
```
root@raspberrypi:~# ls -l /etc/iiab/kiwix_catalog.json
-rw-r--r-- 1 root root 915527 Sep 18 12:07 /etc/iiab/kiwix_catalog.json
```
THEN, on subsequent runs of "iiab-cmdsrv-ctl GET-KIWIX-CAT" Kiwix's catalog is correctly downloaded & installed, *despite* this error indicating problems:
```
root@raspberrypi:~# iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Error": "No Response from IIAB-CMDSRV"}
```
Reconfirmed by the larger ~1.5 MB catalog here: (also confirmed in Admin Console)
```
root@raspberrypi:~# ls -l /etc/iiab/kiwix_catalog.json
-rw-r--r-- 1 root root 1475743 Sep 18 12:31 /etc/iiab/kiwix_catalog.json
```

tim-moody, 2017-09-18:
Received reply:  {"Error": "IIAB-CMDSRV failed to initialize","Alert": "True"} is probably correct. You issued the command before the cmdsrv had initialized.

holta, 2017-09-19:
@georgejhunt helped me nail what appears to be the root cause:

If you run "iiab-cmdsrv-ctl GET-KIWIX-CAT" within about 45sec of starting CMDSRV (e.g. using "systemctl restart iiab-cmdsrv.service" or "cd /opt/admin/cmdsrv/; ./iiab-cmdsrv") it will then permanently and erroneously displays error message "No Response from IIAB-CMDSRV".

This includes all subsequent runs of "iiab-cmdsrv-ctl GET-KIWIX-CAT" (despite the fact such commands Are Working, when run more than ~45sec after CMDSRV was launched).

So: I've now added a "sleep 60" into all load/install scripts @ http://download.iiab.io/6.4/rpi/

holta, 2017-09-19:
FYI @tim-moody has somehow fixed Admin Console with the correct /usr/bin instead of /usr/local/bin ansible paths for Raspbian inside /opt/admin/cmdsrv/cmdsrv.conf (https://github.com/iiab/iiab-admin-console/pull/26) so I've removed the following from all load/install scripts at http://download.iiab.io/6.4/rpi after smoke-testing succeeded:
```
sed -i 's/local\/bin\/ansible/bin\/ansible/g' /opt/admin/cmdsrv/cmdsrv.conf
# ABOVE TEMPORARY FIX FOR ADMIN CONSOLE ON ANSIBLE 2.2.1+ ON RASPBIAN STRETCH:
# Changes /usr/local/bin to /usr/bin (for both Ansible paths) on lines 23 & 24:
# https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates\
/cmdsrv.conf.j2
# To solve this cleanly, Tim may want to create vars/Raspbian.yml alongside:
# https://github.com/iiab/iiab-admin-console/blob/master/vars/Debian.yml

systemctl restart iiab-cmdsrv.service; sleep 60
# "sleep 60" nec due to: github.com/iiab/iiab/issues/261
# Without this pause "iiab-cmdsrv-ctl GET-KIWIX-CAT" (below) will permanently
# issue (false) error messages "No Response from IIAB-CMDSRV"
```

jvonau, 2017-09-19:
https://github.com/iiab/iiab-admin-console/pull/26

holta, 2017-09-19:
@tim-moody can you confirm "cd /opt/iiab/iiab-admin-console/; ./install" (which first installs /opt/admin/cmdsrv for ~4min and then /opt/admin/console for ~2min) provided the needed ~45 sec for CMDSRV to fully launch in the past, such that we never noticed this issue?

(Several smoke tests of the new load scripts in http://download.iiab.io/6.4/rpi seem to confirm this...)
```
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Success": "GET-KIWIX-CAT ."}
```

holta, 2017-09-19:
Success rate w/o sed and w/o "sleep 60" is 2 out of 3 RPi3 master installs so far &mdash; this error message just occurred:
```
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Error": "No Response from IIAB-CMDSRV"}
```

tim-moody, 2017-09-19:
This issue reports two unrelated messages:

 {"Error": "No Response from IIAB-CMDSRV"} means iiab-cmdsrv-ctl timed out waiting for a response even though the command may eventually complete. (it does not necessary mean cmdsrv is not running)

Error": "IIAB-CMDSRV failed to initialize","Alert": "True" which means what it says and we need to know why.

For the first:
the timeout is currently 10 seconds and should be longer.  also considering a timeout loop to prompt whether to continue.

For the second:
check for running and ready cmdsrv needs to be added to iiab-cmdsrv-ctl (WIP). this won't address the root cause, but will prevent commands being sent to cmdsrv when it is not ready

the time for cmdsrv to initialize is hardware dependent

I use factory's make-content-ready, which would probably benefit from having the kalite items done before getting the kiwix catalog to give cmdsrv time to initialize

holta, 2017-09-19:
https://github.com/iiab/iiab-admin-console/pull/27 looks like it may help.

holta, 2017-09-21:
Sufficiently under control as this very rarely now happens with load scripts @ http://download.iiab.io/6.4/rpi/ since 3 days ago &mdash; thanks all for having documented evidence + explanations in case a different version of this ever flares up later.

holta, 2017-09-21:
Addendum: @tim-moody suggests we might move "iiab-cmdsrv-ctl GET-KIWIX-CAT" to the very bottom of scripts like http://download.iiab.io/6.4/rpi/load.txt

closed 2017-09-21 by holta