iiab/iiab #264: "kalite manage retrievecontentpack download en" fails on new install (bad zip file?)

holta, 2017-09-18:
```
[INFO] [2017-09-18 12:09:43,738] kalite: Downloading content pack.
[INFO] [2017-09-18 12:09:43,889] kalite: Downloading content pack from http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
Traceback (most recent call last):
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py", line 224, in run_from_argv
    self.execute(*args, **options.__dict__)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/fle_utils/django_utils/command.py", line 141, in execute
    super(LocaleAwareCommand, self).execute(*args, **kwargs)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py", line 263, in execute
    output = self.handle(*args, **options)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/management/commands/retrievecontentpack.py", line 90, in handle
    self.download(*args, **options)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/management/commands/retrievecontentpack.py", line 147, in download
    zf = download_content_pack(f, lang)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/i18n/base.py", line 398, in download_content_pack
    zf = zipfile.ZipFile(fobj)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
BadZipfile: File is not a zip file
```

holta, 2017-09-18:
Running the same thing ~30 min latet generates a different error oddly:
```
root@raspberrypi:~# /usr/local/kalite/venv/bin/kalite manage retrievecontentpack download en
Traceback (most recent call last):
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py", line 224, in run_from_argv
    self.execute(*args, **options.__dict__)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/fle_utils/django_utils/command.py", line 141, in execute
    super(LocaleAwareCommand, self).execute(*args, **kwargs)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/management/base.py", line 263, in execute
    output = self.handle(*args, **options)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/management/commands/retrievecontentpack.py", line 73, in handle
    self.setup(options)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/updates/management/utils.py", line 55, in setup
    self.progress_log = UpdateProgressLog.get_active_log(process_name=self.process_name)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/updates/models.py", line 168, in get_active_log
    if logs.count() > 0:
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/models/query.py", line 372, in count
    return self.query.get_count(using=self.db)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/models/sql/query.py", line 423, in get_count
    number = obj.get_aggregation(using=using)[None]
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/models/sql/query.py", line 389, in get_aggregation
    result = query.get_compiler(using).execute_sql(SINGLE)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/models/sql/compiler.py", line 840, in execute_sql
    cursor.execute(sql, params)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/backends/sqlite3/base.py", line 366, in execute
    six.reraise(utils.DatabaseError, utils.DatabaseError(*tuple(e.args)), sys.exc_info()[2])
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/db/backends/sqlite3/base.py", line 362, in execute
    return Database.Cursor.execute(self, query, params)
DatabaseError: no such table: updates_updateprogresslog
```
FYI this RPi3 install created by 1-line-installer http://download.iiab.io/6.4/rpi/load-venvtest-vpn.txt

holta, 2017-09-18:
15 min ago, *all* http://box:8008 servers showed the following within Manage tab -> Language tab:
```
Connecting to the server. Please wait...
```
Now they all show this new error:
```
Could not connect to the central server; language packs cannot be downloaded at this time.
```

jvonau, 2017-09-19:
upstream website looks to be working now

holta, 2017-09-19:
Great News!

holta, 2017-09-22:
An ongoing problem for sure, as this has happened for substantial portions of the day over 2 of the last 4 days:

https://groups.google.com/d/msg/unleashkids/62T2aLx7F50/ONV2ROleAgAJ

Ref #257 ("kalite manage retrievecontentpack download en" fails nightly)

holta, 2017-09-22:
KA Lite's on-again-off-again hosting issues this week seem to be attributed to cloudflare.com

So (as discussed on https://webchat.freenode.net/?channels=#kalite) KA Lite has disabled their Cloudflare CDN for now (presumably offering direct access to https://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/... instead?)

As others have noticed e.g. https://community.learningequality.org/t/pantry-learningequality-org-down-502-response-from-cloud-flare/359