iiab/iiab #293: it's time to stop hosting WordPress at download.iiab.io

holta, 2017-09-20:
Even if WordPress had not released an important 4.8.2 security release yesterday, @tim-moody agrees with me it's now time for [roles/wordpress/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/wordpress/defaults/main.yml) to use generic/dynamic URL https://wordpress.org/latest.tar.gz to install directly from WordPress.

Instead of a trail of stale/insecure installers we're caching at http://download.iiab.io/packages that wastes our time for no legit reason (subtracting value far more that adding value...)

Even if WordPress did not auto-upgrade on install (and on an ongoing basis now WordPress now auto-upgrades, with manual intervention removed, and for good reason given sloppy sysadmin practices are a dangerous de facto norm...)

In any case: WordPress (like NextCloud) provides a higher quality QA & hosting uptime than we do, so _humility on our part_ (others are far better at tracking/securing/QA'ing WordPress than we are) means transition to https://wordpress.org/latest.tar.gz now (walking away from hosting WordPress @ download.iiab.io) is the way forward, following in @floydianslips footsteps with Nextcloud's generic installer URL https://download.nextcloud.com/server/releases/latest-12.tar.bz2

Likewise in future we might even consider: https://download.nextcloud.com/server/releases/latest.tar.bz2

_PS let's monitor that offline instances of WordPress continue to work (and self-upgrade on their own, as is now mandatory when they briefly find Internet access...)  Of course there are integration testing/blowback risks, but these are more than worth it &mdash; everybody knows WordPress security is a PHP treadmill &mdash; so if we choose to offer the world's most popular blogging platform, we too like the rest of the world offline+online world we need to step up the plate and be responsive global stewards and citizens._

holta, 2017-09-20:
FYI I very carefully checked for the possibility for WordPress 4.8.2 on just 48-72hours ago(*) but security issues were likely embargoed (kept hush-hush, to contain exploits as much as possible, prior to WordPress servers upgrading themselves worldwide...)

(*) prior and while working on #276 (WordPress 4.8.1 instead of 4.8)

holta, 2017-09-21:
Functional testing of http://box/wordpress also completed.

jvonau, 2017-09-25:
See comment in #296

holta, 2017-09-25:
Re-opening as @jvonau discovered an offline bug at the bottom of #296.

closed 2017-09-26 by holta