iiab/iiab #298: "apt install ansible" smoke tests but is it sufficient?

holta, 2017-09-21:
Earlier @jvonau had suggested:
```
apt -y install ansible python-kerberos python-selinux python-winrm python-xmltodict sshpass bzip2 file findutils gzip tar unzip zip python-keyczar python-boto python-dnspython python-pyrax python-sphere
```
This may be overkill.

FWIW "apt -y install ansible" seems to suffice/survive a full smoke test of http://download.iiab.io/6.4/rpi/load-vpn.txt _with no apparent differences_ (except that it appears faster!)

holta, 2017-09-21:
Functional tests of all the "basic dozen" http://box/apps (per http://FAQ.IIAB.IO question 3) pass with flying colors!

(Beginning with "apt -y ansible" alone on Raspbian, without additional apt installations, leaving us with Ansible 2.2.1...per http://download.iiab.io/6.4/rpi/load.txt)

ASIDE: Ansible 2.4.0 was released yesterday: https://releases.ansible.com/ansible/ (but Raspbian-and-likely-Debian are holding back with 2.2.1 for now)

tim-moody, 2017-09-21:
where did box/apps come from or do you mean http://box/<apps>>

holta, 2017-09-21:
> Functional tests of all the "basic dozen" http://box/apps (per http://FAQ.IIAB.IO question 3) pass with flying colors!

I mean http://box/wordpress, http://box/munin, http://box/maps etc per the listing above.  Perhaps `http://box/[EACH OF ~12 APPS]` is more clear?

holta, 2017-09-21:
"apt install ansible" seems to work really great on Raspbian.

closed 2017-09-21 by holta

holta, 2017-09-21:
Reminder that Ansible 2.2.1 needs a 1-line patch, as documented in all the load scripts @ http://download.iiab.io/6.4/rpi :
```
sed -i 's/StrictVersion/LooseVersion/g' /usr/lib/python2.7/dist-packages/ansible/modules/core/web_infrastructure/htpasswd.py
# ABOVE APPLIES TEMPORARY PATCH https://github.com/ansible/ansible/pull/20202
# needed by munin until Ansible 2.3.2 or 2.4.0, to overcome Ansible bug:
#    TASK [munin : Create admin user] ********************************
#    fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true,
#    "msg": "invalid version number '1.7.0.post20170423015547'"}
```