iiab/iiab #313: Fedora 25: "iiab-make-kiwix-lib" "mysql" command not found

arky, 2017-09-26:
Automated Fedora 25 builds report that 'iiab-make-kiwix-lib' and 'mysql' commands are missing on the machine. 

> ==> fedora: samples/sampler.html
> ==> fedora: samples/test.html
> ==> fedora: samples/test.json
> ==> fedora: sent 67,167 bytes  received 419 bytes  135,172.00 bytes/sec
> ==> fedora: total size is 65,668  speedup is 0.97
> ==> fedora: ./cp-menus: line 31: iiab-make-kiwix-lib: command not found
> ==> fedora: cp: 
> ==> fedora: cannot create regular file '/var/www/html/home/index.html'
> ==> fedora: : No such file or directory
> ==> fedora: files copied
> ==> fedora: ./cp-menus: line 61: mysql: command not found
> ==> fedora: ./cp-menus: line 64: mysql: command not found
> ==> fedora: ./cp-menus: line 68: mysql: command not found
> ==> fedora: database created

Complete output is here https://pastebin.com/PYdNYXdV

jvonau, 2017-09-26:
Tim & Geroge have backed away from Fedora in general since F23 but I wish to bring the support back into focus as ansible is a redhat product. I've aware of some differences between the last successful Fedora installs and what is available in Fedora today. One issue is the renaming of some python/mysql/zmq packages in later Fedora's but the playbooks do not reflect the latest names as the focus has not been of the redhat side of the code.

holta, 2017-11-21:
@jvonau can I defer this to IIAB 6.6?

jvonau, 2017-11-21:
Yes, I'll look at fedora at some point in the future.

holta, 2018-07-18:
@arky please update or close your tkt, if poss?

Thanks, as [IIAB 6.6](http://download.iiab.io/6.6/) release is now converging!

We can also move this to our [IIAB 6.7 Milestone](https://github.com/iiab/iiab/milestone/4) if you prefer.

arky, 2018-07-18:
@holta Looks like I need setup the CI system again to check if issue still exists. Please assign this to me.

holta, 2018-11-04:
> @holta Looks like I need setup the CI system again to check if issue still exists. Please assign this to me.

Oops, sorry @arky I missed this in July!

holta, 2019-05-04:
@arky plz re-open as nec, Thanks!

closed 2019-05-04 by holta