iiab/iiab #314: "Setting ifcfg-WAN True" fails on CentOS

holta, 2017-09-26:
Thanks @arky for discovering this on your Jenkins server:

==> centos: TASK [1-prep : Checking for ifcfg-WAN file - Can Fail] *************************
==> centos: skipping: [127.0.0.1]
==> centos:
==> centos: TASK [1-prep : Setting ifcfg-WAN True] *****************************************
==> centos: fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "The conditional check 'has_ifcfg_WAN.stat.exists' failed. The error was: error while evaluating conditional (has_ifcfg_WAN.stat.exists): 'dict object' has no attribute 'stat'\n\nThe error appears to have been in '/opt/iiab/iiab/roles/1-prep/tasks/detected_redhat.yml': line 6, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Setting ifcfg-WAN True\n ^ here\n"}
==> centos: to retry, use: --limit @/opt/iiab/iiab/iiab.retry
==> centos:
==> centos: PLAY RECAP *********************************************************************
==> centos: 127.0.0.1 : ok=36 changed=7 unreachable=0 failed=1

arky, 2017-09-26:
After few hours and a full meal I further investigated this issue. Looks like this problem is triggered by the fact that the virtual machines does not have a network interfaces named 'wlan*', 'eth*'.  

I couldn't the vagrant to create these interfaces properly. Perhaps it is a limitation of vagrant networking stack.

jvonau, 2017-09-26:
Need state which of the OSs this error applies to, and the actual log would be helpful.

jvonau, 2017-09-26:
I have been playing with our jenkins install hosted by George, some processes need to converge, but forward progress should be pretty rapid.

holta, 2017-09-30:
@arky @jvonau is this important for IIAB 6.4?

Or can we defer to October's IIAB 6.5 not so many days/hours away?

georgejhunt, 2017-09-30:
fixe by pr334

closed 2017-09-30 by georgejhunt