iiab/iiab #315: "Find wifi gateway config if present - Can Fail" fails on Debian

holta, 2017-09-26:
Thanks @arky for discovering this on your Jenkins server:

==> debian: TASK [1-prep : Find wifi gateway config if present - Can Fail] *****************
==> debian: skipping: [127.0.0.1]

==> debian:
==> debian: TASK [1-prep : Setting has_wifi_gw based on ESSID if found - Can Fail] *********
==> debian: fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout_lines'"}
==> debian: to retry, use: --limit @/opt/iiab/iiab/iiab.retry
==> debian:
==> debian: PLAY RECAP *********************************************************************
==> debian: 127.0.0.1 : ok=37 changed=17 unreachable=0 failed=1

Originally posted to #237.

arky, 2017-09-26:
After few hours and a full meal I further investigated this issue. Looks like this problem is triggered by the fact that the virtual machines does not have a network interfaces named 'wlan*', 'eth*'.  

I couldn't the vagrant to create these interfaces properly. Perhaps it is a limitation of vagrant networking stack.

jvonau, 2017-09-26:
I've had a good test run of a more advanced testing branch with George's install of jenkins. In that instance George has the machines coming up with the usual ethX naming, think right now it might be a subtle differences between the setup of vagrant.

jvonau, 2017-09-26:
If the VM is coming up with "predictable interface names" the code should be able to handle that...  PR #319 should fix CentOS again.

jvonau, 2017-09-26:
The testing on jenkins is invalid as 2.2.0 is installed and still lacks the need patch for htaccess #249 
> ok I see what is going on.
==> debian: Cloning into 'iiab'...
==> debian: + cd /opt/iiab/iiab
==> debian: + ./scripts/ansible
ansible 2.2.0 is being installed still while the patch is for 2.2.1.

georgejhunt, 2017-09-26:
I think the curl-me has pull from ditribution AND sed rewrite for htpasswd
in the jv branch which your test uses.

jvonau, 2017-09-26:
Not everybody is going to use the web-load scripts when crafting a custom install. I'd prefer to have the mod in one place to reduce the future upkeep for when it is unneeded, such as when debian might release a later version of ansible. It's in one place to edit rather than update that wiki, update the web-load scripts and elsewhere.

jvonau, 2017-09-27:
Have Tim's original curl-me in mix in jenkens is skewing the results as scripts/ansiable is being used while George's has been updated.
https://github.com/arky/iiab-smoketest/blob/master/curl-me 
https://github.com/georgejhunt/iiab-smoketest/blob/master/curl-me

arky, 2017-09-27:
@jvonau  @georgejhunt  Can you please push the correct curl-me script my repo. Thanks!

arky, 2017-09-27:
@georgejhunt You have to change the git URL in the Jenkin job  to your repo.  

https://github.com/georgejhunt/iiab-smoketest

holta, 2017-09-30:
@arky / All: please resolve or clarify how this offline-testing condition should be addressed in October for IIAB 6.5 starting in not so many days/hours?  Thanks!

jvonau, 2017-11-07:
Think this is resolved? #490

jvonau, 2017-11-12:
invalid

holta, 2017-11-12:
@arky & all please close this now if it's not longer an issue, Thx!

holta, 2017-11-21:
Closing due to non-response.

Please re-open if this re-occurs!

closed 2017-11-21 by holta