iiab/iiab #317: Install OpenVPN/TeamViewer/support tool(s) BEFORE installs begin

holta, 2017-09-26:
[download.iiab.io/6.4/rpi/load-vpn.txt](http://download.iiab.io/6.4/rpi/load-vpn.txt) is on the right track but _not quite enough_ for classic hackers who demand ssh/http/etc and refuse to use TeamViewer, for whatever reason.

So we need a much better way of installing OpenVPN/XSCENET (or [remot3.it](http://remot3.it) formerly known as Weaved) _immediately_ at the beginning of IIAB installs (using #225 or similar).

**This will allow mentoring/implementation/learning relationships to instantly and rapidly blossom around digital libraries like IIAB, in ways that have been sorely lacking to date.**

SEE http://FAQ.IIAB.IO #22 ("How can I remotely manage my Internet-in-a-Box?")

PS remote support tools are of course *not* mandatory for all sorts of security/privacy reasons and others!  i.e. currently those who do not want remote support tools use an installer like: [download.iiab.io/6.4/rpi/load.txt](http://download.iiab.io/6.4/rpi/load.txt)

jvonau, 2017-11-12:
close? install-support can install the vpn standalone

holta, 2017-11-12:
Let's leave this open until install-support is better documented or better integrated with ./iiab-install or wherever we want to take this &mdash; all of which can be deferred to IIAB 6.6 or 7.0!

holta, 2017-11-21:
> Let's leave this open until ./install-support is better documented or better integrated with ./iiab-install or wherever we want to take this — all of which can be deferred to IIAB 6.6 or 7.0!

Deferring to [IIAB 6.6](https://github.com/iiab/iiab/milestone/3).

holta, 2018-09-20:
OpenVPN was moved to Stage 1 (near the beginning) of IIAB's install process.

Thanks to #1141 and PR's #1148 #1151 #1155.

So perhaps we can do _even better_ in future years, but let's declare victory after a year now, with OpenVPN now significantly overhauled with _major_ improvements over the past ~6 weeks.

closed 2018-09-20 by holta