iiab/iiab #318: XSCENET / OpenVPN / TeamViewer / remot3.it redesign

holta, 2017-09-26:
Per @georgejhunt's request, we need new remote support infrastructure, as XSCENET is showing its age (and its risks).

Naturally there are many competing (in fact conflicting!) requirements.  So let's start by gathering some of the more common/dominant possible requirements for this new remote collaboration/support infra:

- works BEFORE Internet-in-a-Box install begins, so screensharing/mentoring is legit: #317
- secures volunteers homes strictly and/or incrementally (security is like an onion…)
- takes the pain out of remote support allowing “Almost All Kinds Of Testing” WHEN escalated/trusted parties so choose
- highly available and highly reliable (self-recovers, holding connection for days/weeks/months) such that it also can (when you're lucky!) work over fragile 3G connections to the developing world

georgejhunt, 2017-09-26:
I think:

   1. Openvpn communication between openvpn clients should be turned off.
   2. Access to openvpn connected clients should be via a second openvpn
   instance on our server which is NATed/masqueraded to the client network
   3. The iptables rules between the two networks will permit conversation
   that originates from the control network--
   4. But blocks access which tries to originate from the client network.
   5. Access to the control openvpn network only via individually generated
   shared keys (and can be revoked on a person by person basis -- this could
   become a headache if many keys were involved)
   6. So clients would be directed toward teamviewer -- which is has shared
   screen, and is probably better for novice -- No support/liability issues.

holta, 2017-09-26:
Thanks @georgejhunt even if I don't understand it all just yet :)

Before finalizing engineering details I'd like to keep the conversation at a higher level addressing human needs, and not just of implementers/ops but that's a good place to start.

To that end, I've refined 6 possible requirements + 2 possible "would-be-nice-to-haves" at the top of this ticket, that I'd prefer we all talk thru before _at the human/collaboration level_ before getting locked into everyone's particular/favorite technologies~

holta, 2019-02-01:
@m-anish points to https://ngrok.com which is another great option, for quick diagnosis of machines behind firewalls/NAT etc.

holta, 2019-02-02:
@tim-moody mentions https://serveo.net which is similar to the above ("Expose local servers to the internet. No installation, no signup.")

Dynamic DNS services like No-IP (https://www.noip.com) are also possible if a higher uptime / enterprise solution is needed.

holta, 2019-02-22:
@m-anish asks about **Mosh** which does much better than ssh with remote connectivity:
> do you think something like this might be useful to have as a package on iiab? https://mosh.org
> [it's] just a simple sudo apt install -y mosh

Should this be added to http://FAQ.IIAB.IO #29 itself?  Specifically:
> "How can I remotely manage my Internet-in-a-Box?"
> http://wiki.laptop.org/go/IIAB/FAQ#How_can_I_remotely_manage_my_Internet-in-a-Box.3F

m-anish, 2019-02-22:
To be clear.. this would be as simple as `sudo apt install -y mosh`

tim-moody, 2019-02-22:
true as long as your laptop runs linux

m-anish, 2019-02-23:
> true as long as your laptop runs linux

true! but i meant on the iiab, it is as easy as that command.

ericnitschke, 2019-11-02:
TeamViewer is a hog, even on the 4B 4GB version, and they're increasingly cutting people off for "suspected commercial use". 

I tried the built-in RealVNC Connect (for connecting on local network), but I'm unable to reach the Pi after an IIAB install.  Are connections blocked somewhere?

I tested Anydesk (https://anydesk.com/en/downloads/raspberry-pi) which seems to be the go-to TeamViewer alternative, also to no avail.  There's an issue with Anydesk on the Pi which makes it crash.  Anydesk on PC sees the Pi, but can't connect.

holta, 2019-11-02:
> TeamViewer is a hog, even on the 4B 4GB version, and they're increasingly cutting people off for "suspected commercial use".

True, but it's also a lifesaver during informal testing of Kids On Computers' client OS based on Raspbian With Desktop, and FWIW "TeamViewer Host" (incoming connections only) has worked well even on minimal RPi 3's.

> I tried the built-in RealVNC Connect (for connecting on local network), but I'm unable to reach the Pi after an IIAB install. Are connections blocked somewhere?

Have you reviewed Firewall needs & settings e.g. https://github.com/iiab/iiab/wiki/IIAB-Networking#firewall-iptables ?

If you nail down a specific firewall need, @jvonau and others can help you solve it, e.g. review the ports here too: https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-ports--services

> I tested Anydesk (https://anydesk.com/en/downloads/raspberry-pi) which seems to be the go-to TeamViewer alternative, also to no avail. There's an issue with Anydesk on the Pi which makes it crash. Anydesk on PC sees the Pi, but can't connect.

Thanks for the insights (I did not know about Anydesk at all).

ericnitschke, 2019-11-02:
Good to know you've had more luck with it! It seems to freeze up two of the Pi's we have in the field (3B+), requiring weekly manual restarts.  In one case I was so fed up I used a wifi-enabled power plug to restart the Pi from afar.

> Have you reviewed Firewall needs & settings e.g. https://github.com/iiab/iiab/wiki/IIAB-Networking#firewall-iptables ?

Didn't know this page existed. *facepalm*. Might be able to add some hacks for adding IIAB to existing community networks.

> If you nail down a specific firewall need, @jvonau and others can help you solve it, e.g. review the ports here too: https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-ports--services

Vielen Dank.

georgejhunt, 2019-11-02:
> TeamViewer is a hog, even on the 4B 4GB version, and they're increasingly
> cutting people off for "suspected commercial use".
>
> True, but it's also a lifesaver during informal testing of Kids On
> Computers' client OS based on Raspbian With Desktop, and FWIW "TeamViewer
> Host" (incoming connections only) has worked well even on minimal RPi 3's.
>
> I tried the built-in RealVNC Connect (for connecting on local network),
> but I'm unable to reach the Pi after an IIAB install. Are connections
> blocked somewhere?
>
> If you are just wanting temporary remote access, you can run "iptables -F"
on the rpi (this flushes the firewall).  Permanent remote access via VNC
requires that you open port 5900 or 5901 using lines similar to
https://github.com/iiab/iiab/blob/5fc4d220578c74f8506592787d79f479de4b3f93/roles/network/templates/gateway/iiab-gen-iptables#L147

ericnitschke, 2019-11-03:
Thanks @georgejhunt !! I ran "iptables -F" and VNC is blazing fast. Cheers.

holta, 2019-11-03:
> Thanks @georgejhunt !! I ran "iptables -F" and VNC is blazing fast. Cheers.

@ericnitschke:

1) Did you try `ports_externally_visible: 5` per the instructions at https://github.com/iiab/iiab/wiki/IIAB-Networking#firewall-iptables ?

2) Should TCP ports 5900 and 5901 be added to `ports_externally_visible: 4` (or possibly even `ports_externally_visible: 3` out of the box??) for VNC implementers in future?

3) I presume we should leave RDP's UDP+TCP port 3389 off for now?

ericnitschke, 2019-11-05:
> 
> 
> > Thanks @georgejhunt !! I ran "iptables -F" and VNC is blazing fast. Cheers.
> 
> @ericnitschke:
> 
>     1. Did you try `ports_externally_visible: 5` per the instructions at https://github.com/iiab/iiab/wiki/IIAB-Networking#firewall-iptables ?

Yes, it worked fine.  No issues after restart.

>     2. Should TCP ports 5900 and 5901 be added to `ports_externally_visible: 4` (or possibly even `ports_externally_visible: 3` out of the box??) for VNC implementers in future?

I think this would be useful - perhaps with a section [here ](http://wiki.laptop.org/go/IIAB/FAQ#How_can_I_remotely_manage_my_Internet-in-a-Box.3F) on how to install/use VNC.  Since it runs on local network it will be much faster for those who want to use the desktop interface but can't use an external monitor.

Only issue is security.  I don't remember if a password is required to access the Pi.  Can someone [install VNC](https://www.raspberrypi.org/documentation/remote-access/vnc/) and check before those ports are opened by default?

holta, 2019-11-05:
Thanks @ericnitschke for helping us decide if 'ports_externally_visible: 3' (default) or 'ports_externally_visible: 4' is better to open up ports 5900 & 5901 for VNC.  If we decide to go with '4', then yep let's document this in http://faq.iiab.io Item 31 ("How can I remotely manage my Internet-in-a-Box?")

Question: is https://www.raspberrypi.org/documentation/remote-access/vnc/ really up-to-date now that VNC server is integrated within Raspbian (Lite & Desktop I presume) in recent years?  As can "instantly" be enabled using `/usr/bin/raspi-config` -> "5 Interfacing Options" -> "P3 VNC - Enable/Disable graphical remote access to your Pi using RealVNC".  _(Perhaps enabling Raspbian built-in VNC server is also possible using Raspbian With Desktop's graphical "Start" menu in the top-left?)_

ericnitschke, 2019-11-06:
>  _(Perhaps enabling Raspbian's built-in RealVNC server is also possible using Raspbian With Desktop's graphical "Start" menu in the top-left?)_

[image]

holta, 2019-11-06:
VNC logins do require a password generally, but still I propose we go ahead with this in coming days:

> `ports_externally_visible: 4` (safer?) is better to open up ports 5900 & 5901 for VNC.

And simultaneously...

> document this in http://FAQ.IIAB.IO Item 31 ("How can I remotely manage my Internet-in-a-Box?")

tim-moody, 2019-11-06:
is desktop required?

closed 2025-01-27 by holta