iiab/iiab #321: GET-KIWIX-CAT "No Response from IIAB-CMDSRV" says "Hit Enter" but jumps to next task

holta, 2017-09-26:
This is likely a very different GET-KIWIX-CAT issue from #261 likely &mdash; @tim-moody is looking into it:
```
...
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.023170948 seconds
Hit Enter to wait another 60 seconds (anything else to stop): Generating a sharing network.
Generating a sharing network invitation--from me, to me!
Done!
[INFO] [2017-09-26 16:04:51,833] kalite: Downloading content pack.
...
```
Summary: "Hit Enter..." appeared but immediately (within about 1 second) it proceeded with KA Lite tasks, without giving me a chance to hit a key.

Context: http://download.iiab.io/6.4/rpi/load-big-vpn.txt was running a full (big) install onto an RPi3.

tim-moody, 2017-09-26:
I am unable to reproduce the behavior in which iiab-cmdsrv-ctl prompts for input and does not wait to get it. On your rpi I issue iiab-cmdsrv-ctl GET-KIWIX-CAT. It returns with a prompt that there was no response in 60 seconds and issues the above "Hit Enter ...". I waited several minutes and it did not continue until I pressed enter.

jvonau, 2017-09-26:
I'd log the job start time in he same line as "No Response from IIAB-CMDSRV" to provide a visual clue.

holta, 2017-09-26:
This issue does not occur at the command line, and is messier than I understood earlier.  It skips past the next command in the bash script (whatever command follows "iiab-cmdsrv-ctl GET-KIWIX-CAT").

Both these issues occurs when "iiab-cmdsrv-ctl GET-KIWIX-CAT" is called from scripts like http://download.iiab.io/6.4/rpi/load-big-vpn.txt using "bash < load-big-vpn.txt" as follows:
```
root@box:~# bash < test1.txt
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.0592889786 seconds
Hit Enter to wait another 60 seconds (anything else to stop): Traceback (most recent call last):
  File "/usr/bin/iiab-cmdsrv-ctl", line 62, in <module>
    resp = raw_input('Hit Enter to wait another ' + str(REQUEST_TIMEOUT/1000) + ' seconds (anything else to stop): ')
EOFError: EOF when reading a line
root@box:~#
```
The above output resulted from test1.txt not having any commands after "iiab-cmdsrv-ctl GET-KIWIX-CAT" in this 2-line script here:
```
#!/bin/bash
iiab-cmdsrv-ctl GET-KIWIX-CAT
```
Here is further confirmation that "iiab-cmdsrv-ctl GET-KIWIX-CAT" is skipping over the following line in the bash script.  As demonstrated by test2.txt here:
```
#!/bin/bash
iiab-cmdsrv-ctl GET-KIWIX-CAT
echo DONE1
echo DONE2
```
Which produces this output:
```
root@box:~# bash < test2.txt
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.0628631115 seconds
Hit Enter to wait another 60 seconds (anything else to stop): DONE2
root@box:~#
```
OBSERVATION 1: the command "echo DONE1" is never executed, while the commands after that (e.g. "echo DONE2") are executed.

OBSERVATION 2: the initial issue goes away (you can now press Enter) if running "./test1.sh" and "./test2.sh" instead of "bash < test1.txt" and "cat < test2.txt"

OBSERVATION 3: if you do hit Enter when running "./test1.sh" or "./test2.sh" it will fails after the SECOND MINUTE as follows:
```
root@box:~# ./test1.sh
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.0545508862 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
Received reply:  {"Error": "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."}
```

holta, 2017-09-27:
RPi3 behavior (using [IIAB 6.4 RC5](https://github.com/iiab/iiab/releases/)) is a bit different this morning on the same machine:
```
root@box:/library/zims# iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.083137989 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
No Response from IIAB-CMDSRV in 123.043766022 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
Received reply:  {"Error": "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."}
```

holta, 2017-09-27:
Ref PR https://github.com/iiab/iiab-admin-console/pull/32 (Changes to get kiwix catalog)

holta, 2017-09-30:
@tim-moody can we resolve this now?

tim-moody, 2017-09-30:
resolve as unable to duplicate

holta, 2017-09-30:
It's straightforward to duplicate @ https://github.com/iiab/iiab/issues/321#issuecomment-332365873 but we can work around this install approach.  Resolving.

closed 2017-09-30 by holta