iiab/iiab #322: "Setting has_wifi_gw based on ESSID if found - Can Fail" fails on Ubuntu 16.04 LTS

holta, 2017-09-26:
Thanks to @germanrs.  ./runansible fails as follows:
```
TASK [1-prep : Setting has_wifi_gw based on ESSID if found - Can Fail] *********
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout_lines'"}
```
This closely resembled #315 ("Find wifi gateway config if present - Can Fail" fails on Debian)

Context:
```
HW: http://www.hystou.com/products/fanless-mini-pc/4th-gen-core-i7-mini-pc/intel-haswell-core-i7-4500u-cpu-fanless-mini-pc-computer-barebone-pc-1953.html

root@ubuntu:~# ifconfig -a
enp2s0    Link encap:Ethernet  HWaddr 00:e0:4c:68:63:2f
          inet addr:10.0.0.7  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:632f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1244 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1072 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:919853 (919.8 KB)  TX bytes:137768 (137.7 KB)

enp3s0    Link encap:Ethernet  HWaddr 00:e0:4c:68:63:30
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:508 (508.0 B)  TX bytes:508 (508.0 B)

wlp4s0b1  Link encap:Ethernet  HWaddr 2c:b0:5d:cf:46:fa
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

jvonau, 2017-09-27:
> "ansible_local": {
   "local_facts": {
    "ansble_version": "2.2.0.0",

Not using the latest web-load. #319

germanrs, 2017-09-27:
Ansible was installed from /opt/iiab/iiab/scripts/ansible, I did run pip uninstall ansible and apt install ansible, now I'm running  2.0.0.2:

root@ubuntu:~:# ansible --version
ansible 2.0.0.2

Then I applied George's fix https://github.com/iiab/iiab/pull/323/files

And ./runansible finished the installation.

holta, 2017-09-29:
@germanrs is this sufficiently resolved?

Please re-open if not!

closed 2017-09-29 by holta