iiab/iiab #324: debian:  [WARNING]: Consider using file module with owner rather than running chown

arky, 2017-09-27:
Harmless warning in to file module in Kiwix task. Report from nightly Jenkins CI run. Please kindly close the bug if it is low priority.  

==> debian: TASK [kiwix : Set kiwix ownership] *********************************************
==> debian: changed: [127.0.0.1]
==> debian:  [WARNING]: Consider using file module with owner rather than running chown
==> debian: 
==> debian: TASK [kiwix : Make an entry in crontab to restart every hour] ******************
==> debian: changed: [127.0.0.1]
==> debian: 
==> debian: TASK [kiwix : Make an entry in crontab to restart every hour] ******************
==> debian: skipping: [127.0.0.1]
==> debian: 
==> debian: TASK [kiwix : Create kiwix-serve service] **************************************
==> debian: changed: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/kiwix-serve.service', u'src': u'kiwix-serve.service.j2', u'mode': u'0644'})
==> debian: changed: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-kiwix-lib', u'src': u'iiab-make-kiwix-lib', u'mode': u'0755'})
==> debian: changed: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-kiwix-lib.py', u'src': u'iiab-make-kiwix-lib.py', u'mode': u'0755'})
==> debian: changed: [127.0.0.1] => (item={u'dest': u'/usr/bin/iiab-make-apache-config.py', u'src': u'iiab-make-apache-config.py', u'mode': u'0755'})

holta, 2017-09-27:
@tim-moody can you explain and/or resolve?

jvonau, 2017-11-12:
code merged

closed 2017-11-12 by holta