iiab/iiab #328: "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."

holta, 2017-09-27:
Error occurs frequently when running "iiab-cmdsrv-ctl GET-KIWIX-CAT" on [IIAB 6.4 RC5](https://github.com/iiab/iiab/releases/) on RPi3, despite a very high-speed Internet connection: (on Raspbian Stretch Lite)
```
root@box:~# iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.083137989 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
No Response from IIAB-CMDSRV in 123.043766022 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
Received reply:  {"Error": "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."}

root@box:~# iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
No Response from IIAB-CMDSRV in 60.027826786 seconds
Hit Enter to wait another 60 seconds (anything else to stop):
Received reply:  {"Error": "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."}
```
Occasionally however it works:
```
root@box:/library/zims# iiab-cmdsrv-ctl GET-KIWIX-CAT
message to send:  GET-KIWIX-CAT
Connecting to server...
Received reply:  {"Success": "GET-KIWIX-CAT ."}
```
@tim-moody is they some reason the downloading of library.xml (appears) to be far glitchier than it ever was in the past, despite an extremely high-speed connection?  Either due to the way we're downloading it or an upstream/hosting issue?

Certainly it feels intermittent, as if the symptom comes in waves (possibly much more frequent at certain times, and much less frequent at certain times...)

Ref #321 (GET-KIWIX-CAT "No Response from IIAB-CMDSRV" says "Hit Enter" but jumps to next task)

jvonau, 2017-09-27:
> {"Error": "GET-KIWIX-CAT ERROR - [Errno 99] Cannot assign requested address."}

Our setting of the hostname is a bit less that ideal on the debian family, the change does not take effect until rebooted, is one of the things I noted.  If you were to omit the grabbing the catalogue from the load script, reboot the box first, then try the command does the situation improve?

holta, 2017-09-27:
>  If you were to omit the grabbing the catalogue from the load script, reboot the box first, then try the command does the situation improve?

Unfortunately this does not help!  (The same problem happens repeatedly after reboot.)

tim-moody, 2017-09-27:
I think the context is xmlsrc = urllib2.urlopen(kiwixLibUrl)

https://stackoverflow.com/questions/11981933/python-urllib2-cannot-assign-requested-address

seems to happen when kiwix access times out

after iiab-admin-console#32 use iiab-get-kiwix-cat utility wrapper as a work around

holta, 2017-09-30:
@tim-moody can we resolve this now?

Or are there still concerns it might come back to bite us stalling the installation of Admin Console when Internet access lags and other such situations?

tim-moody, 2017-09-30:
This problem is likely still there for iiab-cmdsrv-ctl but not for iiab-get-kiwix-cat.

holta, 2017-09-30:
Am deferring this from 6.4 to 6.5/October as a placeholder, in case it merits discussion then.

holta, 2017-10-25:
> This problem is likely still there for iiab-cmdsrv-ctl but not for iiab-get-kiwix-cat.

@tim-moody can we resolve this?

(Or is there action needed for IIAB 6.5 possibly?)

holta, 2017-11-06:
Closed pre @tim-moody recommendation.

closed 2017-11-06 by holta