iiab/iiab #329: http://box/usb "teacher content" does not appear from USB sticks

holta, 2017-09-28:
While USB sticks automount properly to /media/usb1, /media/usb2 etc &mdash; their content (from USB stick directories named "usb") never appears in browsers at http://box/usb &mdash; nor as a link within /library/www/html/local_content

This failure occurs with Ansible 2.2.1 and Ansible 2.4.0 on Raspbian Stretch Lite today.

*Oddly it worked ~48h ago: https://github.com/iiab/iiab/issues/195#issuecomment-332006129 !*

More @ https://github.com/iiab/iiab/pull/316#issuecomment-332689802

holta, 2017-09-28:
Ref #195 Upgrade usbmount 0.0.14 to 0.0.22...for Debian & Raspbian especially (others too?)
Ref #283 Udev usbmount

georgejhunt, 2017-09-28:
starting with a clean sept 7 lite image, 6.4-RC6, using downoad.iiab.io/6.4/rpi/load.txt -> bash (rpi3 hardware):
It works for me. 
So Adam we need to figure out how you got a failure. What was different.

holta, 2017-09-28:
All 3 RPi3's with failing http://box/usb are online for diagnosis here if nec for those who have XSCENET access:

('10.8.0.14', 'Raspbian-Lite-2017-09-07-IIAB-6.4-RC5-big')
('10.8.0.18', 'Raspbian-Lite-2017-09-07-IIAB-6.4-ansible2.4.0-big')
('10.8.0.54', 'Raspbian-Lite-2017-09-07-IIAB-6.4-RC6+')

holta, 2017-09-28:
> ('10.8.0.14', 'Raspbian-Lite-2017-09-07-IIAB-6.4-RC5-big')
('10.8.0.18', 'Raspbian-Lite-2017-09-07-IIAB-6.4-ansible2.4.0-big')
('10.8.0.54', 'Raspbian-Lite-2017-09-07-IIAB-6.4-RC6+'

All 3 above have a proper USB stick inserted,  i.e. including content on stick, within a directory called "USB"

georgejhunt, 2017-09-28:
What is the history of these failing machines?  Did you install from
scratch today after 6.4-RC6? or are these just git pull on current master
and runansible?

holta, 2017-09-28:
All 3 machines were installed from scratch, as per their names but expanded a bit here:

* RC5+ big installed w/ Ansible 2.2.1
* RC5+ big installed w/ Ansible 2.4.0
* RC6 medium installed w/ Ansible 2.4.0

There was never a "git pull" on these 3 machines FWIW!

georgejhunt, 2017-09-28:
Let's try to get to something that might be similar to what I have on my
desk.

on 10.8.0.14 do a git pull, ./runansible --debug, and reboot, then remove
the usb, and reinsert it

holta, 2017-09-28:
10.8.0.14 is my reference testing machine to compare Ansible 2.2.1 so I cannot do a "git pull" there.

But I can do it on any other machine that helps?

georgejhunt, 2017-09-28:
You pick the machine

holta, 2017-09-29:
@georgejhunt there is no improvement with http://box/usb on 10.8.0.54 (RC6, "git pull" doesn't accomplish anything material) even after reboot & removal / re-insertion of USB sticks multiple times.

I tried it both ways, running "umount" prior to pulling the USB stick, and also without running "umount"

As usual the USB stick is correctly mounted to /media/usb0 or /media/usb1, but never appears within /library/www/html/local_content

georgejhunt, 2017-09-29:
Adam,
I cannot find any obvious problems with your 3 machines.

Would you please start over, write a raspbian lite image, and load min
local_vars -- to verify that current master works.

Then we can load medium, and big local_vars to see if we are getting some
sort of conflict.

holta, 2017-09-29:
@georgejhunt thanks!

Load Script http://download.iiab.io/6.4/rpi/load-min-vpn.txt is running and ./runansible begun, on the RPi3 (with USB stick inserted, including Teacher Content) that used to be called:
```
root@box:~# more /etc/iiab/handle
Raspbian-Lite-2017-09-07-IIAB-6.4-ansible2.4.0-big
```
Try 10.8.0.18 (or whatev it becomes) which should be live within ~20min, then it should auto-reboot upon completion of the above Load Script within about 45min.

holta, 2017-09-29:
@georgejhunt here's the new OpenVPN name, that should be live in ~15min:
```
root@raspberrypi:~# more /etc/iiab/handle
Raspbian-Lite-2017-09-07-IIAB-6.4-RC6+-min
```

holta, 2017-09-29:
@georgejhunt the clean install of load-min-vpn.txt completed, and auto-rebooted.

There is nothing in /library/www/html/local_content after the 1st reboot, despite the fact /dev/sda1 correctly auto-mounts to /media/usb1

Removing the USB stick and re-inserting it to different USB ports does not help, with or without a "umount" command prior to yanking, AH WELL.  It automounts as /media/usb0 or /media/usb1 every time, but is not available for teachers/students to view their USB content.

Check it out @ 10.8.0.26 if you can?

georgejhunt, 2017-09-29:
the test machine you made for me is now fully loaded and rebooted.

can you remove and replace the usb stick please

holta, 2017-09-29:
I am not home now but have removed and re-inserted the USB stick several times in several different ways 2h ago (as explained then, on this tkt).

Has something promising changed since?

holta, 2017-09-29:
FYI @georgejhunt same story every time, each time I've (again now) removed & re-inserted teacher's USB stick on 10.8.0.26 in different ways.  (Same as 4h ago, per my earlier comments @ https://github.com/iiab/iiab/issues/329#issuecomment-333164866 explaining this.)

Any other paths we can try?

holta, 2017-09-29:
Clarif for @georgejhunt I also tried inserting multiple USB sticks at the same time (each properly prepared with [Teacher's Content](http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F) in their "usb" directories) which also did not help on this 10.8.0.26 and others.

holta, 2017-09-29:
@georgejhunt & All: I just modified http://BOX.IIAB.IO question 4 to recommend FAT32-formatted USB memory sticks.

Can you please read http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F and confirm this wording it OK or close?  Thanks!

georgejhunt, 2017-09-29:
> IIAB/FAQ#Can_teachers_display_their_own_content.3F and confirm this

I found a reference that suggested that udev was being started before the
read-only disk was remounted rw.

So I stuck a restart in rc.local -- hopefully long after it is rw.

Please reboot, replug, and check for function. I think Adam said that he
was using much faster SD cards than I.  That might explain the difference
in our experience.

Reference:
https://bbs.archlinux.org/viewtopic.php?id=179850

holta, 2017-09-29:
George you appear to have almost solved it, Goddam it :)

After reboot and then removal of stick anyway.  A 2nd stick inserted alongside the 1st stick works too.

It did NOT work immediately after booting, which it should (extremely common teaching scenario, to reboot with Teacher's Content sticks remaining in the IIAB device) but it sounds like we're close!

holta, 2017-09-29:
@georgejhunt I did 5+ reboots of RPi3 10.8.0.26 and [Teacher's Content](http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F) shows up immediately upon reboot at http://box/usb (I guess that 1st boot is not something is too concerning?)

Sounds like you completely nailed it: or are there other timing scenarios we need to test before merging, to try to prevent this from happening again?

jvonau, 2017-09-29:
I'll note roles/usb-lib/templates/umount.d/70-usb-library uses a full path to 'rm' while roles/usb-lib/templates/mount.d/70-usb-library does not have the full path to 'ln'. Think udev needs the full path as there is no environment loaded so there would be not $PATH available to search.

georgejhunt, 2017-09-29:
I know both Tim and Jerry have a dislike for rc.local fixes. So I've
created a unit file which tries to do a udev restart later in the boot
process.

Can you reboot, and repeat the function test

holta, 2017-09-29:
Looks Good: a small battery of ~5 reboots on 10.8.0.26 (removing/re-inserting USB sticks with Teacher's Content in different ways) all worked.

SIDE OBSERVATION: the following URL's sometimes switch back and forth from USB0 to USB1...so schools/teachers thatkeep use 2 different USB sticks resident in their school servers will of course have to stay awake as these URL's bounce around...no worries at all, it's a good thing to Stay Woke :)

http://10.8.0.26/usb/USB0/
http://10.8.0.26/usb/USB1/

holta, 2017-09-30:
Very likely fixed by PR #345 !

Of course: needs testing on a couple different OS/HW scenarios like {Ubuntu on NUC, RPi3, RPi Zero W with USB dongle and eventually others...}

@georgejhunt & All: should this works with SD cards inserted into the side of "big iron" NUC's like the NUC6i3SYH many use ?

holta, 2017-09-30:
UPDATE: Huge progress that the basic functionality now works (i.e. if the teacher remembers to remove & re-insert her USB stick after a cold boot) which has now been checked into IIAB 6.4.

And we remain hopeful USB sticks / Teacher's Content can in future be inserted just prior to a cold boot too.  e.g. if we're lucky and @georgejhunt makes [further progress here](https://github.com/iiab/iiab/pull/345#issuecomment-333329796) in coming days!

holta, 2019-07-26:
Good enough.  In some rare cold boot situations, the stick's content does not appear at http;//box/usb but a manual workaround is sufficient for now (just yank out and re-insert the stick...)

closed 2019-07-26 by holta