iiab/iiab #330: Elgg 2.3.4 regression if installing with Ansible 2.4.0

holta, 2017-09-28:
When installing on Raspbian Stretch Lite using http://download.iiab.io/6.4/rpi/load-ansible2.4.0-vpn.txt or similar.  This is the kind of error that appears in browser: (when trying to visit http://box/elgg)
```
Fatal error in exception handler. Check log for Exception at time 1506558199
```
On reboot, it shows:
```
Fatal error in exception handler. Check log for Exception at time 1506558624
```
Refs: #217 #306 #307 https://github.com/iiab/iiab/pull/316#issuecomment-332694498

holta, 2017-09-28:
This failure of http://box/elgg did not appear when I re-ran a tightened up http://download.iiab.io/6.4/rpi/load-ansible2.4.0-vpn.txt on a completely new Raspbian Stretch Lite!

@jvonau did you solve it within PR #316 somehow?

jvonau, 2017-09-28:
Not that I'm aware of... Would be something to check on after a second run
thou... Didn't you mention you were going to try a sudo runansible as the
pi user at some point?

holta, 2017-09-28:
A plausible theory but it didn't quite hold:

Elgg  2.3.4 installed correctly during the 1st ./runansible on a new RPi3, but then http://box/elgg stopped working after "./runtags elgg" &mdash; then somehow ~10min later http://box/elgg seems to have fixed itself ???  Crazy.

I can't reproduce it anymore (subsequent runs of "./runtags elgg" no longer break it) but here's that (mysteriously clean) output of "./runtags elgg" that seemingly caused its working http://box/elgg to (temporarily for several minutes) break:

http://paste.ubuntu.com/25631359/

Observation 1: http://box/elgg worked (at least once anyway) when "wordpress_install: True" and "wordpress_enabled: True" were in local_vars.yml during Ansible 2.4.0's very 1st run.

Observation 2: this failure does NOT seem to happen on RPi machines built using Ansible 2.2.1 (where ".runtags elgg" can be run as often as you want, and http://box/elgg always continues working, insofar as I can tell!?)

Observation 3: when http://box/elgg mysteriously/magically fixed itself ~10min after I ran "./runtags elgg" on one RPi3... the only thing I did on that RPi3 (during those ~10min, no voodoo applied!) was run functional tests of http://box/nextcloud and http://box/wordpress as listed in Question 3 @ http://FAQ.IIAB.IO

Observation 4: subsequent runs of "./runtags elgg" and subsequent reboots neither fixed not broke Elgg on the 2 RPi3's in question (that were built up on Ansible 2.4.0).  The only difference I know between these 2 RPi'3 (both built with http://download.iiab.io/6.4/rpi/load-ansible2.4.0-vpn.txt on Raspbian Stretch Lite) is... that the RPi3 where http://box/elgg "generally works" is the one where Ansible 2.2.1 was never installed ("apt install ansible" was never run).  But possibly that's just a coincidence...

holta, 2017-09-29:
This is hard to reproduce, but it just happened again on a 3rd RPi3 &mdash; and is not fixed by rebooting, or anything else we know of :/

Here is another example in-browser error when visiting http://box/elgg :
```
Fatal error in exception handler. Check log for Exception at time 1506651604
```
Something in the following ~20 lines during the 2nd execution of ./runansible on RPi3 (IIAB 6.4 RC6 on Raspbian Stretch Lite 2017-09-07) seems to have caused http://box/elgg to break:
```
TASK [elgg : download current version from our copy] ***************************
ok: [127.0.0.1]

TASK [elgg : Determine if software is already expanded] ************************
ok: [127.0.0.1]

TASK [elgg : Expand it to our location] ****************************************
skipping: [127.0.0.1]

TASK [elgg : Create a link to the versioned elgg folder] ***********************
ok: [127.0.0.1]

TASK [elgg : Substitute our parameters in engine/settings.example.php] *********
ok: [127.0.0.1]

TASK [elgg : Copy default .htaccess to the root directory of elgg tree] ********
changed: [127.0.0.1]

TASK [elgg : Modify .htaccess to have RewriteBase as our directory] ************
changed: [127.0.0.1]

TASK [elgg : Change permissions on engine directory so apache can write] *******
ok: [127.0.0.1]

TASK [elgg : Create an upload directory that Apache can write in or elgg] ******
ok: [127.0.0.1]

TASK [elgg : change ownership] *************************************************
ok: [127.0.0.1]

TASK [elgg : Create a user to access the elgg database] ************************
skipping: [127.0.0.1] => (item=127.0.0.1)
skipping: [127.0.0.1] => (item=::1)
skipping: [127.0.0.1] => (item=localhost)

TASK [elgg : Create file to load database] *************************************
changed: [127.0.0.1]

TASK [elgg : Create a mysql database for elgg] *********************************
changed: [127.0.0.1]

TASK [elgg : Load elgg database dump] ******************************************
changed: [127.0.0.1]

TASK [elgg : Remove database dump after load] **********************************
changed: [127.0.0.1]

TASK [elgg : Install config file for elgg in Apache] ***************************
changed: [127.0.0.1]

TASK [elgg : enable elgg] ******************************************************
changed: [127.0.0.1]

TASK [elgg : disable elgg] *****************************************************
skipping: [127.0.0.1]

TASK [elgg : add elgg to service list] *****************************************
ok: [127.0.0.1] => (item={u'option': u'name', u'value': u'elgg-social-netwoking'})
ok: [127.0.0.1] => (item={u'option': u'description', u'value': u'"Elgg is an award-winning social networking engine, delivering the building blocks that enable businesses, schools, universities and associations to create their own fully-featured social networks and applications"'})
ok: [127.0.0.1] => (item={u'option': u'path', u'value': u'/opt/elgg'})
changed: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})

TASK [elgg : Restart apache, so it picks up the new aliases] *******************
changed: [127.0.0.1]
```
_What else should I test, including PR #331 presumably (stricter syntax for Ansible 2.4.x, even if that's going out on a limb?!)_

holta, 2017-09-29:
New evidence shows that running "cd /opt/iiab/iiab; ./runtags elgg" 6 times in a row randomly enables or disables Elgg.

The odds are about 50/50 whether Elgg will work or not work after each run of "./runtags elgg".

Whichever state you are left in, is a permanent state (unaffected by reboots, on these RPi3 machines running 6.4 RC6 anyway, on Raspbian Lite 2017-09-07).

UGLY WORKAROUND BUT IT USUALLY WORKS: run "./runtags elgg" 5 times in a row, and your odds are _about 31-out-of-32_ that Elgg will finally be working at that point (about an hour later, if you have patience anyway :-)

holta, 2017-09-29:
Not sure if this result from @tim-moody is related: (who writes "on my rpi looks like elgg db password is wrong")

> [Fri Sep 29 21:41:54.133521 2017] [:error] [pid 26203] [client 192.168.0.56:50696] Exception at time 1506721314 : fatal error in exception handler : An exception occurred in driver: SQLSTATE[HY000] [1045] Access denied for user 'Admin'@'localhost' (using password: YES)

@jvonau responds:
> little background note on the "Access denied for user 'Admin'@'localhost' (using password: YES)"  mentioned here:  https://github.com/geerlingguy/ansible-role-mysql  about elgg see PR#331 and I'm wondering if the force re-run of mysql in roles/elgg/meta/main.yml is messing things up...

holta, 2017-09-29:
@jvonau elaborates on the idea immediately above:
> How many times do we have to initialize mysql? The answer is once and only once. Could be as simple as removing elgg/meta/main.yml (and the other roles that list mysql as a dependency) that I strongly disagreed with to begin with years ago.

holta, 2017-10-01:
1) My 50/50 stats from ~29h ago were not borne out by a wider sample size: 10+ attempts at "./runtags elgg" including 4+ reboots did not repair Elgg on a clean install of IIAB 6.4 RC6+, despite 10 hours trying, not a single attempt worked.  This is clearly _not_ adequate.

2) Logs within /var/log/apache2/error.log can be searched/grep'd for the error message's timestamp if necessary, e.g. here is an example (pastebin https://paste.fedoraproject.org/paste/kJssxNYBvQmRe-6RIDUYew) and excerpts:

> PHP Notice:  Use of undefined constant EDT - assumed 'EDT' in /opt/elgg-2.3.4/elgg-config/settings.php on line 2...

> PHP Notice:  date_default_timezone_set(): Timezone ID 'EDT' is invalid in /opt/elgg-2.3.4/elgg-config/settings.php on line 2...

> SQLSTATE[HY000] [1045] Access denied for user 'Admin'@'localhost' (using password: YES) in /opt/elgg-2.3.4/vendor/doctrine/dbal/lib/Doctrine/DBAL/Driver/PDOConnection.php:43...

jvonau, 2017-10-01:
Refs #69 #70 #102 ansible reporting is used in default_vars.yml local_tz: "{{ ansible_date_time.tz }}", that is where the 'EDT' comes from, however php expects the format to be the same as what is used in /etc/timezone.

holta, 2017-10-01:
Can anyone yet tell if the root cause is likely:

1) EDT (Timezone format php expects...)

...and/or is there a 2nd root cause with...

2) SQL's "Access denied" issue for user 'Admin' with password 'changeme' ?

tim-moody, 2017-10-01:
https://github.com/XSCE/xsce/blob/release-6.2/roles/elgg/defaults/main.yml had
timezone: America/New_York

I doubt the tz and dbuser are related

tim-moody, 2017-10-01:
unless failure on tz stops script that would create dbuser

tim-moody, 2017-10-01:
MariaDB [elggdb]> SELECT User FROM mysql.user;
+-----------+
| User      |
+-----------+
| nextcloud |
| root      |
| nextcloud |
| root      |
| nextcloud |
| root      |

Admin is not a user

tim-moody, 2017-10-01:
btw this is ansible 2.2.1.0 on rpi

tim-moody, 2017-10-01:
N.B. the db tables were created and populated

tim-moody, 2017-10-01:
if install True and enable False, then runansible will download and expand elgg zip file but it will not create the database user because elgg is not enabled

the next time elgg if elgg is enabled it will not create the user because elgg.stat.exists will be true

tim-moody, 2017-10-01:
this is another instance of locking the door before everyone goes through it

holta, 2017-10-01:
@tim-moody can u verify the same checks on master / Ansible 2.4.0 on RPi3 I just placed on 10.8.0.26 ?

(Or tell me how if easier?)

PS 10.8.0.26 on XSCENET built using http://download.iiab.io/6.4/rpi/load-vpn.txt

tim-moody, 2017-10-01:
this is master

jvonau, 2017-10-01:
Opps little to far, PR #356 addresses the dbuser issue, I'm not a fan of the loading the dbdump as the hostnames between what is in the dump and the actual machine can differ(#179 ).

tim-moody, 2017-10-01:
the dump was already being loaded. I didn't change that.

jvonau, 2017-10-01:
Perhaps it might be better if the load scripts shipped with both install and enabled True? The in the field user could just toggle the service off via local_vars or the admin console should the service be unwanted by user post install but is ready to go otherwise.

jvonau, 2017-10-01:
@tim-moody has #355 that I didn't see, go with that PR but add the disable for redhat from https://github.com/iiab/iiab/pull/331/commits/4cabca69f5f1353b2c7eab58882eb415e60f9371

holta, 2017-10-01:
Clarif: the initial install of Elgg 2.3.4 almost always worked (even with Ansible 2.4.0 over the past 4 days) so I've corrected+clarified the subject line of this ticket &mdash; it's the subsequent installs using "./runtags elgg" or Admin Console that essentially always failed.

holta, 2017-10-02:
Fixed by PR #355 (extensively tested over the past 6 hours)

closed 2017-10-02 by holta