iiab/iiab #332: osm error installing IIAB 6.4 RC5+

holta, 2017-09-28:
Thanks @kananigit who just found this error while running http://download.iiab.io/6.4/rpi/load-big.txt on RPi3

> 2017-09-28 01:40:21,450 p=7260 u=root |  TASK [osm : Install Whoosh 2.6] ************************************************
2017-09-28 01:41:26,507 p=7260 u=root |  changed: [127.0.0.1]
2017-09-28 01:41:26,696 p=7260 u=root |  TASK [osm : Install IIAB with dependencies] ************************************
2017-09-28 01:41:53,247 p=7260 u=root |  changed: [127.0.0.1] => (item=MarkupSafe)
2017-09-28 01:42:13,677 p=7260 u=root |  changed: [127.0.0.1] => (item=pytz)
2017-09-28 01:42:47,099 p=7260 u=root |  failed: [127.0.0.1] (item=Internet-in-a-Box) => {"cmd": "/usr/local/osm/bin/pip2 install --disable-pip-version-check Internet-in-a-Box", "failed": true, "item": "Internet-in-a-Box", "msg": "stdout: Collecting Internet-in-a-Box\n  Downloading Internet-in-a-Box-0.5.10.tar.gz (942kB)\nCollecting Flask>=0.10 (from Internet-in-a-Box)\n  Downloading Flask-0.12.2-py2.py3-none-any.whl (83kB)\nCollecting Flask-Babel>=0.8 (from Internet-in-a-Box)\n  Downloading Flask_Babel-0.11.2-py2.py3-none-any.whl\nCollecting Flask-AutoIndex (from Internet-in-a-Box)\n  Downloading Flask-AutoIndex-0.6.tar.gz (54kB)\nCollecting Flask-SQLAlchemy (from Internet-in-a-Box)\n\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/commands/install.py\", line 353, in run\n    wb.build(autobuilding=True)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/wheel.py\", line 749, in build\n    self.requirement_set.prepare_files(self.finder)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 554, in _prepare_file\n    require_hashes\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_install.py\", line 278, in populate_link\n    self.link = finder.find_requirement(self, upgrade)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 792, in get_page\n    \"Cache-Control\": \"max-age=600\",\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 501, in get\n    return self.request('GET', url, **kwargs)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/download.py\", line 386, in request\n    return super(PipSession, self).request(method, url, *args, **kwargs)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 488, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 609, in send\n    r = adapter.send(request, **kwargs)\n  File \"/usr/local/osm/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py\", line 47, in send\n    resp = super(CacheControlAdapter, self).send(request, **kw)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py\", line 423, in send\n    timeout=timeout\n  File \"/usr/local/osm/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 643, in urlopen\n    _stacktrace=sys.exc_info()[2])\n  File \"/usr/local/osm/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py\", line 315, in increment\n    total -= 1\nTypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n"}

holta, 2017-09-28:
Joshua clarifies that he installed on Raspbian Stretch Lite from 2017-09-07

jvonau, 2017-09-28:
I'll go with older test install where easy_setup was used to upgrade pip in the past then updated to current master.

holta, 2017-09-28:
Has anybody else seen this on Ubuntu?  Or did cosmic rays cause this ;)

(It has never occurred on Raspbian to my knowledge...)

georgejhunt, 2017-09-28:
This thread starts out saying local_vars big on rpi3.
Tim and I both had successful runs on rpi3, and my version has a
functioning OSM -- no tiles loaded (but the search window appears)

We need more clarity I think

holta, 2017-09-29:
@kananigit can you tell us exact steps to reproduce this?

Thanks!

holta, 2017-09-29:
@kananigit asked me to close this ticket, saying this is no longer an issue for him.

closed 2017-09-29 by holta