iiab/iiab #346: fix XSCE_DIR & XSCE_BASE_PATH; consider {{ iiab_base_path }} instead of {{ base_path }}

holta, 2017-09-30:
Let's rename XSCE_DIR to IIAB_DIR in these 5 files:
```
root@box:/opt/iiab/iiab# grep -r XSCE_DIR *
iiab-network:  cd $XSCE_DIR
iiab-network:export ANSIBLE_LOG_PATH="$XSCE_DIR/iiab-network.log"
install-support:export ANSIBLE_LOG_PATH="$XSCE_DIR/iiab-install.log"
roles/1-prep/templates/iiab.env.j2:XSCE_DIR={{ iiab_dir }}
runansible:  cd $XSCE_DIR
runansible:  XSCE_DIR=/opt/iiab/iiab
runansible:export ANSIBLE_LOG_PATH="$XSCE_DIR/iiab-install.log"
runtags:  cd $XSCE_DIR
runtags:export ANSIBLE_LOG_PATH="$XSCE_DIR/iiab-debug.log"
```
Is the name IIAB_DIR confusing given we already have {{ iiab_dir }} in 7 files?  Seem acceptable as they refer to same thing before Ansible is installed (above) and after Ansible is installed (below):
```
root@box:/opt/iiab/iiab# grep -r iiab_dir *
roles/httpd/templates/iiab-info.conf.j2:Alias /info/ {{ iiab_dir }}/docs/html/
roles/httpd/templates/iiab-info.conf.j2:Alias /info {{ iiab_dir }}/docs/html/
roles/httpd/templates/iiab-info.conf.j2:<directory {{ iiab_dir }}/docs/html>
roles/1-prep/templates/iiab.env.j2:XSCE_DIR={{ iiab_dir }}
roles/1-prep/tasks/iiab_ini.yml:    - option: 'iiab_dir'
roles/1-prep/tasks/iiab_ini.yml:      value: '{{ iiab_dir }}'
roles/activity-server/tasks/main.yml:  command: rsync -a {{iiab_dir}}/roles/activity-server/files/lang_templates /library/xs-activity-server/
roles/2-common/tasks/fl.yml:  file: path={{ iiab_dir }}
roles/2-common/tasks/iiab_ini.yml:    - option: 'iiab_dir'
roles/2-common/tasks/iiab_ini.yml:      value: '{{ iiab_dir }}'
vars/default_vars.yml:iiab_dir: "{{ iiab_base }}/iiab"
```
Can we consider renaming {{ iiab_base }} to {{ iiab_base_path }} in these 11 files?
```
root@box:/opt/iiab/iiab# grep -r iiab_base *
roles/debian_schooltool/tasks/main.yml:             dest={{ iiab_base }}
roles/debian_schooltool/tasks/main.yml:  file: src={{ iiab_base }}/{{ schooltool_version }}
roles/debian_schooltool/tasks/main.yml:        dest={{ iiab_base }}/schooltool
roles/debian_schooltool/tasks/main.yml:  shell: command='$( cd {{ iiab_base }}/schooltool; /usr/bin/make ) '
roles/1-prep/templates/iiab.env.j2:XSCE_BASE_PATH={{ iiab_base }}
roles/1-prep/defaults/main.yml:iiab_base_ver: 6.4
roles/1-prep/defaults/main.yml:admin_console_path: "{{ iiab_base }}/admin_console"
roles/1-prep/defaults/main.yml:cmdsrv_path: "{{ iiab_base }}/iiab_cmdsrv"
roles/1-prep/tasks/iiab_ini.yml:    - option: 'iiab_base'
roles/1-prep/tasks/iiab_ini.yml:      value: '{{ iiab_base }}'
roles/moodle/defaults/main.yml:moodle_base: "{{ iiab_base }}/moodle"
roles/kiwix/templates/kiwix-serve.service.j2:ExecStart={{ iiab_base }}/kiwix/bin/kiwix-serve --daemon --port {{ kiwix_port }} --nolibrarybutton --library {{ kiwix_library_xml }}
roles/kiwix/templates/iiab-make-kiwix-lib.py:iiab_base_path = "{{ iiab_base }}"
roles/kiwix/templates/iiab-make-kiwix-lib.py:kiwix_manage = iiab_base_path + "/kiwix/bin/kiwix-manage"
roles/kiwix/templates/iiab-make-kiwix-lib.py:    global iiab_base_path
roles/kiwix/templates/iiab-make-kiwix-lib.py:    iiab_base_path = config.get('location','iiab_base')
roles/kiwix/templates/iiab-make-kiwix-lib.py:    kiwix_manage = iiab_base_path + "/kiwix/bin/kiwix-manage"
roles/kiwix/defaults/main.yml:kiwix_path: "{{ iiab_base }}/kiwix"
roles/kiwix/tasks/kiwix_install.yml:  stat: path={{ iiab_base }}/kiwix/bin/kiwix-serve
roles/kiwix/tasks/kiwix_install.yml:             dest="{{ iiab_base }}"
roles/kiwix/tasks/kiwix_install.yml:  file: path="{{ iiab_base }}/kiwix/bin"
roles/kiwix/tasks/kiwix_install.yml:             dest="{{ iiab_base }}/kiwix/bin"
roles/kiwix/tasks/kiwix_install.yml:  command: "chown -R root:root {{ iiab_base }}"
roles/2-common/tasks/iiab_ini.yml:    - option: 'iiab_base'
roles/2-common/tasks/iiab_ini.yml:      value: '{{ iiab_base }}'
vars/default_vars.yml:iiab_base: /opt/iiab
vars/default_vars.yml:iiab_dir: "{{ iiab_base }}/iiab"
vars/default_vars.yml:pip_packages_dir: "{{ iiab_base }}/pip-packages"
vars/default_vars.yml:yum_packages_dir: "{{ iiab_base }}/yum-packages"
vars/default_vars.yml:downloads_dir: "{{ iiab_base }}/downloads"
```
Making it more understandable, and consistent with XSCE_BASE_PATH (that we'd rename to IIAB_BASE_PATH) below?
```
root@box:/opt/iiab/iiab# grep -r XSCE_BASE_PATH *
roles/1-prep/templates/iiab.env.j2:XSCE_BASE_PATH={{ iiab_base }}
```
That is **if** we can reconcile {{ iiab_base_path }} already being a synonym for {{ iiab_base }} here &mdash; as this looks like it contains some redundant code here &mdash; setting {{ iiab_base_path }} twice, both of which may be unecessary if we agree on an Ansible-wide variable {{ iiab_base_path }} instead:
```
root@box:/opt/iiab/iiab# grep -r iiab_base_path *
roles/kiwix/templates/iiab-make-kiwix-lib.py:iiab_base_path = "{{ iiab_base }}"
roles/kiwix/templates/iiab-make-kiwix-lib.py:kiwix_manage = iiab_base_path + "/kiwix/bin/kiwix-manage"
roles/kiwix/templates/iiab-make-kiwix-lib.py:    global iiab_base_path
roles/kiwix/templates/iiab-make-kiwix-lib.py:    iiab_base_path = config.get('location','iiab_base')
roles/kiwix/templates/iiab-make-kiwix-lib.py:    kiwix_manage = iiab_base_path + "/kiwix/bin/kiwix-manage"
```

holta, 2017-09-30:
@georgejhunt & All: how is `/etc/iiab/iiab.env` created, to help new folk understand?

Certainly it contains 3 references to XSCE that should be fixed:
```
root@box:~# more /etc/iiab/iiab.env
# This is a configuration file for XSCE
# It can sourced in a shell script or read into an application
XSCE_BASE_PATH=/opt/iiab
XSCE_DIR=/opt/iiab/iiab
OS=raspbian
OS_VER=raspbian-9
WWWROOT=/library/www/html
```

holta, 2017-09-30:
Here are 3 more examples, in 3 files prominent files:
```
root@box:/opt/iiab/iiab# grep -r XSDOMAIN *
runansible:XSDOMAIN=""
runansible: XSDOMAIN=`cat /etc/sysconfig/xs_domain_name`
runansible: XSDOMAIN=$1
runansible:if [ x"$XSDOMAIN" != x ]
runansible: ARGS="--extra-vars '{\"iiab_domain\":\"$XSDOMAIN\"}'"
```
Likewise these 2:
```
runtags: echo "XSCE Playbook not found."
```
```
iiab-network: echo "XSCE Playbook not found."
```
Finally running "cd /opt/iiab/iiab; grep -r XS *" will help us root out stale documentation helping us all move forward together, new folks and OLPC alums all now building something new open to all, for [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) !

holta, 2017-10-17:
Ref https://github.com/iiab/iiab-admin-console/issues/37 (remove xsce legacy)

holta, 2017-10-24:
PR #303 may begin this work.

Alternate View: https://github.com/iiab/iiab/search?utf8=%E2%9C%93&q=XSCE_DIR

jvonau, 2017-11-07:
#490

holta, 2017-11-22:
All above stale references to XS or XSCE have been confirmed removed, thanks to @jvonau !

closed 2017-11-22 by holta