iiab/iiab #347: network "FAILED!" red messaging: can this be improved for new contributors?

holta, 2017-09-30:
These loud/red messages during ./runansible may be just warnings, but they frightens off new contributors:

>TASK [network : Tear down any bridge and start fresh] **************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["ip", "link", "set", "br0", "down"], "delta": "0:00:00.017442", "end": "2017-09-30 10:55:54.843294", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-09-30 10:55:54.825852", "stderr": "Cannot find device \"br0\"", "stderr_lines": ["Cannot find device \"br0\""], "stdout": "", "stdout_lines": []}
...ignoring

> TASK [network : and remove the device] *****************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["brctl", "delbr", "br0"], "delta": "0:00:00.028423", "end": "2017-09-30 10:55:55.602927", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-09-30 10:55:55.574504", "stderr": "bridge br0 doesn't exist; can't delete it", "stderr_lines": ["bridge br0 doesn't exist; can't delete it"], "stdout": "", "stdout_lines": []}
...ignoring

What's a better way?  As we're now making great progress cleaning up to welcome new contributors, who really do get derailed by the "false alarms" :

#337 3 red warnings when running "systemctl" (and on boot too?)
#343 "Stop old xsce-cmdsrv" red "FAILED!" + 4 purple WARNINGS during Admin Console install

jvonau, 2017-09-30:
Fixed in one of my later branches that are on hold for 6.5

holta, 2017-09-30:
> Fixed in one of my later branches that are on hold for 6.5

Do you remember which if poss?

(In this [this list](https://github.com/iiab/iiab/milestone/2) presumably?)

jvonau, 2017-09-30:
Here is another one
> ==> ubuntu: TASK [network : Enabling ifcfg slaves] *****************************************
==> ubuntu: skipping: [127.0.0.1] => (item=enp0s10) 
==> ubuntu: skipping: [127.0.0.1] => (item=enp0s8) 
==> ubuntu: skipping: [127.0.0.1] => (item=enp0s9) 
==> ubuntu: 
==> ubuntu: TASK [network : in upgrade from earlier 6.2, delete the resolvconf] ************
==> ubuntu: fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unsupported parameters for (apt) module: enabled Supported parameters include: allow_unauthenticated,autoclean,autoremove,cache_valid_time,deb,default_release,dpkg_options,force,force_apt_get,install_recommends,only_upgrade,package,purge,state,update_cache,upgrade"}
==> ubuntu: ...ignoring
==> ubuntu: 
==> ubuntu: TASK [network : Get the dhcp client daemon used in recent raspbian] ************
==> ubuntu: changed: [127.0.0.1]

jvonau, 2017-09-30:
And one for hostapd:
> ==> ubuntu: TASK [network : Create a config file for hostapd] ******************************
==> ubuntu: skipping: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : Disable the Access Point Hostapd program] **********************
==> ubuntu: changed: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : Use custom systemd unit file to start hostapd] *****************
==> ubuntu: changed: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : ask systemd to reread the unit files for hostapd] **************
==> ubuntu: changed: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : Enable the Access Point Hostapd program] ***********************
==> ubuntu: skipping: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : Checking if slave is active waiting 10 seconds] ****************
==> ubuntu: skipping: [127.0.0.1]
==> ubuntu: 
==> ubuntu: TASK [network : Restart hostapd if slave is inactive] **************************
==> ubuntu: fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout'"}
==> ubuntu: ...ignoring
==> ubuntu: 
==> ubuntu: TASK [postgresql : Install postgresql packages] ********************************
==> ubuntu: skipping: [127.0.0.1] => (item=postgresql) 
==> ubuntu:

holta, 2017-10-01:
In addition to:
#337 3 red warnings when running "systemctl" (and on boot too?)
#343 "Stop old xsce-cmdsrv" red "FAILED!" + 4 purple WARNINGS during Admin Console install

See also:
#351 Using Ansible File module to fix warnings
#352 Using Ansible file module to fix warnings
#353 Using Ansible get_url module to fix warnings

To keep important "CX" improvements from getting lost!

(*) where CX = UX for contributors, as we establish more engaging+meaningful onramps to eager civic contributors across each+all country where we can.

jvonau, 2017-10-06:
list of noted errors thanks to jenkins some are already reported above some are not. 
[run33-noted.txt](https://github.com/iiab/iiab/files/1364021/run33-noted.txt)

holta, 2017-10-14:
@jvonau is this sufficiently under control to resolve now?

jvonau, 2017-10-14:
Let's keep this one open for tracking PR's against as above and close #337 as mongo and sugarizer have been marked resolved. I'll track isc-dhcp-server.service progress from this PR.

jvonau, 2017-11-02:
> TASK [postgresql : Initialize the postgres db] ********************************************************
 [WARNING]: Consider using 'become', 'become_method', and 'become_user' rather than running su

holta, 2017-11-13:
This is sufficiently resolved right?

Plz re-open as nec.

closed 2017-11-13 by holta