iiab/iiab #348: Broken Symlink:  iiab-console.conf.j2 points to /opt/schoolserver

arky, 2017-10-01:
Can someone look into this broken template symlink. 

> $ ls -l iiab/roles/httpd/templates/iiab-console.conf.j2
>
> lrwxrwxrwx 1 arky root 76 Oct  1 11:53 iiab/roles/httpd/templates/iiab-console.conf.j2 -> /opt/schoolserver/xsce/roles/xsce-admin/templates/console/xs-console.conf.j2
>

jvonau, 2017-10-01:
#354 > no. the template should just be removed as it is not used.

jvonau, 2017-10-02:
Think that is a missed remnant of the initial split of cmdsrv/admin-console to its own git repo that then got caught up in the massive xsce -> iiab rename a bit later. S/B safe to remove as the symlink is invalid but will defer to @tim-moody who has git access and can deal directly with the issue.

tim-moody, 2017-10-02:
couldn't have said it better

holta, 2017-10-05:
Per our community/team call agreement (http://minutes.iiab.io) to merge #354 immediately after IIAB 6.4 was released, which I have now done.

closed 2017-10-05 by holta