iiab/iiab #350: debian: Apt: Unsupported parameters for (apt) module

arky, 2017-10-01:
Fatal error from Ansible. Report from nightly Jenkins CI run on default Debian 9 target.

> *****************************************
> 2017-10-01 05:37:51,395 p=4312 u=root |  skipping: [127.0.0.1] => (item=eth1)
> 2017-10-01 05:37:51,406 p=4312 u=root |  skipping: [127.0.0.1] => (item=eth2)
> 2017-10-01 05:37:51,423 p=4312 u=root |  TASK [network : in upgrade from earlier 6.2, delete the resolvconf] ***********************************************************************************
> 2017-10-01 05:37:52,806 p=4312 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unsupported parameters for (apt) module: enabled Supported parameters include: allow_unauthenticated,autoclean,autoremove,cache_valid_time,deb,default_release,dpkg_options,force,force_apt_get,install_recommends,only_upgrade,package,purge,state,update_cache,upgrade"}
> 2017-10-01 05:37:52,807 p=4312 u=root |  ...ignoring
> 2017-10-01 05:37:52,822 p=4312 u=root |  TASK [network : Get the dhcp client daemon used in recent raspbian] ***********************************************************************************
> 2017-10-01 05:37:56,659 p=4312 u=root |  changed: [127.0.0.1]
> 2017-10-01 05:37:56,680 p=4312 u=root |  TASK [network : for upgrades from earlier 6.2, remove br0 file] ***************************************************************************************
> 2017-10-01 05:37:56,862 p=4312 u=root |  ok: [127.0.0.1]
>

arky, 2017-10-01:
Ansible package modules doesn't have "enabled=False" https://github.com/iiab/iiab/blob/master/roles/network/tasks/debian.yml#L16

However turning this one doesn't make the error go away.

jvonau, 2017-10-01:
It's been observed and reported on in the past with the fix pending in one of my branches scheduled for inclusion for 6.5 that was passed over for inclusion for 6.4

holta, 2017-10-01:
Hey everyone sorry [40 things](https://github.com/iiab/iiab/milestone/2) have been deferred from 6.4 to 6.5 already, so we can saw it off and ship soon in coming days :)

PS @jvonau do you have a link handy, to the PR/branch most relevant here?

jvonau, 2017-10-01:
Well that started out as #169 #170 #178 - > refined in #243 #250 -> split out to smaller bits of which some have been merged. #305 covers some testing I did with my revised_net branch that does not have an open PR for at the moment as a rebase is needed and you scheduled that for 6.5.

holta, 2017-10-01:
> Well that started out as #169 #170 #178 - > refined in #243 #250 -> split out to smaller bits of which some have been merged. #305 covers some testing I did with my revised_net branch that does not have an open PR for at the moment as a rebase is needed and you scheduled that for 6.5.

Hugely helpful lineage, Thanks!

jvonau, 2017-10-01:
Well opps, revised_net does have a PR #267 but needs a rebase as master
shifted. PR #303 and new PR would be needed for the 'permissions' branch
mentioned in testing of a PR #305

jvonau, 2017-11-07:
resolved in #490

jvonau, 2017-11-12:
s/b resolved problem code doesn't exist now

closed 2017-11-12 by holta