iiab/iiab #368: avoid initializing MySQL so many times during IIAB's Ansible install?

holta, 2017-10-03:
This takes many minutes during 5-xo-services (and possibly other places too?)

As @jvonau [suggested](https://github.com/iiab/iiab/issues/330#issuecomment-333253611) earlier, this might be a relatively quick win for IIAB 6.5 or 6.6?

(Certainly the vast majority of implementers don't use XO services in 2017/2018!)

holta, 2017-10-03:
Is this also an issue with PostgreSQL, which appears to be faster to install, no matter how many times its own installation checks / initializations happen?

Is Jenkins-assisted Profiling possible?!

(MySQL smells like the real slowpoke during ./runansible but I could be wrong!)

holta, 2017-10-03:
Related issue in /opt/iiab/iiab.yml
#370 comment out & document 5-xo-services as it's slow as a dog
PR #371 comment out 5-xo-services (too damn slow)

jvonau, 2017-10-06:
Turn on timestamps in Jenkins that should give the time required to repeat the execution of the duplicated plays. Now add the time up and compare that to the total time for the run should give a percentage spent doing nothing but repeating the role as percentage.

holta, 2017-10-06:
Added to Monday's call agenda @ http://minutes.iiab.io (Sept 9, 10AM NYC Time) even if MySQL is not the worst offender!

Great topic to deepen our understanding of Jenkins/CI/CD in general~

tim-moody, 2017-10-06:
or look at the install log. my impression is that the time for tasks other than the first install are in the milliseconds.

jvonau, 2017-10-06:
jenkin_ip:8080/job/IIAB-debian-jvonau/35/consoleFull using stock med_vars.

end runasible     12:55:46 ==> debian: 127.0.0.1   : ok=392  changed=292  unreachable=0    failed=0   
start runansible  12:34:11 ==> debian: + ./runansible
Total Time               21:35  = 1295 seconds

first install, is needed:
12:42:05 ==> debian: TASK [mysql : Install MySQL Debian]
12:43:39 ==> debian: changed: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})
total 1min 34secs

now 'reinstall' for elgg that is enabled in the vars file
12:43:39 ==> debian: TASK [mysql : Install MySQL Debian]
12:44:15 ==> debian: ok: [127.0.0.1] => (item={u'option': u'enabled', u'value': True})
 diff  35 secs  Approx 3.5% of total time

Note with ansible 2.4 it appears that if the dependence of mysql for apps that declare mysql in the app's meta/main.yml file that are under the same meta group as where mysq is installed (within the 6-generic-apps group) the declared dependency in the app's role is not repeated during the running of 6-generic-apps after the first app(elgg here) as noted by the lack of a second pass through the myslq role that is declared for owncloud nextcloud wordpress as the group is completed while rachel below is installed in 7-edu-apps and does do a second pass through the mysql role. 

And the 'reinstall' is skipped for services (rachel in this case) that are not initially enabled in the vars file but would repeat the above routine if enabled. 
12:48:41 ==> debian: TASK [mysql : Install MySQL Debian] 
12:48:42 ==> debian: skipping: [127.0.0.1] => (item={u'option': u'enabled', u'value': True}) 

Way better handling in ansible 2.4 since the last time this was profiled in the past, needs some attention as to the keeping of apps and there underlining databases grouped together under the same meta role(6- vis 7- ) as the apps that use postgresql are not grouped as nicely as the ones that use mysql as the backend. A bit of reordering and grouping might make the postgresql apps install much the same way with this change in behaviour/new feature of ansible.

jvonau, 2017-10-07:
Um, I went on to expand on the profiling by introducing http://download.iiab.io/6.4/rpi/local_vars_big.yml into Jenkins environment to get a larger kitchen sink install of what is offered. Among the larger options that are enabled by default is **wondershaper** that I have tested in the past but never worked quite right out of the box however without being enabled does no harm, but **when enabled does cause issues during the install**. I consider to this role unmaintained, and needs further testing and refinement before being useful. PR #382 can wait for forever, but can somebody adjust local_vars_big.yml so the public doesn't get tripped up while possibly using this option with 6.4 please.

jvonau, 2017-10-07:
Not sure when the rpi's were tested or what the contents of big_vars was when tested as the datestamp shows Oct 2 on d/l.iiab.io. Jenkins was testing both debian-9 and ubuntu-16 against big_vars, both showed similar behaviour of stalling the next download in the playbook, until I took drastic action of disabling wondershaper by [brute force](https://github.com/jvonau/iiab/commit/e8e5240fa3f553bb6bca408fbb2e43a2454b9564)in my 'test' branch as a means to diagnose this.

jvonau, 2017-10-07:
To answer why the rpi's may of worked, wondershaper relies of the use of code with "iptables -t mangle" support. Perhaps George's discovery in https://github.com/iiab/iiab/compare/master...georgejhunt:captive-test might shed some light
 >if [ "$captive_portal_enabled" == "True" ];then
   $IPTABLES -t mangle -N internet
>- name: Stock rpi does not have conntrack
  package: name=conntrack state=present 

The rpi may not have the needed iptables support out of the box for those rules, and are failing silently when the iptables rules were loaded, I recall seeing something like 'if-up failed run-parts 15-iptables" trolling the logs, but the network looked as a functional gateway with a normal iptables ruleset, the 'tc' rules that are in wondershaper would of not been loaded into memory. Think it was on the 'big test install" from 2+weeks ago but that is a bit fuzzy and can't be sure, I would not of thought to investigate wondershaper's rules as nobody has confirmed that it works for them and was always disabled since its introduction when I said "needs work". Now fast-forward to where maybe full iptables support **is** available out of the box(debian, ubuntu), now rules don't fail to load and are in memory, traffic is now being shaped but incorrectly.

All conjecture and speculation but no hand-wavey tricks, just a conclusion based on observed behaviour with wondershaper in the mix over the years. This is what I noted years ago with the "needs work" comment and wondershaper was set to disabled since then and to my knowledge never been debugged enough to release. Should somebody seek help with wondershaper somebody other than myself can field the questions and answer session, I'd rather not spend time doing a custom setup, as iptables rules really should to be tailored to the location but others are free to play. 

Seems out of place to be enabled=True in the automated install script, as stalling the install is the result. The question is to be proactive in preventing the situation with an enabled=False as it was historically in the past, until resent introduction of optional installation that is now possible to leave the role out altogether, or wait to see what the resolution is or is not going to be on continued support of wondershaper.

holta, 2017-10-07:
Ok, for now I've changed it to:
```
wondershaper_install: True
wondershaper_enabled: False
```
within http://wiki.laptop.org/go/IIAB/local_vars_big.yml and in these places:
* http://download.iiab.io/6.4/rpi/
* http://download.iiab.io/6.5/rpi/

Monday let's all talk this over (http://minutes.iiab.io) to be sure we're all on the right track here, and this is not affecting others in ways we don't yet realize~

holta, 2017-10-27:
@jvonau explained that Roles have moved from meta/main.yml to tasks/main.yml so let's try to prevent {MySQL & PostgreSQL} from installing multiple times based on dependencies.

Even if only 2 apps using PostgreSQL
* Pathagar
* Moodle
* possibly authserver?

holta, 2017-10-27:
@jvonau as we work this out, see Lines 91-147 of [roles/1-prep/tasks/computed_vars.yml](https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/computed_vars.yml#L91) where MySQL and PostgreSQL's _de facto defaults_ are set!

holta, 2017-10-27:
Possible PR @jvonau's outlined:
https://github.com/iiab/iiab/compare/master...jvonau:DB

commit 322fd0a5 (2017-10-28): Merge pull request #472 from jvonau/DB

Db - avoids MySQL & PostgreSQL Install-A-Thon

Closes #368
closed 2017-10-28 by holta