iiab/iiab #369: Novice usability - 'Gateway' not functioning as gateway out of the box.

jvonau, 2017-10-03:
Might want to reconsider the default, might be better as an opt-out option, rather than having the novice user wonder why the internet can't be accessed after a successful installation and is not offered an option with the express install 'load-*' method . As noted from reports the stock default of iiab_gateway_enabled: False that would be used in conjunction with load*.txt that download the respective local_vars.yml file doesn't not allow for out-of-the-box conductivity without manual user intervention after the installation. The end user would have to edit the local_vars.yml, toggle iiab_gateway_enabled: True, then run ./iiab-network, perhaps while logged in via ssh. Also true for vars/local_vars.yml for that matter but you get change to edit that file before beginning.

holta, 2017-10-03:
Most of  the developing world users I've talked to don't want Internet to work for client machines associated with IIAB, as they could not possibly afford to pay for it.

As we move into "middle-class" (2nd world?) countries however, this will likely begin to change.

A clean UX/UI would be wonderful at that point, in conjunction with @tim-moody dream to make the most common operator/implementer functions instantly changeable, without involving Ansible.

jvonau, 2017-10-03:
Valid point, but I can't see somebody in a developing country installing via any sort of internet, as you say " they could not possibly afford to pay for it".  Thinking those situations would entail shipping or sneaker-net type of approach to having the hardware(or SDCard) show up for use. There would be an intermediary step of a deployment person who has to add content and other customizations prior to shipping so that would be the logical point to select that option at that point of time while users of the express curl/load method could have less post installation work and could get on with testing. Speaking of testing does the changing of network roles work? I don't recall that being part of the functional testing now, the bar was once set higher prior to release. Once image creation can be done via Jenkins, I have developed a method to be able to add/edit files within an image prior to minimizing the image(even minimized but that takes longer) and shipping that might be of interest. Dreams are possible but require real work to make happen, not lip service and wishing.

holta, 2017-10-03:
> changing of network roles work

I personally don't test that, as many/most of the IIAB communities I service focus on the core Internet-in-a-Purse / "knowledge hotspot" functionality, and these folk do not choose to customize the network.

But let's try to test that before "Wedn/Thurs release" if possible, as big iron implementers may be rare today, but a few still do exist.

jvonau, 2017-10-03:
If the desire is to move away from ansible for the Admin Console then these variable that are part of iiab-gen-iptables should be moved to a common location to be sourced for use from that location. Hey that sounds like a good idea anyway, and makes a clean break from some more the OLPC-XS heritage/baggage while avoiding having to re-write iiab-gen-iptables. I would suggest /etc/iiiab/iiab.env be used as that location as that is the intended purpose of that file. 

> LANIF=`cat /etc/sysconfig/iiab_lan_device`
WANIF=`cat /etc/sysconfig/iiab_wan_device`
MODE=`grep iiab_network_mode_applied  /etc/iiab/iiab.ini | gawk '{print $3}'
gw_block_https={{ gw_block_https }}
ssh_port={{ ssh_port }}
gui_wan={{ gui_wan }}
gui_port={{ gui_port }}
iiab_gateway_enabled={{ iiab_gateway_enabled }}
services_externally_visible={{ services_externally_visible }}
calibre_port={{ calibre_port }}
kiwix_port={{ kiwix_port }}
kalite_server_port={{ kalite_server_port }}
block_DNS={{ block_DNS }}
/etc/sysconfig/xs_httpcache_on

To start with, that can be fast-track for 6.5, that I'm willing to help and kind of fits with #364 now ;)  

Next would be to duplicate the functionality of the network role in a different programming language, not impossible but no small feat even if you scale back dansguardian, squid, switch to dmsq and if raspbian is not used "predicable interface names" come into play. Once upon a time that was working with bash in the pre-ansible days, but the references were removed with the repo switch. Should not be too hard to pull off, grew a few more features since but all of the detection code was reused in ansible as shell anyway. I'm not volunteering to do the re-write but could be available for questions should somebody take up the task. Happy Recruiting!!

jvonau, 2017-11-13:
Implemented

closed 2017-11-13 by jvonau