iiab/iiab #372: Ubuntu: Downloading ka_lite_static failed

arky, 2017-10-03:
Report from Ubuntu CI VM. 

> 
> 
> ==> ubuntu: 
> ==> ubuntu: TASK [kalite : Install ka-lite with pip] ***************************************
> ==> ubuntu: fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/local/kalite/venv/bin/pip2 install ka-lite-static==0.17.3", "failed": true, "msg": "stdout: Collecting ka-lite-static==0.17.3\n  Downloading ka_lite_static-0.17.3-py2-none-any.whl (40.7MB)\n\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/commands/install.py\", line 335, in run\n    wb.build(autobuilding=True)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/wheel.py\", line 749, in build\n    self.requirement_set.prepare_files(self.finder)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 620, in _prepare_file\n    session=self.session, hashes=hashes)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 821, in unpack_url\n    hashes=hashes\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 659, in unpack_http_url\n    hashes)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 882, in _download_http_url\n    _download_url(resp, link, content_file, hashes)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 603, in _download_url\n    hashes.check_against_chunks(downloaded_chunks)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/utils/hashes.py\", line 46, in check_against_chunks\n    for chunk in chunks:\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 571, in written_chunks\n    for chunk in chunks:\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/utils/ui.py\", line 139, in iter\n    for x in it:\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/download.py\", line 560, in resp_read\n    decode_content=False):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py\", line 357, in stream\n    data = self.read(amt=amt, decode_content=decode_content)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py\", line 314, in read\n    data = self._fp.read(amt)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/filewrapper.py\", line 63, in read\n    self._close()\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/filewrapper.py\", line 50, in _close\n    self.__callback(self.__buf.getvalue())\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/controller.py\", line 275, in cache_response\n    self.serializer.dumps(request, response, body=body),\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/serialize.py\", line 87, in dumps\n    ).encode(\"utf8\"),\nMemoryError\n"}
> ==> ubuntu: 	to retry, use: --limit @/opt/iiab/iiab/iiab.retry
> ==> ubuntu: 
> ==> ubuntu: PLAY RECAP *********************************************************************
> ==> ubuntu: 127.0.0.1                  : ok=236  changed=160  unreachable=0    failed=1

holta, 2017-10-03:
@arky is this an intermittent download failure that might recover when running the following?
```
   cd /opt/iiab/iiab
   ./runtags kalite
```
(Does your CI system allow you any post-op intervention as such?)

arky, 2017-10-03:
@holta This looks like Python Pip cache failure. 

The download file is actually quite small. 

> $ pip2 install ka-lite-static==0.17.3
> Collecting ka-lite-static==0.17.3
>   Downloading ka_lite_static-0.17.3-py2-none-any.whl (40.7MB)
>     100% |████████████████████████████████| 40.7MB 33kB/s 
> Installing collected packages: ka-lite-static
> Successfully installed ka-lite-static-0.17.3
>  
Writing a patch for this.

georgejhunt, 2017-10-03:
> @georgejhunt <https://github.com/georgejhunt>, @jvonau
> <https://github.com/jvonau> & @germanrs <https://github.com/germanrs> can
> you clarify if you too do/don't face this issue, with IIAB 6.4 RC9
> <https://github.com/iiab/iiab/releases/> especially?

within the last hour, I had a successful VM built of local_vars_big.yml on
ubuntu

jvonau, 2017-10-03:
PR #373 a duplicate of commit https://github.com/jvonau/iiab/commit/f1c7ece62a62a75dbc9e42f42057668bc684311d in my test branch, [written](https://github.com/jvonau/iiab/commit/917c3c9c7af64573b6380fe7f9b35577caa64d98} while sorting out Ubuntu in the jen-mem testing branch. In my humble opinion, I think cherry-pick is your friend and the right thing to do when using someone else's ideas/changes. This is a temporary hack to workaround the current default memory allocation in jenkins. Once Jenkins' memory is bumped up the need for this patch goes away, but could be useful in other low memory situations, but George & I agreed that at this time it should be put on hold pending review/testing on real hardware. @georgejhunt Did you bump the memory in Jenkins?

holta, 2017-10-05:
Moved to IIAB 6.5 per agreement on call http://minutes.iiab.io

holta, 2017-10-05:
@georgejhunt is concerned this may be intermittent and hard to test.

jvonau, 2017-10-05:
Source of the original idea [Stack overflow: memory-error-while-using-pip-install-matplotlib](https://stackoverflow.com/questions/29466663/memory-error-while-using-pip-install-matplotlib)
Related Refs [pypa: Excessive memory use when caching large packages #2984](https://github.com/pypa/pip/issues/2984)

holta, 2017-10-09:
Unanimous to push #373 (done) even if we're not 100% sure this solves the problem &mdash; at least this moves the research forward.

@jvonau asks should the equivalent be done for OSM, which seems to manifest itself in the same way (let's publish that OSM Ansible failure here if poss!)

holta, 2017-10-13:
Resolving for now.

All feel free to re-open if this pattern flares up in a describable (even if intermittent!) way.

PR #373 (2017-10-09): Disable pip caching for ka_lite_static
Patch for downloading ka_lite_static failure. Fixes #372
commit 40a58925 (2017-10-09): Disable pip caching for ka_lite_static

Patch for downloading ka_lite_static failure. Fixes #372
closed 2017-10-13 by holta