iiab/iiab #375: when not first_run

georgejhunt, 2017-10-04:
The error in an email to xsce-devel titled Current master failing on Debian 9.1? appears to be from:
https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/detected_network.yml (first 15 lines or so).

This may have resulted from inadequately tested use of the first_run variable when it is in the False state.
I believe PR 363 would defer the problem by forcing the code to always take a path that is more tested. 

The other approach would be to engage in more test/fix/test, which at this point in the release cycle, makes no sense to me.

holta, 2017-10-04:
The 2 OS's that communities are actually asking for (Raspbian & Ubuntu) face no material issues to date with IIAB 6.4 RC9.

So let's not exaggerate the consequence of a small glitch on Debian (:

This can be fixed in coming weeks after 6.4 is released if the Debian community assists us here &mdash; and can even be retrofit into 6.4 if folks are hopefully willing to test such a narrow fix.

holta, 2017-10-04:
@tim-moody's issue pasted in from early today:

> Linux medbox-iiab 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u5 (2017-09-19) x86_6
> 
> 2017-10-04 12:50:34,291 p=10366 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "cat /etc/sysconfig/iiab_domain_name", "delta": "0:00:00.002688", "end": "2017-10-04 12:50:34.267529", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-10-04 12:50:34.264841", "stderr": "cat: /etc/sysconfig/iiab_domain_name: No such file or directory", "stderr_lines": ["cat: /etc/sysconfig/iiab_domain_name: No such file or directory"], "stdout": "", "stdout_lines": []}
> 
> current master
> 
> looks for /etc/sysconfig. don't think that exists on debian
> 
> intel deb 9

holta, 2017-10-04:
Aside: http://box/elgg has a glitch on Ubuntu is seems, but this is NOT a blocker preventing the release of IIAB 6.4 :)

jvonau, 2017-10-05:
This sounds like a tainted test subject, as in /etc/iiab/iiab.env exists on what should be a pristine machine. http://<hadron_ip:8080/ shows that Debian Ubuntu has no issues installing into an untouched by human hands environment using https://github.com/jvonau/iiab-smoketest/blob/jv/curl-me-debian as the base for the install. Note that factory is disabled as the same as it is on all the web-load scripts http://download.iiab.io/6.4/rpi/load-big.txt that are in general use now. Might want to do the same and join the mainstream development trend.

holta, 2017-10-05:
fyi @germanrs servicing schools in the Dominican Republic told me tonight that everything works great with IIAB 6.4 RC9 on Ubuntu, even Elgg 2.3.4, Caliber etc.

This is fantastic news, and truly speaks for many, built upon our quite rigorous testing on Raspbian and Ubuntu both.

Let's allow the release to move forward without delay as customers are now demanding, and permanently put our egos aside each+all to collaborate in earnest (Truly Their Is No "i" In Team).

(To repair Debian as best we can in the weeks immediately after release, even though it does not materially affect field communities at this time, it serves a strategic interest in the longer term.  So let's warmly welcome Debian contributors here who might help us make this field/testing/global coverage hopefully happen increasingly release-by-release for years to come!)

holta, 2017-10-05:
@germanrs was correct http://box/elgg works on IIAB 6.4 RC9 on Ubuntu 16.04 LTS: I simply needed to click "reload" and Elgg 2.3.4 came up for me too!

holta, 2017-10-05:
I've added to following Known Issue to our [IIAB 6.4 Release Notes](url):

> Debian 9.2 might require a patch in order for software installation to complete, [in certain conditions](https://github.com/iiab/iiab/issues/375). Watch this space in coming weeks and/or if you're a Debian pro please do [contact us](http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F) if you might help, Thanks!

holta, 2017-10-05:
["Do Everything from Scratch" installation instructions](https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch) have also been clarified:

> On Raspbian, Ubuntu or Debian (may need a TBA patch under certain conditions in [October 2017](https://github.com/iiab/iiab/issues/375)) doing everything from scratch involves a few simple steps [...]

holta, 2017-10-05:
During our live community/team call (http://minutes.iiab.io) we determined:
1) This is not an issue on Ubuntu and Raspbian, as confirmed with multiple runs of ".runansible"
2) "apt purge ansible" is sometimes necessary on Debian, to assure you're running Ansible 2.4.x (check using "ansible --version").  Then 2nd & subsequent runs of "cd /opt/iiab/iiab; ./runansible" have been confirmed by @floydianslips (JoshD) to work :)

closed 2017-10-05 by holta