iiab/iiab #377: Changing homepage through admin interface failing

kananigit, 2017-10-05:
Right now if you try to change the homepage from the admin interface the server breaks and you lose access to the homepage. I tried to change to wordpress severally but it broke each time. @jvonau  adviced that I do the change in local_vars.yml then do a runtags <wordpress>i nstead.  This worked for me but it would be nice if its fixed because it gets kinda frustrating if you don't know what is happening.

holta, 2017-10-05:
Thx Kanani!

Can you clarify if changing "iiab_home_url: /home" to "iiab_home_url: /wordpress" in [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) is what works for you?  (Followed by running "cd /opt/iiab/iiab; .runansible" of course :)

jvonau, 2017-10-05:
PR #378

kananigit, 2017-10-05:
I changed "iiab_home_url: /home" to "iiab_home_url: /wordpress" in local_vars.yml  before i did                 cd /opt/iiab/iiab; .runansible" in the first install and it worked for me.

holta, 2017-10-22:
@tim-moody do you have time to look into this week?

holta, 2017-10-22:
@tim-moody if possible review or improve upon #378 if this is the fix?

holta, 2017-10-24:
@kananigit can we resolve this now, preferably with your official testing of #378 merged into master yesterday morning?  Thanks!

holta, 2017-10-25:
@kananigit plz re-open if your issue's not resolved, Thanks!

closed 2017-10-25 by holta