iiab/iiab #379: Checking iiab_domain_name fails on stretch

tim-moody, 2017-10-06:
on stretch vps with current master

TASK [1-prep : Checking iiab_domain_name] *****************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "cat /etc/sysconfig/iiab_domain_name", "delta": "0:00:00.003359", "end": "2017-10-06 11:55:46.939076", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-10-06 11:55:46.935717", "stderr": "cat: /etc/sysconfig/iiab_domain_name: No such file or directory", "stderr_lines": ["cat: /etc/sysconfig/iiab_domain_name: No such file or directory"], "stdout": "", "stdout_lines": []}

holta, 2017-10-06:
Are you using Ansible 2.4.0 ?  We agreed on the call yesterday to put that into the "From Scratch" installation instructions which I did — sounds like the same #375 Debian story you're likely facing again?

holta, 2017-10-06:
Also be sure you're running on a clean OS, Debian 9.1 or 9.2 or whatever?

As JoshD (@floydianslips) helped us document yesterday, "apt purge ansible" is his recommendation to remove Ansible 2.2.1 from Debian.

tim-moody, 2017-10-06:
ansible was installed with scripts ansible. the os is clean

holta, 2017-10-06:
What does "ansible --version" say to triple-check?

georgejhunt, 2017-10-06:
I encountered this problem also, last night,  when the qemu emulator was
taking so long, and I wanted to go to bed.  I copied the image to an sd
card, and ran ansible to see the results.  The emulation was in the middle
of 3-base-server, and had not created the file (which is looked for in the
second clause below).

So I believe we have generated a more fragile playbook with the first_run
variable.
Snippet where Tim's error occurred:

- name: Checking iiab_domain_name
  shell: "cat /etc/sysconfig/iiab_domain_name"
  register: prior_domain
  when: not first_run

# above always registers
- name: Checking for prior domain name
  set_fact:
    iiab_domain: "{{ prior_domain.stdout }}"
  when: not first_run and prior_domain.stdout != "lan" and
prior_domain.stdout != ""

holta, 2017-10-06:
I'm unable to reproduce this on Raspbian or Ubuntu.

Can someone tell me exactly how to reproduce this, preferably following the instructions we agreed to during yesterday's call? (Ansible 2.4.0)  Or is this some kind of intermittent thing?

FWIW Step 5. of the ["Do Everything from Scratch" installation instructions](https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch) currently say:

> 5. Ansible 2.4.x is required on Debian and recommended on all OS's so that multiple runs of "./runansible" can complete: run command "ansible --version" to check and if necessary "apt purge ansible" to be sure Ansible 2.2.x is removed, prior to Ansible 2.4.x's installation below.

georgejhunt, 2017-10-06:
>    1. Ansible 2.4.x is required on Debian and recommended on all OS's so

I have not tried it. but I think the following will replicate the error:

On a clean install, runansible however you will. When you see the "TASK
[3-base-server]"  message, hit ctl-c, and runansible again.

jvonau, 2017-10-06:
Actually I can see this, ctl-c out of runansible before the network role is reached will produce this result as iiab.env is written in 1-prep-prep.yml but /etc/sysconfig/iiab_domain_name is written in network. I should learn how to argue stronger for my point of view, instead of giving in and going with the flow over backing off the config_vars nonsense. At any rate moving the creation of iiab.env to a point after the creation of /etc/sysconfig/iiab_domain_name, to take care of this hole in coverage should mitigate this, PR#381 and would fit into #369 should things go in that direction.

Like I have said in meetings the hostname handing can be be revised to handle the debian family better(currently needs reboot) as that was not really addressed with the port and the long standing 010-iiab.conf.j2 has ServerName {{ iiab_hostname }} that will never get reset from iiab-from-console.yml that is a carry over from past times.

tim-moody, 2017-10-06:
what if /etc/sysconfig/ doesn't exist?

this is debian 9, not raspbian, not ubuntu.

jvonau, 2017-10-06:
Gets created during 2-common/tasks/fl.yml

tim-moody, 2017-10-06:
but was already needed in TASK [1-prep : Checking iiab_domain_name]

jvonau, 2017-10-07:
That is why I opted to move writing config_vars originally and keying in on that file as the "I'm done marker", nothing references that file during the install but here we are..... Now from the voice discussion about having config_vars respected during runansible, Tim's expectation that config_vars should be included for consideration and is [documented](https://github.com/iiab/iiab/wiki/IIAB-Variables) as such but makes no mention of editing or removing the file breaks the user expectation of what is edited in local_vars will be used in during a re-install(runansible), which is the mantra of the day. Seeing as the admin-console no longer actually preforms any part of the initial install, as in the install-console days, but is now an add-on feature by virtue of it's new location maybe it's time let the command line tools function as command line tools.

holta, 2017-10-12:
TK Kang is currently facing something similar, with a failure to install IIAB 6.4 on RPi3 right after:
```
TASK [1-prep : Checking iiab_domain_name **************
```
He will try again with a new Raspbian Stretch OS from 2017-09-07.

holta, 2017-10-13:
@tim-moody please resolve or give us all a status report (how to reproduce this?) when possible!

tim-moody, 2017-10-13:
I have no ability to test this. If you think it is fixed, close it. It is possibly rectified by #381

holta, 2017-10-13:
@floydianslips to bring closure here, can you do a full smoke test of 6.5/master on Debian 9.2 (using http://download.iiab.io/6.5/rpi/load.txt or similar?)

holta, 2017-10-25:
Issue is resolved from what I can tell.

@tim-moody plz re-open if this can be reproduced.

closed 2017-10-25 by holta