iiab/iiab #383: IIAB Branding Identity

arky, 2017-10-08:
Add Internet-in-a-Box (IIAB) brand identity to all landing pages. This might include a logo and matching color palette.  The logo could be animated gif logo/motion graphic to convey the purpose of the project. 

The purpose of task is to increase brand recognition of IIAB.

holta, 2017-10-08:
Don't get frightened away!  But do also see:

IIAB logo - call for designs
http://iiab.io/t/iiab-logo-call-for-designs/32

holta, 2017-11-22:
> IIAB logo - call for designs
http://iiab.io/t/iiab-logo-call-for-designs/32

All logos were lost (as organized over many months by @darkency & others!) due to the very sad loss of our Discourse forums on the http://iiab.io server.

@m-anish is working to see if these can be recovered from backup/disk in Delhi, India.

Am deferring this tkt to IIAB 6.6 when we'll know a lot more.

holta, 2018-05-13:
Ref #789

arky, 2018-08-31:
Adding this as UI/UX task list for #1055

holta, 2019-11-01:
This is largely a duplicate of tkt #842 so I'm closing: let's consolidate the discussion in one place!

closed 2019-11-01 by holta