iiab/iiab #384: 2 obsolete checkboxes in Admin Console?

holta, 2017-10-09:
These 2 checkboxes appear in most all installs:
```
[ ] Services for XO Laptops such as Registration, Backup, and the Activity Server
[ ] Collect statistics on the use of Sugar Activities.
```
Are they at all useful/functional at this time?

Should we hide them and/or document their limitations for [IIAB 6.5](https://github.com/iiab/iiab/milestone/2)?

Especially given 5-xo-services was commented out of the current [iiab.yml](https://github.com/iiab/iiab/blob/master/iiab.yml) as these painfully slows down RPi installs especially:

#370 "comment out & document 5-xo-services as it's slow as a dog"
PR #371 "comment out 5-xo-services (too damn slow)"

holta, 2017-10-09:
@tim-moody wrote:
> Make sure they are install false, not just missing 

Each machine's https://github.com/iiab/iiab/blob/master/vars/default_vars.yml contains:
```
sugar_stats_install: False
sugar_stats_enabled: False
```
Each machine's http://wiki.laptop.org/go/IIAB/local_vars.yml contains:
```
# sugar_stats_install: False
# sugar_stats_enabled: False
```
Where else should I look to explain these 2 "zombie" checkboxes?

PS Everything else under 5-XO-SERVICES appears likewise (False in default_vars, commented out in local_vars).

tim-moody, 2017-10-09:
xo-services is due to the name change to xo_services. sugar_stats was also changed from sugar-stats. admin console needs this change as well.

tim-moody, 2017-10-09:
actually adm cons already has xo_services. please confirm  that this is an issue since you added xo_services_install: False.

holta, 2017-10-09:
[default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml) contains the following: (on all these stock IIAB 6.4 machines)
```
xo_services_enabled: False
```
"xo_services_install:" (True or False) is missing above.  Is this in any way related to the "Services for XO Laptops" checkbox appearing on all these stock 6.4 machines?

[local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) contains the following: (on all these stock IIAB 6.4 machines)
```
# xo_services_install: False
# xo_services_enabled: False
```

holta, 2017-10-10:
Ref https://github.com/iiab/iiab-admin-console/issues/35 "name changes (- to _)"

tim-moody, 2017-10-10:
put xo_services_install: False in default_vars

holta, 2017-10-11:
> put xo_services_install: False in default_vars

PR #389 does this...and makes [default_vars.yml](https://github.com/iiab/iiab/pull/389/files) (more) readable.

holta, 2017-10-11:
This appears solved thanks to PR #389.

@tim-moody please re-open as nec, Thx!

closed 2017-10-11 by holta

holta, 2017-10-24:
Further improved with https://github.com/iiab/iiab-admin-console/pull/36 ("sugar-stats to sugar_stats")

holta, 2017-10-24:
Related:
#458 "Kiwix checkbox not working well in Admin Console"
iiab/iiab-admin-console#38 "change 3 instances of kiwix_serve to kiwix to fix Kiwix checkbox?"