iiab/iiab #385: intermittent OSM Ansible failure, related to Cache-Control?

holta, 2017-10-09:
@jvonau asked me to post this IIAB-6.4-on-RPi failure from earlier this morning, in relation to the very similar kalite issue (#372) @arky reported:  

TASK [osm : Install IIAB with dependencies] ************************************
changed: [127.0.0.1] => (item=MarkupSafe)
changed: [127.0.0.1] => (item=pytz)
failed: [127.0.0.1] (item=Internet-in-a-Box) => {"changed": false, "cmd": "/usr/local/osm/bin/pip2 install --disable-pip-version-check Internet-in-a-Box", "failed": true, "item": "Internet-in-a-Box", "msg": "stdout: Collecting Internet-in-a-Box\n  Downloading Internet-in-a-Box-0.5.10.tar.gz (942kB)\nCollecting Flask>=0.10 (from Internet-in-a-Box)\n\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/commands/install.py\", line 353, in run\n    wb.build(autobuilding=True)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/wheel.py\", line 749, in build\n    self.requirement_set.prepare_files(self.finder)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 554, in _prepare_file\n    require_hashes\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/req/req_install.py\", line 278, in populate_link\n    self.link = finder.find_requirement(self, upgrade)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/index.py\", line 792, in get_page\n    \"Cache-Control\": \"max-age=600\",\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 501, in get\n    return self.request('GET', url, **kwargs)\n  File \"/usr/local/osm/local/lib/python2.7/site-packages/pip/download.py\", line 386, in request\n    return super(PipSession, self).request(method, url, *args, **kwargs)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 488, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 609, in send\n    r = adapter.send(request, **kwargs)\n  File \"/usr/local/osm/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py\", line 47, in send\n    resp = super(CacheControlAdapter, self).send(request, **kw)\n  File \"/usr/local/osm/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py\", line 423, in send\n    timeout=timeout\n  File \"/usr/local/osm/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 643, in urlopen\n    _stacktrace=sys.exc_info()[2])\n  File \"/usr/local/osm/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py\", line 315, in increment\n    total -= 1\nTypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n"}

arky, 2017-10-09:
Alright the " --no-cache-dir" flag is a work around. However we need to find the root cause of the 
cache failures.

Here are few questions to aid the investigation: 

_1. Is there a change in the location of pip cache ?
2. Is there a change in the pickle format of the pip cache?
3. Does CacheControl package store incomplete downloads? Can we clear these downloads before running pip?_

**Possible Solution: 
Does upgrade to 'cachecontrol' package fix this problem?** 
 
Here is the upstream bug in pip and cachecontrol packages: 

> https://github.com/pypa/pip/issues/2674

jvonau, 2017-10-09:
> Does upgrading 'cachecontrol' package fix this problem?

You are referring to inside the virtual environment correct? I'd prefer to limit any changes to just the virtual environment and not the underlying OS's python packages. I'd be inclined to think that the latest python version supplied by the OS might be 'too new' for what is now 4 year old code and iiab might want something a bit older(just a guess). The source for IIAB is on github if you want to review what is in the requirements file. However this appears to be intermittent and might be related to not being able to retrieve the a python package for whatever reason(timeouts, site issues), with a mishandled error masking what is really occurring. Gut feeling says proxy issue... @holta Did this occur now with big_vars in the mix, as that one has squid enabled, but I'm at a loss to make the connection as other installs work correctly.

holta, 2017-10-09:
>  Did this occur now with big_vars in the mix, as that one has squid enabled

No, this OSM error above occurred with a clean install of http://download.iiab.io/6.4/rpi/load.txt

(using [medium-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml))

arky, 2017-10-11:
@jvonau 

How about disabling CacheControl in ~/.pip/pip.conf ?

commit 00235c67 (2017-10-12): Merge pull request #400 from arky/bug-385

Disable pip caching for osm (Fixes #385)
commit 0d22dcb2 (2017-10-12): Disable pip caching for osm (Fixes #385)
closed 2017-10-12 by holta