iiab/iiab #391: [RFC] Pull Request Template

arky, 2017-10-11:
To further improve the contribution pathways, we need to add a Pull Request template in _docs/PULL_REQUEST_TEMPLATE_ or _.github/PULL_REQUEST_TEMPLATE_. It would be included when someone is reporting a bug. More information[ here](https://help.github.com/articles/creating-a-pull-request-template-for-your-repository/)

[image]

Please kindly add your comments to improve this template by adding comments in this bug report.

## Fixes Bug # 

## Description of changes proposed in this pull request. 

## Mention a team member for further information or comment using @ name

holta, 2017-10-14:
Self-reminder...such stuff can+should be set here: https://github.com/iiab/iiab/community

See also: https://help.github.com/articles/setting-guidelines-for-repository-contributors/

jvonau, 2017-11-12:
merged

closed 2017-11-12 by holta