iiab/iiab #392: urgent: kalite consistently blocks IIAB installs on RPi

holta, 2017-10-11:
Certainly this blocked 1 install last night and 2 this morning, so this is not an isolated/intermittent download failure as I initially assumed (it's probably blocking all IIAB 6.4 installs as well!)

Is this really as simple as a failure to download non-existent https://www.piwheels.hostedpi.com/simple/importlib/ ?  (last line in error msg below)

(Or possibly a python issue?  Hypothetically relating to "apt dist-upgrade" which (as of a few hours ago) applies 29 packages above & beyond Raspbian Lite from 2017-09-07)

TASK [kalite : Install ka-lite dependencies with pip] **************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/local/kalite/venv/bin/pip2 install --no-cache-dir -r /opt/iiab/pip-packages/kalite.txt", "failed": true, "msg": "stdout: New python executable in /usr/local/kalite/venv/bin/python2\nAlso creating executable in /usr/local/kalite/venv/bin/python\nInstalling setuptools, pkg_resources, pip, wheel...done.\nRunning virtualenv with interpreter /usr/bin/python2\nCollecting django==1.5.12 (from -r /opt/iiab/pip-packages/kalite.txt (line 9))\n  Downloading Django-1.5.12-py2.py3-none-any.whl (8.3MB)\nCollecting docopt<0.7,>=0.6 (from -r /opt/iiab/pip-packages/kalite.txt (line 11))\n  Downloading https://www.piwheels.hostedpi.com/simple/docopt/docopt-0.6.2-py2.py3-none-any.whl\nCollecting South==1.0.2 (from -r /opt/iiab/pip-packages/kalite.txt (line 12))\n  Downloading https://www.piwheels.hostedpi.com/simple/South/South-1.0.2-py2.py3-none-any.whl (136kB)\nCollecting django-js-reverse==0.5.0 (from -r /opt/iiab/pip-packages/kalite.txt (line 13))\n  Downloading django_js_reverse-0.5.0-py2.py3-none-any.whl\nCollecting ply==3.4 (from -r /opt/iiab/pip-packages/kalite.txt (line 15))\n  Downloading ply-3.4.tar.gz (138kB)\nCollecting slimit==0.8.1 (from -r /opt/iiab/pip-packages/kalite.txt (line 17))\n  Downloading slimit-0.8.1.zip (88kB)\nCollecting six==1.10.0 (from -r /opt/iiab/pip-packages/kalite.txt (line 18))\n  Downloading six-1.10.0-py2.py3-none-any.whl\nCollecting django-appconf==1.0.1 (from -r /opt/iiab/pip-packages/kalite.txt (line 19))\n  Downloading django_appconf-1.0.1-py2.py3-none-any.whl\nCollecting peewee==2.6.4 (from -r /opt/iiab/pip-packages/kalite.txt (line 20))\n  Downloading peewee-2.6.4.tar.gz (464kB)\nCollecting python-dateutil==2.4.2 (from -r /opt/iiab/pip-packages/kalite.txt (line 21))\n  Downloading python_dateutil-2.4.2-py2.py3-none-any.whl (188kB)\nCollecting django-annoying==0.8.4 (from -r /opt/iiab/pip-packages/kalite.txt (line 22))\n  Downloading django-annoying-0.8.4.tar.gz\nCollecting cherrypy==3.3.0 (from -r /opt/iiab/pip-packages/kalite.txt (line 23))\n  Downloading https://www.piwheels.hostedpi.com/simple/CherryPy/CherryPy-3.3.0-py2.py3-none-any.whl (448kB)\nCollecting ifcfg==0.10.1 (from -r /opt/iiab/pip-packages/kalite.txt (line 24))\n  Downloading ifcfg-0.10.1-py2-none-any.whl\nCollecting importlib==1.0.3 (from -r /opt/iiab/pip-packages/kalite.txt (line 25))\n\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/commands/install.py\", line 342, in run\n    requirement_set.prepare_files(finder)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/req/req_set.py\", line 554, in _prepare_file\n    require_hashes\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/req/req_install.py\", line 278, in populate_link\n    self.link = finder.find_requirement(self, upgrade)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/usr/local/kalite/venv/local/lib/python2.7/site-packages/pip/index.py\", line 795, in get_page\n    resp.raise_for_status()\n  File \"/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/models.py\", line 893, in raise_for_status\n    raise HTTPError(http_error_msg, response=self)\nHTTPError: 404 Client Error: Not Found for url: https://www.piwheels.hostedpi.com/simple/importlib/\n"}

PS temporary workaround is to set all 3 kalite lines to False within /opt/iiab/iiab/vars/local_vars.yml

holta, 2017-10-11:
I've reconfirmed this failure with a 4th attempt at a clean install on Raspbian Lite 2019-09-07 on RPi3 with the usual load script http://download.iiab.io/6.5/rpi/load.txt

Does anybody know if this might also affects Ubuntu 16.04 LTS, Debian 9.2 and/or CentOS 7.4?

holta, 2017-10-11:
#162 (venv for kalite) might also hypothetically be related?

holta, 2017-10-11:
As an experiment, "./runtags kalite" works to install a clean KA Lite 0.17.3 onto Ubuntu, after I deleted (actually moved) all of /library/ka-lite

So this failure (appears) to be isolate to Raspbian alone.

jvonau, 2017-10-11:
>  raise HTTPError(http_error_msg, response=self)\nHTTPError: 404 Client Error: Not Found for url: https://www.piwheels.hostedpi.com/simple/importlib/\n"} 

Seems like a new addition to the mix. Don't think virtual environments have anything to do with, this one a 404  The link https://www.piwheels.hostedpi.com/ looks to be for python3 with no importlib when viewing https://www.piwheels.hostedpi.com/simple/ 

Find and disable where that pip repo is enabled in raspbian or --no-use-wheel might be a workaround.

holta, 2017-10-11:
> Find and disable where that pip repo is enabled in raspbian or --no-use-wheel might be a workaround.

Huge thanks @jvonau.

I could use help from everyone understanding how to make either above happen.

jvonau, 2017-10-11:
If this is a resent change upstream, the package might be pulled in during scritps/ansible. Can you find/post the apt install log to this ticket, might shed some light. Disabling "apt-get update" might buy some time till the next raspbian release, but that is just a guess without knowing what is in the apt.log.

jvonau, 2017-10-11:
> cat  /etc/pip.conf
[global]
extra-index-url=https://www.piwheels.hostedpi.com/simple

holta, 2017-10-11:
Commenting out those 2 lines within /etc/pip.conf works to get "./runtags kalite" (as a short-term workaround anyway!)
```
#[global]
#extra-index-url=https://www.piwheels.hostedpi.com/simple
```
What's the best way to get this rolled out quickly, so IIAB 6.4 and 6.5/master installers can get back on their feet?

holta, 2017-10-11:
1. "apt install python-pip" within scripts/ansible is the root cause, creating the 2017-09-25 file /etc/pip.conf (presumably due to a new update that was published by Raspbian-or-others in recent days?)
```
# apt -a list python-pip
Listing... Done
python-pip/testing,now 9.0.1-2+rpt1 all [installed]
python-pip/stable 9.0.1-2 all
```
2. Placing "rm /etc/pip.conf" into load scripts like http://download.iiab.io/6.5/rpi/load.txt feels a bit too ruthless.  As does commenting out every line, using a sed script.
But any such ideas are possible, to installers back in motion, who are all DOA for the moment!

jvonau, 2017-10-11:
Don't be pointing to scripts/ansible as the cause it's just the first point where pip in mentioned, pip will get installed anyway later in the various roles. Which deb package repo that pip is being installed from is the source of this issue. I'd be looking to maybe 'fix' the deb version installed at 9.0.1-2 and not use 9.0.1-2+rpt1 but need to look at where all the other packages that are provided by raspbian come from and where/how there are defined. Any workaround may not survive an apt update unless that deb is excluded, but the question asked in chat was "Interesting that we're on the "testing" version not the stable!Sent on:10:27 am", and that is a good one, perhaps one for upstream Raspbian?

holta, 2017-10-11:
So that nobody is prevented from installing Internet-in-a-Box, this crude workaround was placed in the [IIAB 6.4](http://download.iiab.io/6.4/rpi/) & [IIAB 6.5/master](http://download.iiab.io/6.5/rpi/) install/load scripts -- right after Ansible 2.4.0 is installed and right before ./runansible is run:
```
   rm /etc/pip.conf
   touch /etc/pip.conf
   # TEMPORARY WORKAROUND to deal with https://github.com/iiab/iiab/issues/392
```
This certainly works -- a complete install of http://download.iiab.io/6.5/rpi/load-min.txt was tested to be 100% sure.

However let's try to nail a cleaner solution during Thursday's 10AM NYC Time call (http://minutes.iiab.io).

_Of course Raspbian & the publisher of python-pip might solve this for us at any moment, by cleaning up their package?!_

holta, 2017-10-17:
FWIW as we monitor the situation going forward in coming weeks &mdash; here are today's results, just to keep an eye on it &mdash; towards removing the "rm /etc/pip.conf; touch /etc/pip.conf" crude workaround in coming weeks hopefully:
1. "apt install python-pip" on a completely clean install of Raspbian Lite 2017-09-07 no longer installs /etc/pip.conf _(something changed for the better since last week, yay!)_
2. [scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) (and perhaps other parts of IIAB?) still do install the problematic /etc/pip.conf (possibly as a result of scripts/ansible using "deb http://ppa.launchpad.net/ansible/ansible/ubuntu xenial main" in /etc/apt/sources.list, and there may be good reasons for that...)

jvonau, 2017-10-17:
Point one, something changed upstream again(backed out of a change)? Point two, perhaps ansible's PPA is enabling a testing repo somewhere? Really one would need a `apt -a list python-pip` prior to and after adding ansible's PPA to get a better handle on why there is a difference, if there is a difference now as point one is where the altered /etc/pip.conf came from.

holta, 2017-10-17:
Oops maybe it's the same as last week after all.  After "apt install python-pip" on a virgin Raspbian, `apt -a list python-pip` shows:
```
python-pip/stable 9.0.1-2 all [installed]
```
BUT after "apt update; apt dist-upgrade" it shows the following:
```
python-pip/testing,now 9.0.1-2+rpt1 all [installed]
python-pip/stable 9.0.1-2 all
```

holta, 2017-10-18:
FYI after basic experimentation that it seems to work, I've shortened this temporary workaround within [IIAB 6.5's install/load scripts](http://download.iiab.io/6.5/rpi/) to:
```
rm /etc/pip.conf
# touch /etc/pip.conf
# TEMPORARY WORKAROUND to deal with https://github.com/iiab/iiab/issues/392
```

holta, 2017-10-26:
Here are some others experiencing the same serious issue:

> /etc/pip.conf is badly broken - Raspberry Pi Forums
> https://www.raspberrypi.org/forums/viewtopic.php?t=194886

jvonau, 2017-11-03:
upstream issue https://github.com/bennuttall/piwheels/issues/20

holta, 2017-11-03:
Great news via Ben Nuttall (also RPi Fdtn community manager) that the upstream bug appears fixed on Oct 27 after ~2 months: https://github.com/bennuttall/piwheels/issues/20#issuecomment-340108171

So this should hopefully be resolved in Raspbian itself in coming weeks.

bennuttall, 2017-11-06:
Sorry for the issue! It's fixed now and there's no need to comment out the pip config or `rm` the `pip.conf` file or anything.

When we soft-launched piwheels, we soon realised that there was an issue with pip replacing dots with hyphens when searching for dependencies, which messed up some installations, and we configured the web server to account for this, and the issue is now resolved.

For more info on this, see my explanation in https://github.com/learningequality/ka-lite/issues/5533#issuecomment-342021564

jvonau, 2017-11-06:
Thanks for the quick response @bennuttall  and as you explained in ka-lite tracker pywheels is python3 only. The install in question is setting up a python2 virtual environment and importlib is being pulled in but is not found(I knew why, it's not on the site) without the rm /etc/pip.conf workaround.  Any pointers would be helpful.

bennuttall, 2017-11-06:
piwheels currently only builds for Python 3, so `importlib` does not exist in our index (because we don't have a successful build of it). The 404 is the same as if a package merely didn't exist on PyPI, e.g. https://pypi.python.org/pypi/blahblahblah

The fact piwheels is missing this package should not make the slightest difference to anyone trying to install `importlib`. Either they're trying to install it for Python 3 and it will fall back to PyPI and fail, or they're trying to install it for Python 2, it will not be found on piwheels, but found on PyPI from where it will be downloaded and installed no problem.

Either piwheels has a wheel, and the user gets an install faster, or it doesn't, and the user gets the sdist from PyPI, as before.

I don't understand how the `pip.conf` is causing a problem, because pip should use it in *addition* to PyPI, not instead of.

Observe running `pip install importlib -v` in a Python 2 virtualenv (`-v` for verbose output). You'll see that pip discovers two indexes to search:

```
2 location(s) to search for versions of importlib:
  * https://pypi.python.org/simple/importlib/
  * https://www.piwheels.hostedpi.com/simple/importlib/
```

then it searches both indexes for available files, and finds several from PyPI and none from piwheels. It then determines the latest version from all files available, and downloads `importlib-1.0.4.zip`. When running in a Python 3 environment, the installation fails. Since piwheels is Python 3 only at present, we do not have a build of `importlib`, which is Python 2 only.

Similarly, installing `importlib2` (which is Python 3 compatible):

```
pip install importlib2 -v
```

It finds several files (source distributions) from PyPI and several files (bdist wheels) from piwheels. It chooses the latest wheel (`importlib2-3.5.0.2-py3-none-any.whl`) from piwheels as preference to sdist from PyPI. It installs no problem.

holta, 2017-11-08:
Very promising development from @bennuttall !

> Ok we discovered that there's a known bug in pip which breaks when additional indexes 404. The bug is still open in [ [pypa/pip#4605](https://github.com/pypa/pip/issues/4605) -> [pypa/pip#4580](https://github.com/pypa/pip/issues/4580) ], and we can't expect a fix to find its way into Debian any time soon.
>
> However, we (piwheels) have mitigated this by creating empty directories for packages we haven't successfully built, which means pip gets an empty list of files rather than a 404 (which is an enhandled exception causing the fatal error).
>
> Hopefully that means that this is no longer an issue here.

Above Excerpt from: https://github.com/learningequality/ka-lite/issues/5533#issuecomment-342822199

_And to confirm https://www.piwheels.hostedpi.com/simple/importlib/ currently does respond (unlike in recent weeks) showing the following that we'll test very soon:_
```
Links for importlib
```

bennuttall, 2017-11-08:
Sorry I meant to post here too - glad you found it anyway!

holta, 2017-11-08:
Thank you to @bennuttall & his piwheels team for creating stub site https://www.piwheels.hostedpi.com/simple/importlib/ which indeed solved the problem!

closed 2017-11-08 by holta

bennuttall, 2017-11-08:
Thanks @waveform80 :+1: