iiab/iiab #395: 2017-10-11 kiwix-tools to press testing coverage

holta, 2017-10-11:
Tomorrow I'd like to merge the new [2017-10-11 kiwix-tools](http://download.kiwix.org/nightly/2017-10-11/) into the [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) release cycle, so the following issues (@mgautierfr has been working hard on over the past 2 weeks) get more testing coverage:

* Byte range https://github.com/kiwix/kiwix-tools/pull/84 (test with /wikipedia_en_medicine_2017-09/skin/error.php ?)
* Systematic crash of kiwix-serve if page not available https://github.com/kiwix/kiwix-tools/issues/83
* Make kiwix-server multi-threaded https://github.com/kiwix/kiwix-tools/pull/82
* Use sudo to install pip dependencies in travis https://github.com/kiwix/kiwix-tools/pull/78
* Look for renamed kiwix-compile-resources script as well https://github.com/kiwix/kiwix-tools/pull/78
* adding custom url location https://github.com/kiwix/kiwix-tools/pull/77 (with @swiftugandan)

holta, 2017-10-19:
@tim-moody :
1) can you battle-test kiwix-tools with your favorite selections of the latest content in coming days?
2) or do you prefer we upgrade kiwix-tools to the very latest first, based on https://github.com/kiwix/kiwix-tools/commits/master ?

mgautierfr, 2017-10-20:
Please wait end of this week, release is coming :)

holta, 2017-10-21:
Thanks @mgautierfr !

Re-opening #395 here just so we can keep track, with #440 "Adding content to kiwix fail on iiab(master) and debian 9.2"

holta, 2017-10-25:
Obviated by #460

commit 352a0540 (2017-10-12): Merge pull request #396 from holta/master

update kiwix-tools 2017-09-28 to 2017-10-11

closes #395
closed 2017-10-25 by holta