iiab/iiab #401: Nextcloud logins/logouts remain unacceptably slow for some

holta, 2017-10-12:
On Thu, Oct 12, 2017 at 11:36 AM, T.K. Kang wrote:
> Took 1 minute to get login screen and 1 minute to login after password is typed in :-(  with [ http://box/nextcloud ]
>
> For [ http://box/elgg ] .. it is nearly 1 sec to load and login...
>
> with [ http://schoolserver/nextcloud ] the time to get login screen is normal. Login takes 1/2 minutes
>
> In the ... anything to do with name resolution?

Jerry thinks you might be onto something.

Rough Observations:

1. This Nextcloud nightmare (logins/logouts taking 1-to-2min each) goes away for me after the 1st login, in some conditions (like my home?)  Jerry's theory is that reverse DNS (or some such in my particular router) mitigates the worsts of this problem.
2. [Nextcloud 12.0.3](https://download.nextcloud.com/server/releases/) (from Sept 20th, that is currently installed as part of IIAB 6.4) appears better than [Nextcloud 12.0.2](https://download.nextcloud.com/server/releases/) (from Aug 14th).  This problem used to happen in my home as well in August.  It no longer does today.  Anyway something sped up dramatically in September and I'm not sure quite why.
3. Naturally even when the above problem is brought under control, Nextcloud still takes 15 seconds (or more) to login IF it hasn't been used by anybody in hours, as various jobs/databases/whatev need to we woken up and brought into RAM etc.

Recap: nobody is asking for perfection here.  Just a reality check &mdash; no teachers will use Nextcloud if logins regularly (or even intermittently) take 1-2 MIN.  Even 30 sec is unacceptably slow, when you only have your students attention for "30MIN AT BEST" during many lessons, in Haiti or any other country...

holta, 2017-10-12:
@tim-moody writes:
> my belief is that nextcloud does a lot of initialization on first access. we used to do some of that during the ansible run for owncloud, but may not be doing it for nextcloud. easy to verify by timing first and subsequent logins.

holta, 2017-10-13:
See #266 "Nextcloud logins painfully slow (1 to 2 min) on RPi3" even if we don't yet have the final word.

holta, 2017-10-13:
TK Kang asks us if we should be concerned about these warnings upon login to http://box/nextcloud/index.php/settings/admin ?

> Security & setup warnings
> 
> It's important for the security and performance of your instance that everything is configured correctly. To help you with that we are doing some automatic checks. Please see the Tips & Ticks section and the documentation for more information.
> 
> * **Your data directory and your files are probably accessible from the Internet. The .htaccess file is not working. It is strongly recommended that you configure your web server in a way that the data directory is no longer accessible or you move the data directory outside the web server document root.**
> 
> * **You are accessing this site via HTTP. We strongly suggest you configure your server to require using HTTPS instead as described in our [security tips](http://box/nextcloud/index.php/settings/admin/tips-tricks).**
> 
> * No memory cache has been configured. To enhance your performance please configure a memcache if available. Further information can be found in our [documentation](https://docs.nextcloud.com/server/12/go.php?to=admin-performance).
> 
> * The PHP OPcache is not properly configured. For better performance we recommend to use following settings in the php.ini:
> 
>     opcache.enable=1
>     opcache.enable_cli=1
>     opcache.interned_strings_buffer=8
>     opcache.max_accelerated_files=10000
>     opcache.memory_consumption=128
>     opcache.save_comments=1
>     opcache.revalidate_freq=1
> 
> Please double check the [installation guides ↗](https://docs.nextcloud.com/server/12/go.php?to=admin-install), and check for any errors or warnings in the log.

holta, 2017-10-13:
@jvonau asks if our ~5-year-old /etc/php.ini might exacerbate this?  See:

* #69 "use of /etc/php.ini" we carefully researched in late June / early July 2017
* Nextcloud 12 configuration requirements: https://docs.nextcloud.com/server/12/admin_manual/installation/source_installation.html#php-ini-tips-label

holta, 2017-10-14:
TK Kang discovered that his Nextcloud logins sped up after about 24h after IIAB's install.

Good to know.  The deeper quest to explain this continues...

holta, 2017-10-18:
TK hopes to try NEXTBERRY for comparison, so we all persist together trying to solve this:

https://www.techandme.se/nextberry-rpi/

jvonau, 2017-10-18:
code is on github https://github.com/techandme/NextBerry

holta, 2017-10-20:
More from TK:

> [am also trying NextCloudPi which] has official backing:

https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/

> I took the NextcloudPi for a drive and I must say it is SLICK.
> Downloaded the image *.bz2 image which as about 820M. Unzip and the sd image was about 2.9G.
> Using Etcher I could transer the image to my mini SD in a few minutes.
> 
> Insert into the Rpi3 and it boot as usual. Login with pi will get you a logo and I could then do a sudo raspi-config for config. It has a NextCloudPi option for configuration.
> 
> I manage to get into Nectcloud in my network with http://192.168.2.100.  First login time is acceptable and subsequent login is fast.
> 
> http://192.168.2.100:4443 and I could get into a web admin. I chose to upgrade and did a static address config to 192.168.2.188. Reboot and all work well with http://192.168.2.188.
> 
> Now I want to try the headless install to see if I could put a SSH file in the boot on a virgin install as by default now RPI3 has ssh switched off by default.

holta, 2017-10-23:
During our community/team call (http://minutes.iiab.io) @jvonau suspects a re-installed + correctly configured resolvconf might help too!

holta, 2017-11-23:
I don't see any hope for completely fixing this for IIAB 6.5 so am deferring to IIAB 6.6 when Next loud 13.0.x will give us another shot.

holta, 2018-04-25:
See merged PR's from Dec 4 & Feb 8:
PR #591 Nextcloud 12 playbook cleaner: download findable & explicit reinstall/upgrade
PR #664 Nextcloud 12.x -> 13.x

arky, 2018-07-08:
Quickly looked into this issue. It turns out to problem with database (mariadb) performance issue. Traditionally databases tend to be IO, Memory resource hogs.  

Can someone problem some additional performance information using any of the tools mentioned in the link and also logs for further analysis.  

https://www.tecmint.com/mysql-performance-monitoring/

holta, 2018-11-03:
Let's declare victory.  Nextcloud 14.03 logins works tolerably well even on Raspberry Pi today :)

closed 2018-11-03 by holta