iiab/iiab #409: Travis CI fails b/c whitespace etc? How to fix/automate?

holta, 2017-10-13:
Thanks @Arky if you have time to help us find the best code linting/cleaningscripts in the coming week?

So https://travis-ci.org/iiab/iiab becomes more directly actionable for all!  e.g. please all help deepen this list of Travis Tuning ideas we might talk about during our Monday Oct 16 community/team call: (http://minutes.iiab.io)

a. https://docs.travis-ci.com/user/notifications/ & other customizable/impactful options?
b. How can whitespace & similar lint warning/fails be overcome efficiently, in an automated way?
c. Is Frank Lamar available to meet with us in the coming days, to review all CI/CD strategy+tactics?

arky, 2017-10-14:
@holta I have fixed most of the whitespace problems in the codebase. A pull request is on the way after the CI tests completes. 

I think a using a code editor (atom.io, vi, emacs) with yaml linting enabled is best way of preventing these kind of issues in the future.

jvonau, 2017-10-14:
Being early adopters of ansible, all the code started out being written in the pre-ansible-1.4(wow how fast 5-year passes) time frame when there were no best practices guidelines, ansible-lint, ansible-review(available only via pip right now) and without really being reviewed other than for works/not works over time. New features have been introduced into ansible that have not been been taken advantage of, the old it works why mess with things for fear of breakage have been the general rule.

holta, 2017-10-14:
Ref #411 "Why don't we use Ansible handlers ?"

jvonau, 2017-10-14:
> WARN: Best practice "YAML should be correctly indented" not met:
roles/1-prep/tasks/computed_vars.yml:9: indentation should increase by 2 chars

For the most part those types of warning for indents during set_facts, the code matches what is shown as examples on http://docs.ansible.com/ansible/latest/set_fact_module.html. I have to ask the question, are the examples incorrect on the website?

arky, 2017-10-20:
@jvonau The author of the ansible-review has some thoughts on Ansible YAML formatting errors. Patches are welcome. :) 

https://github.com/willthames/ansible-review/issues/15

holta, 2017-11-21:
@arky & @jvonau there is MASSIVE progress here in recent days/months, thank you:

#550 

Should we close this tkt now?

closed 2018-06-19 by holta