iiab/iiab #411: Encourage using Ansible handlers

arky, 2017-10-14:
This is coding convention/issue that has been surfaced by ansible CI code review tool. 

Perhaps we could start using ansible handlers in our code. 

What is a Ansible handler? 
> Handlers are just like regular tasks in an Ansible playbook but are only run if the Task contains a notify directive and also indicates that it changed something. For example, if a config file is changed, then the task referencing the config file templating operation may notify a service restart > handler.

Here is an example where such handlers could be used. 

```
iiab/roles/network/tasks/rpi_debian.yml:74: [EANSIBLE0016] Tasks that run when changed should likely be handlers

- name: dhcpd may be affected
  service: name=bind9 state=started
  when: interface.changed
```

tim-moody, 2017-10-14:
handlers were tried in the past and didn't seem to work as advertised, but that was a few ansible versions back. the most obvious use would be for things like mysql that are dependencies for more than one role, but the roles usually need them done in advance in order to perform configuration tasks rather than at the end.

jvonau, 2017-10-14:
Recalling looking at how the handlers are used within ansible, handlers run at the end of the playbook(or was that at the end of a role, little fuzzy) after everything else is configured. Some roles lend themselves really well to using a handler, while others notability network need to be broken up and rearranged to be able to use a handler due to timing of the sequence of events. Not against the idea but sense resistance might be high, and time consuming affair.

holta, 2017-11-21:
Deferring this design discussion to to IIAB 6.6

arky, 2018-07-01:
Closing this as Travis CI is no longer used.

closed 2018-07-01 by arky