iiab/iiab #412: Captive Portal!

holta, 2017-10-14:
New users (rich and poor, literate and non-literate) often struggle to type in http://box and http://box.lan into their phones, even when they do know a Latin-based alphabet.

And increasingly many do not know a Latin-based alphabet, as new countries and communities get involved, outside of major cities etc!

It is possible to force client machines' browsers to open a page automatically in support of new users — however each OS has its own quirks.

@tim-moody, @m-anish and @georgejhunt have done extensive research here since 2016: _please do contact them if you can help._

jvonau, 2017-10-14:
How about a look at what code that is hidden from view that exists now contains? Smells like a hidden fork for profit of the few otherwise.

holta, 2017-10-14:
@jvonau my understanding (unfortunately extremely limited!) is that these diverse code efforts, while all over the map, are all public in place likes https://coova.github.io

Some of the original "DNS jail" work is part of other Kiwix-connected and RACHEL related projects if I recall correctly?

But @tim-moody can share much more during our upcoming community calls (http:/minutes.iiab.io) as he's gathered the most comprehensive overall view of all these diverse community efforts last I checked, and possibly @georgejhunt too.

Solving this browser/human literacy issue which is much more complex that it appears on the surface (hence the reason this struggle has gone on+off repeatedly for about 2 years now) given that browser tricks+popups used by Chrome, Microsoft, Molliza, iOS, Android itself (etc) are not only very subtle, often undocumented, but also ever-evolving...

@darkenvy & @floydianslips have also done some basic R&D here too, along these lines?
https://github.com/nodogsplash/nodogsplash (and the related NoCatSplash)

Also Ivan Savov @ Learning Equality (KA Lite) had a suggestion for us after our August http://OFF.NETWORK hackathon too:
https://github.com/learningequality/kolibri-sandisk-connect/blob/master/ansible/roles/system/files/dnsd
https://github.com/learningequality/kolibri-sandisk-connect/blob/master/ansible/roles/system/files/resolve_all_dns_to_host_ip.py

_PS there's also an opposing view that this entire "Captive Portal" effort may be way more trouble than it's worth (?) when a giant sticker (http://BOX.LAN) pasted on these handheld Internet-in-a-Box devices solves ~99% of the problem nearly instantly..._

slaporte, 2017-11-16:
I think a captive portal that directs users to the main menu is a great idea. 👍  It will make it easier for users to get to information quickly. 

I see this has been researched/discussed extensively already, but wanted to add my +1 for the feature generally.

holta, 2017-11-23:
Question for @deldesir & all: is https://github.com/stemid/captiveportal a good approach?

slaporte, 2017-11-23:
@holta Do you want people to log in through captive portal before accessing anything, or is it enough just to load the menu when they connect? 

If you just want to load the menu, I think you may be able to do it with something simpler than stemid/captiveportal. Simpler is usually better, in my view!

tim-moody, 2017-11-23:
iptables -t nat -A PREROUTING -i br0 -p tcp -j DNAT --to-destination 172.18.96.1 is simple, but not perfect in that the device internet test url remains at the top of the browser and it takes the home page to be the network signin page. I agree the goal is to find something that makes a full transition to the home page as simply as possible.

holta, 2017-11-25:
Small Progress Report from @tim-moody : [nodogsplash](https://github.com/nodogsplash/nodogsplash) compiles on Raspbian Stretch, unlike Raspbian/Debian Jessie earlier this year!

holta, 2017-12-07:
Captive Portal Work is ongoing as we transition from [IIAB 6.5](https://github.com/iiab/iiab/milestone/2) to [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) &mdash; see @tim-moody PR #597 evolving prototype now merged!

holta, 2017-12-07:
IETF has a new Captive Portal working group since Mar/Sep 2017, which (possibly) bodes well for an open-standards solution here in future years &mdash; thanks @tim-moody for finding this: https://datatracker.ietf.org/doc/draft-ietf-capport-architecture/

tim-moody, 2017-12-07:
Most captive portals follow the IETF definition:

 Captive Network: A network which limits communication of attached
   devices to restricted hosts until the user has satisfied Captive
   Portal Conditions, after which access is permitted to a wider set of
   hosts (typically the internet).

Unfortunately this is not our use case exactly in that we want to direct to a home page whether the user has 'signed in' or not.

The gist is to add ICMP messages to indicate a captive portal situation.

holta, 2018-07-06:
@m-anish, @jvonau, @tim-moody working on #826 -> PR #870 which is a promising approach based on https://github.com/nikosft/captive-portal/blob/master/captive_portal.py

holta, 2018-10-07:
It's taken 2 years, but Captive Portal PRs #1179 #1180 have now been merged.

_Broad community testing_ is now needed to refine the UX across as many phones/tablets/laptops as possible — collating this fieldback at #1182

holta, 2019-01-04:
Please move the design/feedback to #1182 now that our (largely working) Captive Portal product is entering circulation, Thanks!

closed 2019-01-04 by holta