iiab/iiab #417: Hardcoded path in iiab.yml

arky, 2017-10-17:
The global iiab.yml uses hardcoded path /etc/iiab/config_vars.yml. I found that this practice makes it difficult to implement Travis CI variable and syntax checks. 

https://github.com/iiab/iiab/blob/master/iiab.yml#L9

>$ ansible-playbook --check iiab.yml -i tests/inventory
ERROR! vars file /etc/iiab/config_vars.yml was not found

tim-moody, 2017-10-17:
all the var paths are hardcoded. the problem is not the path, but the fact that the file is created by the script so doesn't exist initially. same for local_vars.yml.

arky, 2017-10-18:
@tim-moody Am no expert, but it seems to me this needs to be refactored using Ansible best practices. 

 https://github.com/iiab/iiab/blob/2203107f85cd72f890cc6ce46560b1c5966a0e88/runansible#L3-L6

Perhaps putting these files under  `playbook_dir` (it contains the playbook base directory) make the code more portable and Ansible standards compliant than putting them under /etc/ directory.

jvonau, 2017-10-18:
Ref #362, not only hard-coded paths there is machine specific local_facts detection occurring. Maybe have an iiab-test.yml that omits /etc/iiab/config_vars.yml and substitutes an arbitrary os but otherwise is the same as iiab.yml for the testing for syntax checks?

arky, 2017-10-20:
@tim-moody @jvonau 

I have implemented all the functionality of the runansible script using few pre_tasks in my test.yml.  Thanks @geerlingguy !

```
 pre_tasks:
    - name: "Create /etc/iiab directory"
      file:
        path: "/etc/iiab"
        state: "directory"

    - name: "Create custom fact directory"
      file:
        path: "/etc/ansible/facts.d"
        state: "directory"
    - name: "Copy Medium.localvars"
      copy:
        src: ../vars/medium.localvars
        dest: ../vars/local_vars.yml
        mode: 0755
    - name: "Insert custom fact file"
      copy:
        src: ../scripts/local_facts.fact
        dest: /etc/ansible/facts.d/local_facts.fact
        mode: 0755

````

holta, 2017-11-21:
@arky if you can explain this proposal during a live call for IIAB 6.6 in coming weeks, that'd greatly help!

holta, 2018-06-19:
Installation of IIAB 6.5+ has changed a lot since 8 months ago.

It now happens with script ./iiab-install that depends on iiab-stages.yml ( all of which remain in /opt/iiab/iiab ).

Please re-open this ticket as nec -

closed 2018-06-19 by holta