iiab/iiab #418: Sugarizer failed @ "Set up apt sources on is_debuntu"

holta, 2017-10-17:
Is this yet another download failure that blocked installation while running http://download.iiab.io/6.5/rpi/load-big.txt on Raspbian Lite?  Seems so as the problem went away when running "./runtags sugarizer"

>TASK [sugarizer : Set up apt sources on is_debuntu] ****************************
 [WARNING]: Consider using get_url or uri module rather than running curl
>
>fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "curl -sL https://deb.nodesource.com/setup_6.x | bash -", "delta": "0:00:21.732492", "end": "2017-10-17 00:33:25.192321", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-10-17 00:33:03.459829", "stderr": "Warning: apt-key output should not be parsed (stdout is not a terminal)\ngpg: can't connect to the agent: IPC connect call failed", "stderr_lines": ["Warning: apt-key output should not be parsed (stdout is not a terminal)", "gpg: can't connect to the agent: IPC connect call failed"], "stdout": "\n## Installing the NodeSource Node.js v6.x repo...\n\n\n## Populating apt-get cache...\n\n+ apt-get update\nHit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease\nHit:2 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial InRelease\nHit:3 http://archive.raspberrypi.org/debian stretch InRelease\nIgn:4 http://linux.teamviewer.com/deb stable InRelease\nIgn:5 http://linux.teamviewer.com/deb preview InRelease\nHit:6 http://linux.teamviewer.com/deb stable Release\nHit:7 http://linux.teamviewer.com/deb preview Release\nReading package lists...\n\n## Confirming \"stretch\" is supported...\n\n+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_6.x/dists/stretch/Release'\n\n## Adding the NodeSource signing key to your keyring...\n\n+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -\nError executing command, exiting", "stdout_lines": ["", "## Installing the NodeSource Node.js v6.x repo...", "", "", "## Populating apt-get cache...", "", "+ apt-get update", "Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease", "Hit:2 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial InRelease", "Hit:3 http://archive.raspberrypi.org/debian stretch InRelease", "Ign:4 http://linux.teamviewer.com/deb stable InRelease", "Ign:5 http://linux.teamviewer.com/deb preview InRelease", "Hit:6 http://linux.teamviewer.com/deb stable Release", "Hit:7 http://linux.teamviewer.com/deb preview Release", "Reading package lists...", "", "## Confirming \"stretch\" is supported...", "", "+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_6.x/dists/stretch/Release'", "", "## Adding the NodeSource signing key to your keyring...", "", "+ curl -s https://deb.nodesource.com/gpgkey/nodesource.gpg.key | apt-key add -", "Error executing command, exiting"]}

Ref #193 (umbrella Sugarizer 0.8 -> 0.9 ticket covering many associated issues).

jvonau, 2017-10-18:
https://deb.nodesource.com/setup_6.x contains:
> echo 'deb https://deb.nodesource.com/${NODEREPO} ${DISTRO} main' > /etc/apt/sources.list.d/nodesource.list
echo 'deb-src https://deb.nodesource.com/${NODEREPO} ${DISTRO} main' >> /etc/apt/sources.list.d/nodesource.list

[nodesource.list.txt](https://github.com/iiab/iiab/files/1395839/nodesource.list.txt) is the file contents as installed by setup_6.x

holta, 2017-10-22:
Am resolving as this has not re-occured in over a dozen subsequent clean installs of IIAB on Raspbian Lite and Raspbian Desktop.  Plz re-open as nec.

closed 2017-10-22 by holta