iiab/iiab #421: why does Kiwix install when local_vars.yml asks it not to?

holta, 2017-10-18:
Why does Kiwix install itself into /opt/iiab/downloads and /opt/iiab/kiwix/bin etc even when I put the following in [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) ?
```
kiwix_serve_install: False
kiwix_serve_enabled: False
```

holta, 2017-10-18:
Just to confirm http://box:3000 does indeed work.

_Even when I wish it did not here &mdash; we all love Kiwix but rapid installs in diverse scenarios are very important too !!_

tim-moody, 2017-10-18:
because https://github.com/iiab/iiab/blob/master/vars/default_vars.yml has kiwix_install: True and the variables are not consistent

tim-moody, 2017-10-18:
the standard is <role>_install and enabled (see kalite)
(enabled should have been enable)

closed 2017-10-18 by holta