iiab/iiab #423: installs cannot be restarted if interrupted [1-prep : Checking iiab_domain_name]

holta, 2017-10-18:
Let's try to fix this as part of [IIAB 6.5](url) as this annoyance has affected basically everyone, given that interruptions of lengthy IIAB installations are realistically a fact of life :-)

Example error:

> TASK [1-prep : Checking iiab_domain_name] **************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "cat /etc/sysconfig/iiab_domain_name", "delta": "0:00:00.009257", "end": "2017-10-17 22:19:51.789293", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-10-17 22:19:51.780036", "stderr": "cat: /etc/sysconfig/iiab_domain_name: No such file or directory", "stderr_lines": ["cat: /etc/sysconfig/iiab_domain_name: No such file or directory"], "stdout": "", "stdout_lines": []}

jvonau, 2017-10-18:
#381 is the quick fix with further food for thought in #424 and #369

holta, 2017-10-19:
This blocked another installation for me, like yesterday.  Same Pattern: some playbook's download/mirror failure (e.g. it was [httpd : Install httpd required packages] this time) interrupts the installation of IIAB &mdash; which leads you to rerun ./runansible &mdash; but disturbingly that cannot proceed due to [1-prep : Checking iiab_domain_name].

Ref #379

holta, 2017-10-20:
All evidence so far indicates @jvonau's #381 has fixed this issue.  Thanks all for further testing during Ansible's early stages like:

1-PREP
2-COMMON
3-BASE
4-SERVER-OPTIONS

closed 2017-10-20 by holta