iiab/iiab #430: Enable simultaneous Hotspot/AP & client-WiFi on RPi's internal WiFi?

holta, 2017-10-19:
If achievable, this would be particularly useful during IIAB installation/curation/testing.

And potentially also during field updates of Content Packs &mdash; though Wi-Fi access to the Internet can be quite rare within many rural/developing countries, where pricey 3G mobile data is usually far more common (though efforts like Google Station are seeking to change this...)

Simultaneous Hotspot/AP and WiFi-as-client could build upon #197 "Permit installation of IIAB over Wi-Fi instead of Ethernet".

tim-moody, 2017-10-19:
http://imti.co/post/145442415333/raspberry-pi-3-wifi-station-ap

holta, 2017-12-06:
> http://imti.co/post/145442415333/raspberry-pi-3-wifi-station-ap

@jvonau adds this interesting read from Oct 27 & onwards:

> How-to: AP+wifi client on Zero W with Stretch Lite
https://www.raspberrypi.org/forums/viewtopic.php?t=196263

jvonau, 2017-12-06:
iw list for .26 
> $ iw list
Wiphy phy0
	max # scan SSIDs: 10
	max scan IEs length: 2048 bytes
	max # sched scan SSIDs: 16
	max # match sets: 16
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
		 * P2P-device
	Band 1:
		Capabilities: 0x1020
			HT20
			Static SM Power Save
			RX HT20 SGI
			No RX STBC
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (disabled)
			* 2472 MHz [13] (disabled)
			* 2484 MHz [14] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * tdls_oper
		 * start_sched_scan
		 * start_p2p_device
		 * crit_protocol_start
		 * crit_protocol_stop
		 * connect
		 * disconnect
	Supported TX frame types:
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * managed: 0x40 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
	valid interface combinations:
		 * #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1,
		   total <= 3, #channels <= 1
		 * #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1,
		   total <= 4, #channels <= 1
	Device supports scan flush.

https://superuser.com/questions/1148811/adding-virtual-interface-to-wlan0-with-iw-device-or-resource-busy-16 suggests that dual mode support requires something like 
valid interface combinations:
#{ managed, AP }  (like the  #{P2P-client, P2P-GO above})
is needed in iw list's output while https://wiki.archlinux.org/index.php/Software_access_point#Wireless_client_and_software_AP_with_a_single_Wi-Fi_device suggests messing with virtual interfaces and mac addresses to make this work. BTW channel 14 is forbidden in North America http://www.wirelesshack.org/router-channel-14-the-banned-channel.html

holta, 2017-12-07:
@tim-moody suggests putting a version of #543 (@jvonau's reduced version of @georgejhunt's #431) into https://github.com/iiab/iiab-factory/tree/master/box/rpi for people to try &mdash; e.g. the script currently proposed for /opt/iiab/iiab/scripts/rpi-wifi-gw

@jvonau is confident the hostapd part will work, but there are concerns "faking" 2 virtual devices might require MAC address problems, or wrong channels etc...

holta, 2019-03-01:
These 2 PR's tried to achieve "dual WiFi" but QA was never achieved (they were never merged) possibly due to flaky WiFi firmware within RPi:
- PR #697 by @georgejhunt "Wifi3 - simultaneous AP & WiFi-as-client on RPi (using dnsmasq)"
- PR #748 by @jvonau "rpi-test: wpa_supplicant AP mode simultaneous with WiFi as client"

holta, 2019-04-04:
> These 2 PR's tried to achieve "dual WiFi" but QA was never achieved (they were never merged) possibly due to flaky WiFi firmware within RPi:
> 
> * PR #697 by @georgejhunt "Wifi3 - simultaneous AP & WiFi-as-client on RPi (using dnsmasq)"

@georgejhunt just closed the above.

Should this tkt or any below also be closed?

(Conversely we should keep things open if we/others will continue to improve on and/or learn from these.)

> * PR #748 by @jvonau "rpi-test: wpa_supplicant AP mode simultaneous with WiFi as client"

See also PR #1114 by @jvonau "Method to create ap0 and have hostapd use it [for simultaneous up AND down WiFi]"

holta, 2019-04-05:
Also under consideration for closure:

#431 by @georgejhunt "wifi as both upstream and downstream"
#543 by @jvonau "gh-rpi-wifi-gw2: WiFi as both upstream & downstream"

jvonau, 2020-02-27:
#2266

jvonau, 2020-04-29:
#2337

holta, 2020-05-02:
Implemented experimentally by PR #2380 on Raspberry Pi and many regular PC laptops too!

holta, 2020-05-02:
> Implemented experimentally by PR #2380 on Raspberry Pi and many regular PC laptops too!

DRAFT instructions for newcomer IIAB implementers just getting started...are here:

https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#whats-new

Related:
- #796 "Rapid networking & other IIAB config changes that can't wait ~20min for Ansible"
- #2096 "NGINX & ./runrole & ./iiab-network [Dual WiFi] & IIAB Maps need approachable docs for newbie/legacy implementers"
- #2379 "Explain how to connect IIAB's internal Wi-Fi to the Internet: how to enter SSID & password"

closed 2020-05-02 by holta