iiab/iiab #432: "./runtags kiwix" install fails every time during WiFi install

holta, 2017-10-19:
Current master fails consistently when installing over WiFi (instead of Ethernet) on Raspbian Desktop 2017-09-07 on RPi3 using http://download.iiab.io/6.5/rpi/load.txt

> TASK [kiwix : Get the kiwix software] ******************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: timeout: timed out
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "failed to create temporary content file: timed out"}

Clarif: this happens even when "ping mit.edu" and DNS work.

holta, 2017-10-19:
Lacking any other ideas, I rebooted the RPi3:

1st Reboot Results: it's no longer possible to select a WiFi connection from the top-right of Raspbian Desktop.  So I inserted a live Ethernet cable to try to restore connectivity.  With no luck on either front, I ran "reboot" (with the live Ethernet cable inserted) which took several minutes to power down.

2nd Reboot Results: still no WiFi, nor Ethernet &mdash; eth0 shows "169.254.144.110/16" when you type "ip a".

Hopefully @jvonau or others have ideas!

jvonau, 2017-10-19:
> 2nd Reboot Results: still no WiFi, nor Ethernet — eth0 shows "

Glad the step was taken to chart the waters with wifi as upstream. Quick
suggestion, leave the ethernet cable attached and run ./iiab-network, just
to see if one can change the network from the command line without other
tools.

jvonau, 2017-10-19:
Can you retrieve the /etc/network.interfaces.d/iiab for 432 and attach it to the issue please.... first example of what the code did while wifi was active as gateway... maybe systemctl status dhcpcd systemctl status networking if you can would be helpful.. Shoot, should of thought of that before the above...  Think deleting /etc/network.interfaces.d/iiab and a reboot should put the rpi back online via wifi, or otherwise not interfere with a wired connection.

holta, 2017-10-20:
@jvonau asked me to paste in /etc/network/interfaces.d/iiab from this RPi3 running Raspbian Desktop 2017-09-07 (with apt updates) so here it is:
```
# Generated by IIAB
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# iiab_network_mode is Gateway

#################   GATEWAY   #########################
auto br0
iface br0 inet static
    bridge_ports   eth0
    bridge_maxwait 0
    address 172.18.96.1
    netmask 255.255.224.0
    dns-nameservers 172.18.96.1
# change for debian 9, let dhcpcd.conf set the static ip
#  if gui_static_wan == True
auto wlan0
iface wlan0 inet manual
```
Ethernet (but not WiFi) was restored by moving this file (to /root for now) and then rebooting, at which point /etc/network/interfaces.d/iiab appeared as follows:
```
# Generated by IIAB
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

# iiab_network_mode is Gateway

auto wlan0
iface wlan0 inet manual

#################   GATEWAY   #########################
auto br0
iface br0 inet static
    bridge_ports  wlan0
    bridge_maxwait 0
    address 172.18.96.1
    netmask 255.255.224.0
    dns-nameservers 172.18.96.1
# change for debian 9, let dhcpcd.conf set the static ip
#  if gui_static_wan == True
auto eth0
iface eth0 inet manual
```
Perhaps I have removed the Ethernet cable prior to rebooting?

In any case, I then continued with another ./runansible (using Ethernet instead of via WiFi) which succeeded in getting past the Kiwix section obviously, even if failing at Sugarizer with the following:

> TASK [sugarizer : Create the express framework for node.js - ALL less F18] *****
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "npm install", "delta": "0:00:00.009218", "end": "2017-10-19 21:08:50.205256", "failed": true, "msg": "non-zero return code", "rc": 127, "start": "2017-10-19 21:08:50.196038", "stderr": "/bin/bash: npm: command not found", "stderr_lines": ["/bin/bash: npm: command not found"], "stdout": "", "stdout_lines": []}

Finally, the results "systemctl status dhcpcd" contained (and still do contain) the worrisome line **if_addaddress6: Permission denied** as seen here:
```
● dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: e
nabled)
   Active: active (running) since Thu 2017-10-19 17:58:35 EDT; 2h 20min ago
  Process: 398 ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -b (code=exited, status=0/SU
CCESS)
 Main PID: 417 (dhcpcd)
   CGroup: /system.slice/dhcpcd.service
           └─417 /sbin/dhcpcd -q -b

Oct 19 17:58:37 box.lan dhcpcd[417]: DUID 00:01:00:01:21:44:2c:4f:b8:27:eb:94:20
:79
Oct 19 17:58:37 box.lan dhcpcd[417]: eth0: IAID eb:c1:75:2c
Oct 19 17:58:37 box.lan dhcpcd[417]: eth0: adding address fe80::dca5:1ff:e46d:13
66
Oct 19 17:58:37 box.lan dhcpcd[417]: if_addaddress6: Permission denied
Oct 19 17:58:37 box.lan dhcpcd[417]: eth0: soliciting an IPv6 router
Oct 19 17:58:38 box.lan dhcpcd[417]: eth0: soliciting a DHCP lease
Oct 19 17:58:43 box.lan dhcpcd[417]: eth0: probing for an IPv4LL address
Oct 19 17:58:48 box.lan dhcpcd[417]: eth0: using IPv4LL address 169.254.144.110
Oct 19 17:58:48 box.lan dhcpcd[417]: eth0: adding route to 169.254.0.0/16
Oct 19 17:58:48 box.lan dhcpcd[417]: eth0: adding default route
```

holta, 2017-10-20:
@jvonau I got the "/bin/bash: npm: command not found" error to go away (and Sugarizer role to complete on the install begun via WiFi) by replacing this:
```
- name: Install sugarizer required packages - is_debuntu
  package: name={{ item }}
           state=present
  when: internet_available and is_debuntu
  with_items:
- nodejs
```
With this:
```
- name: Install nodejs=6.* which also installs npm - is_debuntu
  package: name=nodejs=6.*
           state=present
  when: internet_available and is_debuntu
```
Can this strategy work when "non is_debuntu" as well, allowing further/substantial code cleanup of [roles/sugarizer/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml) ?

holta, 2017-10-20:
Networking challenges (after a WiFi-based install of IIAB completes) continue @ #438

Am closing #432 as I've now done 2 subsequent IIAB-installs-via-WiFi without Kiwix's install failing.  Plz do re-open this ticket as nec, if this reoccurs!

closed 2017-10-20 by holta